zephyrproject-rtos/zephyr #57198: bsim module must be cloned for west to work

jgl-meta, 2023-04-24:
**Describe the bug**
When running west a prompt that looks like the following appears. 
`
FATAL ERROR: failed manifest import in bsim (tools/bsim)
  Failed importing "west.yml" from revision "908ffde6298a937c6549dbfa843a62caab26bfc5"
  Hint: for this to work:
          - bsim must be cloned
          - its manifest-rev ref must point to a commit with the import data
        To fix, run:
          west update
`
this seems to be a little backwards since it forces pulling in additional modules, outside of the minimum necessary to build.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the getting started guide without cloning additional modules.
2. Run a west command such as update or init
You will get an error message that looks like the one above.

**Expected behavior**
We should still be able to build without the bsim module. Or at least we should provide a work around to build without it.

**Environment (please complete the following information):**
 - OS: Seen on MacOS and Linux

cfriedt, 2023-04-24:
I remember this was a topic at a TSC Meeting recently. It didn't occur to me at the time that it would create a mandatory module requirement. It's kind of against the ideas discussed in #54276 and #55056 .

Would be interesting to investigate in a bit of depth - I'm sure we can come up with a solution. Who is the best assignee for it?

carlescufi, 2023-04-24:
> Run a west command such as update or init

@jgl-meta `west init` will never fail this way. `west update` should not either unless you pass it an additional param.

cfriedt, 2023-04-24:
I updated the description - able to repro fairly easily with `west status`.

aescolar, 2023-04-24:
CC @mbolivar 
(To the best of my understanding) The issue is just a technicality around west. If the manifest has a project with an import, west needs that project locally to do anything beyond "init" and "update", as otherwise what west tries to do is collate the whole manifest (which requires the submanifests from the imports) and will fail when it does not have the submanifests from the imports.
Note that the "bsim" project is not the whole bsim simulator. It is just 5 files which include the submanifest that is being imported.

aescolar, 2023-04-24:
Created an issue in west to track possible improvements https://github.com/zephyrproject-rtos/west/issues/652

nordicjm, 2023-04-25:
@cfriedt I think you accidentally edited my comment, so I'm reverting it and putting your comment here:

@nordicjm - you can also run west update [module names...] to get only the modules you need without editing west.yml.

Same here though - this is a showstopper for our workflow. I haven't looked into the relevant west code myself, so I can't really speculate at a solution. There were some possibilities mentioned in a linked issue though.

cfriedt, 2023-04-25:
> @cfriedt I think you accidentally edited my comment, so I'm reverting it and putting your comment here:

@nordicjm - sorry about that - fat fingers and not enough ☕ . Thanks for catching it 👍

nordicjm, 2023-04-26:
@nashif pinging you as I vaguely remember you were the one I saw comment that non-zephyr repos weren't allowed in west.yml, can we not move these files in to zephyr or do something with it? The rule doesn't really make any sense since the `bsim-manifest` repo is now required to be present to do anything with zephyr and it imports the west.yml from that repo, that repo is hosted on zephyr and adds third party repos as per https://github.com/zephyrproject-rtos/babblesim-manifest/blob/main/west.yml so I don't really see the point in what moving it to a separate repo but making that repo required did

aescolar, 2023-04-26:
@nordicjm Please note that people is already using this bsim_manifest repo and the bsim project in the main manifest to fetch the simulator. We can't just remove it, or move things around without careful thought.
So, "let's  just revert that commit" is hardly a good solution. That will just cause another minor breakage for whoever is doing adhoc things (instead of a full west update), but adapted to needing bsim (by maybe doing a `west update bsim` in their scripting); but more importantly it will cause a worse break for those who started getting and using the simulator in this way. This was added because quite many users requested it. I can only imagined some of those are using it already.

This issue of "being forced to at the very least fetch the bsim manifest repo, if you don't want to do a plain west update" to be able to work with west with the zephyr main manifest, seems to me a relatively minor inconvenience we can live with while we do something nicer (maybe in the west side).
So far I haven't heard of any other issue about this, apart from some users who were not following the default workflow having been surprised, and maybe annoyed, that their workflow broke, and they did not know why. But once you know why, fixing it is adding 1 extra command (and using 400KB of disk space and 1 second for an extra git fetch).

nordicjm, 2023-04-26:
> @nordicjm Please note that people is already using this bsim_manifest repo and the bsim project in the main manifest to fetch the simulator. We can't just remove it, or move things around without careful thought.

I don't remember being asked for thoughts for having this change added in the first place, hence why I'm bringing it up here. This change has not been put into any releases so the true number of users it will cause problems for is not known, it has actually completely broken a private CI task I have for synchronising zephyr to a different git server because I can no longer run e.g. `west list` to just get a list of repos.

aescolar, 2023-04-26:
Sorry, but why is this a release blocker? The only background I can see in the TSC mom that @cfriedt mentioned it during the meeting even that it was not in the TSC meeting agenda.
I would request that we don't overblow minor issues..

cfriedt, 2023-04-26:
@aescolar - the release blocker tag is mainly to get flagged at the next release meeting.

I honestly don't think the ideal solution will be a lot of work, so I don't want you to be worried.

Given that the bsim module is so small, is there a possibility it can be added to the main repo? I did see that you presented a number of possible solutions as well. Thanks for doing that.

You can unassign yourself if it helps for now.

aescolar, 2023-04-26:
> @aescolar - the release blocker tag is mainly to get flagged at the next release meeting.

Instead of discussing this at the next release meeting (Please note those meetings are at a quite inconvenient time for people working in Europe), could you please check if the solution @mbolivar proposed for your particular problem is ok?
Otherwise it seems we are not spending our time wisely.

aescolar, 2023-04-26:
Removing the "release blocker", as this seems to require meta to understand their problem, and evaluate a proposed solution.

marc-hb, 2023-04-27:
>  I'm not against bsim particularly, I am against this whole concept in general. Once one module is allowed, that's it for allowing all modules to work like this. 

Agreed 200%. This is simply not scalable at all.

User-friendliness is critical, especially around version control which too many developers don't want to hear about. I understand the #55696 desire to make life easier for babblesim users but what about users of other modules? Can they all stuff their default locations in zephyr/west.yml too? This is obviously not scalable.

Ironically, all other users now have to know about the secret `west update bsim` trick. How user friendly!

Downloading everything by default with a plain `west update` is slow and not user friendly either - and obviously not scalable. It should always be possible to `west update only_what_I_need` out of the box.

The official `zephyr/west.yml` should be kept simple and point only at official zephyrproject-rtos projects, directly or indirectly. It probably does not need any import.
 
This is the flawed, "manifest stuffing" opt-in logic in a nutshell:

- Stuff default manifests with many locations by default for user friendliness
- group-disable them by default  because most users don't want them (`group-filter: [-babblesim]`)
- Ask users who want opt-ins to "disable the disable"

The usual, scalable opt-in approach is simply download and drop opt-ins in a well known `submanifests/` location. Almost as user friendly, a familiar approach, infinitely scalable and no "double negation". I already pointed at this flawed opt-in approach in 2021 in  https://github.com/zephyrproject-rtos/west/pull/481 but I was "overruled" :-)

marc-hb, 2023-04-27:
> as this seems to require meta to understand their problem

This sounds quite dismissive. Commit 01a12f658c5b broke `west update only_what_I_need` and there's nothing specific to Meta about that. Meta just happened to hit this first, this commit was merged less than one week ago. Reverting 01a12f658c5b make everything instantly work again.

`west update only_what_I_need` has been working with `zephyr/west.yml` since forever so I think this is major regression that deserves a priority higher than "low". This type of manifest stuffing of the default `zephyr/west.yml` is also crossing a major line IMHO.

PS @jgl-meta: I took the liberty to add `west update only_what_I_need`  to your description above, please review.

marc-hb, 2023-04-28:
> @nordicjm Please note that people is already using this bsim_manifest repo and the bsim project in the main manifest to fetch the simulator. We can't just remove it, or move things around without careful thought.
So, "let's just revert that commit" is hardly a good solution. That will just cause another minor breakage for whoever is doing adhoc things (instead of a full west update), but adapted to needing bsim (by maybe doing a west update bsim in their scripting); but more importantly it will cause a worse break for those who started getting and using the simulator in this way. This was added because quite many users requested it. I can only imagined some of those are using it already.

I don't think this matters. This is a development branch so breakages are expected regularly (e.g. #55290, #57127, #57298,...). What matters is how things will work in the future. This commit has been merged only one week ago whereas `west update only what I need` has been working since the start of the project; so if we cared about the past (we don't) then this commit should absolutely be reverted.

cfriedt, 2023-04-28:
> Removing the "release blocker", as this seems to require meta to understand their problem, and evaluate a proposed solution.

So again (I've already mentioned this) - please do not single out Meta. We've already heard from at least 3 or 4 companies that this is a problem. Those companies include (in chronological order FWIU):
1. Intel
2. Meta
3. STM
4. Nordic

Anas also mentioned it shortly before going on vacation on Discord.

jgl-meta, 2023-05-02:
In the release meeting we have discussed this issue and it is agreed that this is a release blocker.

nashif, 2023-05-03:
The part I did not know about is the fact that the group filter works on the imported repos and not the entry in the west manifest itself, so:

```
  group-filter: [-babblesim]

  #
  # Please add items below based on alphabetical order
  projects:
    - name: bsim
      repo-path: babblesim-manifest
      revision: 908ffde6298a937c6549dbfa843a62caab26bfc5
      import:
        path-prefix: tools
```

means that the filtering does not apply to the bsim module and it can't be filtered out,  `group-filter: [-babblesim]
` works on the repos listed in the babbesim-manifest repo, so this bsim module becomes mandatory, i.e. you will always need it.

@mbolivar-nordic  @aescolar   What is the rationale or constraints for not allowing filtering when `import` is used? why can't we filter out the parent repo?

aescolar, 2023-05-03:
> @mbolivar-nordic @aescolar What is the rationale or constraints for not allowing filtering when import is used? 

@nashif It is described here: https://github.com/zephyrproject-rtos/west/issues/543

yperess, 2023-05-15:
I'm curious what's the status of this issue? For me running `west update bsim` isn't working. I have to revert https://github.com/zephyrproject-rtos/zephyr/commit/01a12f658c5b8c4bd985a581e3753fd67d1ee10a locally

mbolivar-nordic, 2023-05-15:
We are working on a new version of west to solve this problem

nordicjm, 2023-05-25:
> After a discussion between @mbolivar-nordic @carlescufi @tejlmand, @57300 and me , it was agreed @mbolivar-nordic will continue with option 7 of [zephyrproject-rtos/west#652](https://github.com/zephyrproject-rtos/west/issues/652) : [zephyrproject-rtos/west#653](https://github.com/zephyrproject-rtos/west/issues/653) as a fix for this

So long as this does not need someone to change the configuration locally using a local west configuration file and it can be done through west manifests, that's fine, otherwise it's a non-starter.

nashif, 2023-05-25:
> So long as this does not need someone to change the configuration locally using a local west configuration file and it can be done through west manifests, that's fine, otherwise it's a non-starter.

agree with that if I understood option 7 correctly:  If I have bsim in the manifest, but not in my workspace, I will not be able to use west correctly and `wst status` gives me an error, option 7 will let me exclude `bsim` or any entry in the manifest and things will start working again? Hows does this solve the problem of west not working correctly due to missing bsim? Most likely one would end up reverting the bsim manifest commit like @yperess  did in one of his PRs before looking into setting options in west config

But maybe I am wrong about what options 7 is supposed to do.

yperess, 2023-05-25:
> agree with that if I understood option 7 correctly:  If I have bsim in the manifest, but not in my workspace, I will not be able to use west correctly and `wst status` gives me an error, option 7 [...]

I would 100% just revert it locally. I don't think it would cross my mind to check the west config.

marc-hb, 2023-05-25:
> option 7 will let me exclude bsim or any entry in the manifest and things will start working again? Hows does this solve the problem of west not working correctly due to missing bsim?

Because it would be disabled _by default_.

Copying part of https://github.com/zephyrproject-rtos/west/pull/660#issuecomment-1560747701
```diff 
--- a/west.yml
+++ b/west.yml
@@ -15,6 +15,8 @@
 # information.
 
 manifest:
+  version: "1.1"
+
   defaults:
     remote: upstream
 
@@ -23,6 +25,7 @@ manifest:
       url-base: https://github.com/zephyrproject-rtos
 
   group-filter: [-babblesim]
+  import-group-filters: false
 
   #
   # Please add items below based on alphabetical order
@@ -30,6 +33,8 @@ manifest:
     - name: bsim
       repo-path: babblesim-manifest
       revision: 908ffde6298a937c6549dbfa843a62caab26bfc5
+      groups:
+        - babblesim
       import:
         path-prefix: tools
     - name: canopennode
```

marc-hb, 2023-05-25:
As far back as [2021](https://github.com/zephyrproject-rtos/west/pull/481#issuecomment-778000066) I found stuffing manifests with URLs _disabled by default_  not intuitive  when you can just [drop random `.yml` in a well known `submanifests.d/`](https://github.com/zephyrproject-rtos/zephyr/issues/57198#issuecomment-1526052490) subdirectory instead (or delete some)

Two years later this approach might finally work?

mbolivar-nordic, 2023-05-31:
New approach: https://github.com/zephyrproject-rtos/west/pull/664

carlescufi, 2023-05-31:
@nashif @yperess @nordicjm with the approach in https://github.com/zephyrproject-rtos/west/pull/664  all you need to do is to type this *one single time* on your command-line:

`west config manifest.project-filter -- -bsim`

carlescufi, 2023-05-31:
> So long as this does not need someone to change the configuration locally using a local west configuration file and it can be done through west manifests, that's fine, otherwise it's a non-starter.

@nordicjm this only requires you to alter the local config **once**, since this setting will be then stored in your `~/.westconfig` once and forever.

nordicjm, 2023-06-05:
> @nordicjm this only requires you to alter the local config **once**, since this setting will be then stored in your `~/.westconfig` once and forever.

But can't be changed by a manifest itself which e.g. does an import as the original solution discussed was able to do

carlescufi, 2023-06-05:
> But can't be changed by a manifest itself which e.g. does an import as the original solution discussed was able to do

Unfortunately that's correct. We have tried in every way possible to make that work but we have not been able to.

> > @nashif @yperess @nordicjm with the approach in [zephyrproject-rtos/west#664](https://github.com/zephyrproject-rtos/west/pull/664) all you need to do is to type this **one single time** on your command-line:
> > `west config --global manifest.project-filter -- -bsim`
> 
> By this I assume you mean **+** bsim to enable it not, not **-** bsim to disable it since users should not have to disable this, it should not be present by default?

No, users will have it by default, at least for the time being. But you will be able to disable it permanently locally so that it doesn't affect you at all.

nordicjm, 2023-06-05:
> No, users will have it by default, at least for the time being. But you will be able to disable it permanently locally so that it doesn't affect you at all.

This is not a solution anyone with this problem has agreed to, so doesn't resolve anything for anyone with the above issues. @cfriedt

caco3, 2023-06-05:
Is there a workaround yet so I can test with the `3.4-RC2`?
I get this error:
```
CMake Error at /opt/zephyrproject-v3.4.0-rc2/zephyr/cmake/modules/zephyr_module.cmake:77 (message):
  ERROR: ManifestImportFailed: project <Project bsim
  ('/opt/zephyrproject-v3.4.0-rc2/tools/bsim') at
  908ffde6298a937c6549dbfa843a62caab26bfc5> value west.yml
```

Also since `3.4`, I have to call west with `sudo` to have the `build `option available.
(Note that I do not use `west.yaml` but use zephyr in `/opt`).

closed 2023-06-08 by nashif