zephyrproject-rtos/zephyr #57204: Output printf() on multiple zephyr,console set UARTs

bdesterBE, 2023-04-24:
**Describe the bug**
I have been trying to find a solution to be able to output the Zephyr console to multiple UART peripherals simultaneously. The solution I have found is setting `zephyr,console` in my board's .dts file to multiple UARTs on separate lines, i.e.,
```
chosen {
    zephyr,console = &uart7;
    zephyr,console = &usart3;
}
```
I have noticed this works to output printk() calls to both UARTs, but not printf() calls. printf() will only output to the second UART that `zephyr, console` is set to in the .dts. I am wondering if what I am doing is a hack and is not known/intended behavior to begin with, if configuration for printf() (presumably with use of `CONFIG_STDOUT_CONSOLE`) just does not account for this and could possibly be fixed to do so, or if possibly there is another configuration that already exists that I am missing to enable this.

**Expected behavior**
Calls to printf() output to each peripheral that `zephyr,console` is set to in a .dts file.

**Environment**
 - Custom board based off nucleo_h743zi (STM32 H7)
 - OS: Linux, MacOS
 - Zephyr v3.3.0
 - Zephyr SDK 0.16.0

jgl-meta, 2023-04-25:
We don't expect this work. The second line should override the first.

closed 2023-04-25 by jgl-meta

bdesterBE, 2023-04-26:
Thanks for the response. I'm still curious if there is currently a proper way of being able to print to multiple UARTs that I am not aware of. I wouldn't want to go this route as a solution to find that it ends up being patched out in a later version of Zephyr.