zephyrproject-rtos/zephyr #57219: drivers: spi: stm32: STM32H7 SPI don't work when when the core clock is high (>200Mhz)

valkjsaaa, 2023-04-25:
STM32H7 SPI driver doesn't work when the core clock is high (>200Mhz).

I am using STM32H743VI (custom board) and STM32H723ZG (nucleo_H723ZG) with LSM6DSL, when I set the main clock to be above 200Mhz, I can't read out the chip ID from LSM6DSL.

Workaround that I have tried:
* reduce SPI speed (I have tried 10M, 5M, 1M, no luck)
* reduce APB clock by increasing `hpre`, `d2ppre1`. helpful, but still can't get by 200Mhz. Notably, when I'm using 3.3V with nucleo_H723ZG, getting 200Mhz is easy. With 1.8V, I have to reduce APB to less than 10Mhz and some configurations may work.
* As noted above, it seems that this issue could be voltage related, 3.3V IO is slightly better than 1.8V IO.
* Also tried older Zephyr up to 3.0, still doesn't work

How to reproduce:
* Use nucleo_H723ZG at 3.3V, connect LSM6DSL to SPI 1 pins.
* Apply `400mhz.diff` as attached.
* Build and flash using `west build -b nucleo_h743zi samples/sensor/lsm6dsl/ && west flash -r stm32cubeprogrammer`
* Run serial monitor
* It would say `sensor: device not ready.`
* Further inspection shows that the CHIP_ID read from LSM6DSL wrong.

As a comparison, apply `200mhz.diff` at 3.3V, it would print gyroscope and accelerometer values as expected.

Environment (please complete the following information):
 - OS: macOS
 - Toolchain: Zephyr SDK
 - Zephyr v3.3.0 or 07c6af3b8c35c1e49186578ca61a25c76e2fb308

[200mhz.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/11319679/200mhz.diff.txt)
[400mhz.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/11319680/400mhz.diff.txt)

gautierg-st, 2023-04-26:
I tried your PLL configuration in STM32CubeMX and in your 400MHz case, it told me that the SPI kernel clock is too high. See the following diagram.
[image]
I would advise you to use this tool to find a good configuration for your clocks as it immediately tells you if there is a problem.

valkjsaaa, 2023-04-27:
[400mhz.diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/11338955/400mhz.diff.txt)
Sorry. My bad. Just tried something like this and it still doesn't work.

valkjsaaa, 2023-05-08:
any updates on this issue?

valkjsaaa, 2023-05-13:
I did some extra experiments for this bug.

I used a logic analyzer to check the signal (with 400mhz.diff.txt) and this is the result:

[image]
As you can see, the CS select is dangerously close to a clock change from SPI initialization.
Here is a zoomed-in view. The CS is happening after the clock down-edge, but still very close.
[image]

So I tried a fix by adding a bit of wait between SPI enable and CS signal sent out and it worked!
[spi_wait.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/11468428/spi_wait.patch.txt)

Here is the signal under the logic analyzer:
[image]

Is a `k_busy_wait(1);` a clean solution, or is there a better approach to this? I'm happy to submit a pull request to fix this error if there is a clean fix.

valkjsaaa, 2023-05-16:
Any thoughts on this @gautierg-st ?

gautierg-st, 2023-05-17:
Good catch about that. After seeing your patch, I remembered looking in the errata sheet for the H743 (ES0392) and the problem you are encountering seems to be listed:
[image]
So I agree with the solution, but it should apply only to STM32H7 series (and mention the errata in a comment).
Could you make a PR for this?

PR #58227 (2023-06-05): drivers: spi: stm32h7: add a small delay after enabling
Add a small delay after enabling SPI to prevent transfer stalling.
Fixes #57219