zephyrproject-rtos/zephyr #57222: STM32CO - Add support stm32c011xx

Ereaey, 2023-04-25:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
the soc of c0 being supported, it would be necessary to take advantage of it to include the first packages directly

**Describe why you are asking for this support?**
I plan to use this version of c0 but I don't have a card yet.

**Additional context**
https://www.st.com/en/microcontrollers-microprocessors/stm32c011j6.html

erwango, 2023-04-25:
@Ereaey Don't hesitate to contribute support for this SoC if you need it.
Here is an example on how to add a SoC support on an already existing series: https://github.com/zephyrproject-rtos/zephyr/pull/57126

closed 2023-07-09 by github-actions[bot]

erwango, 2023-07-10:
@Ereaey Please reopen if required