zephyrproject-rtos/zephyr #57236: Building MCUboot fails, -DCONFIG, own pem file

JiiZR, 2023-04-25:
**Describe the bug**
I have generated own signature file and I tried to build MCUboot but kconfig fails.
I have tried also default file (root-rsa-2048.pem) but build still fails.

 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. generate pem file: \bootloader\mcuboot\scripts\python imgtool.py keygen -k own-root-rsa-2048.pem -t rsa-2048
2. copy to pem file to bootloader\mcuboot
3.  west build -p always -b  b_u585i_iot02a boot\zephyr -DCONFIG_BOOT_SIGNATURE_KEY_FILE="own-root-rsa-2048.pem"
4. See error

**Expected behavior**
build should be OK

**Impact**
build fails

**Logs and console output**

C:\Data\ZEP\zephyr_latest\bootloader\mcuboot>west build -p always -b  b_u585i_iot02a boot\zephyr -DCONFIG_BOOT_SIGNATURE_KEY_FILE="own-root-rsa-2048.pem"
-- west build: making build dir C:\Data\ZEP\zephyr_latest\bootloader\mcuboot\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.2
-- Found Python3: C:/Python311/python.exe (found suitable exact version "3.11.2") found components: Interpreter
-- Cache files will be written to: C:/Data/ZEP/zephyr_latest/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/Data/ZEP/zephyr_latest/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: b_u585i_iot02a
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.0 (C:/Build/zephyr-sdk-0.16.0)
-- Found toolchain: zephyr 0.16.0 (C:/Build/zephyr-sdk-0.16.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Data/ZEP/zephyr_latest/zephyr/boards/arm/b_u585i_iot02a/b_u585i_iot02a.dts
-- Found devicetree overlay: C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/dts.cmake
C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/zephyr.dts:709.7-732.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/zephyr.dts:627.30-656.5: Warning (spi_bus_bridge): /soc/octospi@420d2400: node name for SPI buses should be 'spi'
<stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'

C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/misc/generated/extra_kconfig_options.conf:2: warning: malformed string literal in assignment to BOOT_SIGNATURE_KEY_FILE (defined at C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/boot/zephyr/Kconfig:129). Assignment ignored.Parsing C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'C:/Data/ZEP/zephyr_latest/zephyr/boards/arm/b_u585i_iot02a/b_u585i_iot02a_defconfig'
Merged configuration 'C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration 'C:/Data/ZEP/zephyr_latest/bootloader/mcuboot/build/zephyr/misc/generated/extra_kconfig_options.conf'

error: Aborting due to Kconfig warnings

CMake Error at C:/Data/ZEP/zephyr_latest/zephyr/cmake/modules/kconfig.cmake:329 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/Data/ZEP/zephyr_latest/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  C:/Data/ZEP/zephyr_latest/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Data/ZEP/zephyr_latest/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:31 (find_package)

-- Configuring incomplete, errors occurred!
←[91mFATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=C:\Python311\python.exe' '-BC:\Data\ZEP\zephyr_latest\bootloader\mcuboot\build' -GNinja -DBOARD=b_u585i_iot02a -DCONFIG_BOOT_SIGNATURE_KEY_FILE=own-root-rsa-2048.pem '-SC:\Data\ZEP\zephyr_latest\bootloader\mcuboot\boot\zephyr'
←[0m

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain zephyr-sdk-0.16.0
 - latest codes from github

**Additional context**
This is not MCUboot problem, more like tool related issue. There are workarouds but I would like to use this -DCONFIG.

nordicjm, 2023-04-25:
You have not escaped the command, try: `west build -p always -b b_u585i_iot02a boot\zephyr '-DCONFIG_BOOT_SIGNATURE_KEY_FILE="own-root-rsa-2048.pem"'`

Closing as user error, not a bug.

JiiZR, 2023-04-25:
Hmm, Did you try to build ?
I'am getting:

C:\Data\ZEP\zephyr_latest\bootloader\mcuboot>west build -p always -b  b_u585i_iot02a boot\zephyr '-DCONFIG_BOOT_SIGNATURE_KEY_FILE="own-root-rsa-2048.pem"'
-- west build: making build dir C:\Data\ZEP\zephyr_latest\bootloader\mcuboot\build pristine
-- west build: generating a build system
CMake Warning:
  Ignoring extra path from command line:

   "'-DCONFIG_BOOT_SIGNATURE_KEY_FILE=own-root-rsa-2048.pem'"

nordicjm, 2023-04-25:
```
west build -p always -b nrf52840dk_nrf52840 boot/zephyr '-DCONFIG_BOOT_SIGNATURE_KEY_FILE="own-root-rsa-2048.pem"'
-- west build: making build dir /tmp/aa/bootloader/mcuboot/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/aa/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.26.3
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.10") found components: Interpreter 
-- Cache files will be written to: /home/jamie/.cache/zephyr
-- Zephyr version: 3.3.99 (/tmp/aa/zephyr)
...
MCUBoot bootloader key file: /tmp/aa/bootloader/mcuboot/own-root-rsa-2048.pem
-- Configuring done (4.6s)
-- Generating done (0.1s)
-- Build files have been written to: /tmp/aa/bootloader/mcuboot/build
-- west build: building application
ninja: error: '/tmp/aa/bootloader/mcuboot/own-root-rsa-2048.pem', needed by 'zephyr/autogen-pubkey.c', missing and no known rule to make it
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /tmp/aa/bootloader/mcuboot/build
```

nordicjm, 2023-04-25:
You might need to escape the quotes as \" on windows

JiiZR, 2023-04-26:
So, Windows syntax is:

west build -p auto -b b_u585i_iot02a boot\zephyr -DCONFIG_BOOT_SIGNATURE_KEY_FILE=\"own-root-rsa-2048.pem\"

Thanks for the support and I'am sorry about the hassle.