zephyrproject-rtos/zephyr #57242: drivers: flash: shell: merge flash sample into flash shell

jeppenodgaard, 2023-04-25:
**Is your enhancement proposal related to a problem? Please describe.**
`zephyr/drivers/flash/flash_shell.c` and `zephyr/samples/drivers/flash_shell/src/main.c` has a lot of the same functionality which leads to redundancy.

**Describe the solution you'd like**
Merge flash sample code into flash shell. Empty the flash sample main and just build the flash shell in the flash sample.

**Additional context**
Discussion started here: https://github.com/zephyrproject-rtos/zephyr/pull/57233

kartben, 2023-09-14:
Just spotted this as part of reviewing existing driver samples. @jeppenodgaard would you be able to work on a pull request addressing this? Ultimately the sample should just be a main with vanilla flash shell enabled.

jeppenodgaard, 2023-09-14:
@kartben I think there are some pretty nice functionality in the sample that is worth merging.
I can give it a shot (although IMO not the most fun issue to work on), if nobody else is really keen on it :)

Here is what the commands looks like today (183b84d0e22d95541b6c6950a0b6b595fed2e3c4)

The shell commands :
```
$ flash --help
flash - Flash shell commands
Subcommands:
  erase      :[<device>] <page address> [<size>]
  read       :[<device>] <address> [<Dword count>]
  test       :[<device>] <address> <size> <repeat count>
  write      :[<device>] <address> <dword> [<dword>...]
  load       :[<device>] <address> <size>
  page_info  :[<device>] <address>
```
The sample commands:
```
$ flash_sample --help
flash_sample - Flash related commands.
Subcommands:
  erase             :<off> <len>
                     Erase <len> bytes from device offset <off>, subject to
                     hardware page limitations.
  page_count        :Print the number of pages on the flash device.
  page_erase        :<page> [num]
                     Erase [num] pages (default 1), starting at page <page>.
  page_layout       :[start_page] [end_page]
                     Print layout of flash pages in the range [start_page,
                     end_page], which is inclusive. By default, all pages are
                     printed.
  page_read         :<page> <len> OR <page> <off> <len>
                     Read <len> bytes from given page, starting at page offset
                     <off>,or offset 0 if not given. No checks are made that
                     bytes read are all within the page.
  page_write        :<page> <off> <byte1> [... byteN]
                     Write given bytes to given page, starting at page offset
                     <off>. No checks are made that the bytes all fall within
                     the page. Pages must be erased before they can be written.
  read              :<off> <len>
                     Read <len> bytes from device offset <off>.
  set_device        :<device_name>
                     Set flash device by name. If a flash device was not found,
                     this command must be run first to bind a device to this
                     module.
  write             :<off> <byte1> [... byteN]
                     Write given bytes, starting at device offset <off>.
                     Pages must be erased before they can be written.
  write_block_size  :Print the device's write block size. This is the smallest
                     amount of data which may be written to the device, in
                     bytes.
  write_unaligned   :<off> <byte1> [... byteN]
                     Write given bytes, starting at device offset <off>.
                     Being unaligned, affected memory areas are backed up,
                     erased, protected and then overwritten.
                     This command is designed to test writing to large flash
                     pages.
  write_pattern     :<off> <len>
                     Writes a pattern of (0x00 0x01 .. 0xFF 0x00 ..) of
                     length<len> at the offset <off>.
                     Unaligned writing is used, i.e. protection and erasing are
                     automated.This command is designed to test writing to large
                     flash pages.
  ```

jeppenodgaard, 2023-11-08:
Would it make sense to close this issue even though https://github.com/zephyrproject-rtos/zephyr/pull/63546 does not merge any code?

jeppenodgaard, 2023-11-23:
@kartben what do you think?

kartben, 2023-11-23:
Ya I wasn't sure, not a fan of losing track of what used to be nice command and taking the risk that someone might reinvent the wheel sometime in the future, not knowing that we used to have a few more nice commands written already.
But happy to close as won't fix if you think that's the right thing to do. Thanks for the extra ping!