zephyrproject-rtos/zephyr #57263: llvm/arm-clang fails to build STM32 platforms

galak, 2023-04-25:
when building with arm-clang or llvm for arm we get the following error:

```
/home/galak/git/zephyr/drivers/interrupt_controller/intc_exti_stm32.c:231:9: error: initializer element is not a compile-time constant
```

To reproduce, you can grab the ARM LLVM prebuilt here:

https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/releases

```
export ZEPHYR_TOOLCHAIN_VARIANT=llvm
export LLVM_TOOLCHAIN_PATH=<PATH_TO_LLVM>
```

```
west build -p -b nucleo_f767zi  -T tests/kernel/common/kernel.common
```

erwango, 2023-05-17:
Issue is coming from the method used to fill this array:
https://github.com/zephyrproject-rtos/zephyr/blob/fdb631c5d703e11e07f8d10e9c079b49135832fe/drivers/interrupt_controller/intc_exti_stm32.c#L209-L211
which is the way I've found to access the values of the following property:
```
			line-ranges = <0 1>, <1 1>, <2 1>, <3 1>,
				      <4 1>, <5 5>, <10 6>;
```
This could be replaced by something like:
```
	static const struct stm32_exti_range line_range_##idx = {	\
		DT_PROP_BY_IDX(node_id, line_ranges, UTIL_BY_TWO(idx)), \
		DT_PROP_BY_IDX(node_id, line_ranges, UTIL_INC(UTIL_BY_TWO(idx))) \
	};	
```

This would require the addition of 
```
#define UTIL_INC(x) UTIL_PRIMITIVE_CAT(UTIL_INC_, x)
#define UTIL_BY_TWO(x) UTIL_PRIMITIVE_CAT(UTIL_BY_TWO_, x)
with :
#define UTIL_INC_0 1
#define UTIL_INC_1 2
#define UTIL_INC_2 3
...
#define UTIL_BY_TWO_0 0
#define UTIL_BY_TWO_1 2
#define UTIL_BY_TWO_2 4
...
```

`UTIL_INC` used to be present, but was removed sometime ago in https://github.com/zephyrproject-rtos/zephyr/pull/32692 with no obvious replacement.

galak, 2023-05-17:
maybe we should re-introduce UTIL_INC.

erwango, 2023-05-17:
Unrelated: @galak, note that the LLVM compilation didn't complete, as I got the following error:
```
lang-16: warning: argument unused during compilation: '--specs=nosys.specs' [-Wunused-command-line-argument]
~/st/stm32cubeclt_1.11.1/GNU-tools-for-STM32/bin/arm-none-eabi-ld.bfd: zephyr/libzephyr.a(cbprintf_complete.c.obj): in function `extract_conversion':
~/zephyrproject/zephyr/lib/os/cbprintf_complete.c:645: undefined reference to `__aeabi_memclr8'
~/st/stm32cubeclt_1.11.1/GNU-tools-for-STM32/bin/arm-none-eabi-ld.bfd: zephyr/libzephyr.a(cbprintf_complete.c.obj): in function `encode_uint':
~/zephyrproject/zephyr/lib/os/cbprintf_complete.c:802: undefined reference to `__aeabi_uldivmod'
~/st/stm32cubeclt_1.11.1/GNU-tools-for-STM32/bin/arm-none-eabi-ld.bfd: zephyr/drivers/serial/libdrivers__serial.a(uart_stm32.c.obj): in function `LL_USART_SetBaudRate':
~/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_usart.h:1029: undefined reference to `__aeabi_uldivmod'
~/st/stm32cubeclt_1.11.1/GNU-tools-for-STM32/bin/arm-none-eabi-ld.bfd: ~/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/stm32f4xx_ll_usart.h:1029: undefined reference to `__aeabi_uldivmod'
~/st/stm32cubeclt_1.11.1/GNU-tools-for-STM32/bin/arm-none-eabi-ld.bfd: modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a(stm32f4xx_hal_rcc.c.obj): in function `HAL_RCC_GetSysClockFreq':
~/zephyrproject/modules/hal/stm32/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_rcc.c:916: undefined reference to `__aeabi_uldivmod'
clang-16: error: ld.lld command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

PR #58197 (2023-06-06): stm32: interrupt_controller: Access line-range prop elements at build time
Fixes #57263

Involves the re-introduction of UTIL_INC and new UTIL_X2 method.
For coherency UTIL_DEC is added as well.

EDIT: Fixes regression cause by https://github.com/zephyrproject-rtos/zephyr/pull/58421/commits/ba5e50c20006677788ac3a839227c09a10eaec88 (breaking communication on H5/U5/L5 targets)
closed 2023-06-06 by nashif