zephyrproject-rtos/zephyr #57267: RFC: API Change: Bluetooth Mesh

LingaoM, 2023-04-26:
- Problem Description: https://github.com/zephyrproject-rtos/zephyr/pull/57215
   - Change Define Mesh element as const, the reason is that the Mesh Profile clearly stipulates that Mesh nodes cannot change their own element definitions.
 - Proposed Change (detailed):
   - The `elem->addr` has been removed, just replace with API `bt_mesh_elem_addr_get`
 - Detailed RFC:
   - Function call changes
        The `elem->addr` has been removed, just replace with API `bt_mesh_elem_addr_get`
   - Device Tree changes (source and bindings)
      No
   - Kconfig option changes
      No
 - Dependencies:
    Any access `elem->addr` should replace with `bt_mesh_elem_addr_get`.

carlescufi, 2023-05-04:
@LingaoM you are missing a couple of steps:
https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes

In particular, you need to send an email to devel@ and bring the topic to the Architecture WG meeting (you don't need to be present, just ask for the topic to be discussed there).

LingaoM, 2023-05-05:
> @LingaoM you are missing a couple of steps: https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes
>
> In particular, you need to send an email to devel@ [...]

Thanks for the reminder, done.

LingaoM, 2023-11-20:
Merged

closed 2023-11-20 by LingaoM