zephyrproject-rtos/zephyr #57272: mimxrt1050_evk_qspi: mcuboot serial recovery: image upload fails after updating to Zephyr v3.3.0

kunohel, 2023-04-26:
**Describe the bug**
After upgrading to Zephyr v3.3.0 the image uploading during serial recovery mode is not working anymore.
I was working well before when I was using Zephyr V3.1.0

**To Reproduce**

usb_cdc_acm_log_recovery.conf is customized a bit to enable LED:
```
CONFIG_LOG=y

# The build won't fit on the partition allocated for it without size
# optimizations.
CONFIG_SIZE_OPTIMIZATIONS=y

# Serial
CONFIG_SERIAL=y
CONFIG_UART_LINE_CTRL=y

# MCUBoot serial
CONFIG_MCUBOOT_SERIAL=y
CONFIG_BOOT_SERIAL_CDC_ACM=y

CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_BACKEND_RTT=n

CONFIG_MCUBOOT_INDICATION_LED=y
```

and usb_cdc_acm.overlay have been customized to:

```
/ {
    aliases {
        mcuboot-led0 = &green_led;
	mcuboot-button0 = &user_button;
    };

	chosen {
		zephyr,uart-mcumgr = &zephyr_udc0;
	};
};

&zephyr_udc0 {
	cdc_acm_uart0 {
		compatible = "zephyr,cdc-acm-uart";
	};
};
```
I have tried without "chosen" also.

Steps to reproduce the behavior:
1. west build -b mimxrt1050_evk_qspi zephyrproject/bootloader/mcuboot/boot/zephyr/ -d build-boot -- -DOVERLAY_CONFIG="usb_cdc_acm_log_recovery.conf" -DDTC_OVERLAY_FILE="usb_cdc_acm.overlay;app.overlay"
2. west flash -d build-boot/
3. Build an app to use for uploading: 
4. west build -b mimxrt1050_evk_qspi -d build-boot-app zephyrproject/bootloader/mcuboot/samples/zephyr/hello-world/ -- DCONFIG_BOOTLOADER_MCUBOOT=y '-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE="zephyrproject/bootloader/mcuboot/root-rsa-2048.pem"'
5. sudo ~/go/bin/mcumgr --conntype serial --connstring "/dev/ttyACM1,baud=115200,mtu=512" image upload build-boot-app/zephyr/zephyr.signed.bin -l debug

**Expected behavior**
That the image uploading would succeed

**Logs and console output**
The output from debug console
```
*** Booting Zephyr OS build zephyr-v3.3.0 ***
I: Starting bootloader
I: Enter the serial recovery mode
I: Device configured
W: endpoint 0x1 already occupied
I: Erasing range 0x%jx:0x%jx
I: Writing at 0x0 until 0x129
I: RX: 0x0
I: TX
I: Writing at 0x129 until 0x27d
I: RX: 0x0
I: TX
```

The output from mcumgr
```
00000060  6f 4f 63 44 4c 46 32 42  72 61 54 4b 42 50 73 50  |oOcDLF2BraTKBPsP|
00000070  74 63 4b 39 76 70 64 58  2b 2b 58 53              |tcK9vpdX++XS| 
DEBU[2023-04-26 08:52:40.665] Tx serial
00000000  0a                                                |.| 
DEBU[2023-04-26 08:52:40.685] Tx serial
00000000  04 14                                             |..| 
DEBU[2023-04-26 08:52:40.685] Tx serial
00000000  76 61 5a 4c                                       |vaZL| 
DEBU[2023-04-26 08:52:40.685] Tx serial
00000000  0a                                                |.| 
DEBU[2023-04-26 08:52:40.761] Rx serial:
00000000  06 09 41 41 6f 44 41 41  41 41 41 41 46 43 41 59  |..AAoDAAAAAAFCAY|
00000010  54 4b                                             |TK| 
DEBU[2023-04-26 08:52:40.761] Decoded input:
00000000  03 00 00 00 00 01 42 01                           |......B.| 
DEBU[2023-04-26 08:52:40.761] rx nmp response: 00000000  03 00 00 00 00 01 42 01                           |......B.| 
DEBU[2023-04-26 08:52:40.761] Failure decoding NMP rsp: Invalid response: EOF
```

**Environment (please complete the following information):**
 - OS: Linux 22.04
 - Toolchain Zephyr 0.15.2
 - Zephyr v3.3.0

nordicjm, 2023-04-26:
Can you give a try with https://github.com/zephyrproject-rtos/zephyr/pull/57256 ?

kunohel, 2023-04-26:
After some more investigation, it seems like the qspi is being flashing correctly, but the response received after each bulk is wrong - causing it to take a very long time to flash, and the progress bar from mcumgr also stays at 0%.

nordicjm, 2023-04-26:
How long does it take to perform the QSPI erase/write? During that time, the module is probably "dead" but the go application will resend packets if it does not get a response which might cause problems with them. I think someone else was reporting an issue with USB CDC packets but on an STM32 board, @Frank-Buss maybe?

kunohel, 2023-04-26:
A quick analysis based on logging timestamps, writing does not take a long time

Output from minicom, where "Writing at 0xX until 0xXXX" is just before entering write operations, and "RX" is after.
So less than 3ms per 340 bytes.

[2023-04-26 13:09:43.087] I: Writing at 0x0 until 0x129
[2023-04-26 13:09:43.090] I: RX: 0x0
[2023-04-26 13:09:43.090] I: TX
[2023-04-26 13:09:46.094] I: Writing at 0x129 until 0x27d
[2023-04-26 13:09:46.097] I: RX: 0x0
[2023-04-26 13:09:46.098] I: TX
[2023-04-26 13:09:49.179] I: Writing at 0x27d until 0x3d1
[2023-04-26 13:09:49.182] I: RX: 0x0
[2023-04-26 13:09:49.183] I: TX
[2023-04-26 13:09:52.264] I: Writing at 0x3d1 until 0x525
[2023-04-26 13:09:52.266] I: RX: 0x0
[2023-04-26 13:09:52.267] I: TX
[2023-04-26 13:09:55.349] I: Writing at 0x525 until 0x679
[2023-04-26 13:09:55.352] I: RX: 0x0
[2023-04-26 13:09:55.352] I: TX

kunohel, 2023-04-26:
But when only 340 bytes is sent every 3 seconds it's taking a long time - I will try using another mcumgr version

Frank-Buss, 2023-04-26:
> How long does it take to perform the QSPI erase/write? During that time, the module is probably "dead" but the go application will resend packets if it does not get a response which might cause [...]

I found multiple problems. When erasing external flash, it needed too long and the timeout didn't work, because it is hardcoded to 3s. I fixed it:
https://github.com/apache/mynewt-newtmgr/pull/198
Not merged since a month, I guess the project is dead.

The Go program is horrible in other ways as well, which is the reason I started my own mcumgr-client program. I then encountered a problem with USB CDC in Zephyr with STM32 microcontrollers, see the notes in my program how to fix it:
https://github.com/vouch-opensource/mcumgr-client#notes

The underlying problem might be different, but I guess no point in investigating it more deeply, because the new Zephyr USB stack is around the corner.

nordicjm, 2023-04-26:
Can you also retry using `main` of zephyr (remember to `west update`) and see if the problem is still present there as maybe it has been fixed?

kunohel, 2023-04-27:
> 

Just tested with main and there is even more issues - mcuboot stops quite early in main without even initializing the usb.

It seems like something is blocking when entering the serial recovery mode

main.c
```
490:    BOOT_LOG_INF("Enter the serial recovery mode");
491:    rc = boot_console_init();     <----- get stuck here
```

nordicjm, 2023-04-27:
Can you attach a debugger and get a backtrace?

kunohel, 2023-04-27:
Here is some debugging:

```
*** Booting Zephyr OS build zephyr-v3.3.0-3225-ga004a85ccf53 ***
[00:00:00.001,000] <dbg> os: k_sched_unlock: scheduler unlocked (0x800090a0:0)
[00:00:00.001,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:00:00.001,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
[00:00:00.001,000] <dbg> mpu: region_init: [3] 0x80013440 0x150b000a
[00:00:00.001,000] <inf> mcuboot: Starting bootloader
[00:00:00.001,000] <dbg> os: setup_thread_stack: stack 0x80012300 for thread 0x80008ac0: obj_size=832 buf_s0
[00:00:00.001,000] <inf> mcuboot: Enter the serial recovery mode
[00:00:00.001,000] <dbg> usb_device: usb_enable: lock usb_enable_lock mutex
[00:00:00.001,000] <dbg> os: z_impl_k_mutex_lock: 0x800090a0 took mutex 0x800083c4, count: 1, orig prio: 0
[00:00:00.031,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:00:00.031,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
[00:00:00.031,000] <dbg> mpu: region_init: [3] 0x80012300 0x150b000a
[00:00:00.228,000] <dbg> os: z_tick_sleep: thread 0x80008ac0 for 300 ticks
[00:00:00.228,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:00:00.228,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
[00:00:00.228,000] <dbg> mpu: region_init: [3] 0x80013440 0x150b000a
[00:00:00.258,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:00:00.258,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
[00:00:00.258,000] <dbg> mpu: region_init: [3] 0x80012300 0x150b000a
[00:00:00.312,000] <dbg> os: z_tick_sleep: thread 0x80008ac0 for 300 ticks
[00:00:00.312,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:00:00.312,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
```

it is stuck in a loop or something, a minute later, it is still printing the same with a high frequency:
```
[00:01:46.098,000] <dbg> os: z_tick_sleep: thread 0x80008ac0 for 300 ticks
[00:01:46.098,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x3
[00:01:46.098,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x3
[00:01:46.098,000] <dbg> mpu: region_init: [3] 0x80013440 0x150b000a
```

nordicjm, 2023-04-27:
Maybe this fixes the issue you are currently seeing: https://github.com/zephyrproject-rtos/zephyr/pull/57313

nordicjm, 2023-04-27:
Seems that NXP has their own MPU file which is different to the normal ARM one. @mmahadevan108 @dlech seems like a bug in the NXP MPU code?

kunohel, 2023-04-27:
> > Could it be related to the newer updates with cbor? Since the issue seems to be found here in the mcumgr tool.
> > ~/go/pkg/mod/mynewt.apache.org/[newtmgr@v0.0.0-20201028150837-60b2da78788c](mailto:newtmgr@v0.0.0-20201028150837-60b2da78788c)/nmxact/nmp/decode.go
> > ```go
> > cborCodec := new(codec.CborHandle)
> > dec := codec.NewDecoderBytes(body, cborCodec)
> > 
> > if err := dec.Decode(r); err != nil {
> > return nil, fmt.Errorf("Invalid response: %s", err.Error())
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Maybe mcuboot is not responding with a valid cbor message anymore?
> 
> If the output on the go application is correct then there is a missing 0x0a character from the end of the response. The response decodes to `00 0a 03 00 00 00 00 01 42 01 84 ca` which is length 0x000a, OP 0x03 (write response), data length 0x0000, group: 0x0001 (application), sequence: 0x42, command: 0x01, crc: 0x84ca
> 
> So you are getting the correct response **minus** the expected newline character, maybe check with something like wireshark USB monitoring to see if you are really getting the final 0x0a character or not.

Just did a test monitoring usb traffic.

This is where the reponse is received by mcumgr
```
DEBU[2023-04-27 14:15:32.553] Tx serial
00000000  0a                                                |.| 
DEBU[2023-04-27 14:15:32.555] Rx serial:
00000000  06 09 41 41 6f 44 41 41  41 41 41 41 47 34 41 58  |..AAoDAAAAAAG4AX|
00000010  6a 41                                             |jA| 
DEBU[2023-04-27 14:15:32.555] Decoded input:
00000000  03 00 00 00 00 01 b8 01                           |........| 
DEBU[2023-04-27 14:15:32.555] rx nmp response: 00000000  03 00 00 00 00 01 b8 01                           |........| 
DEBU[2023-04-27 14:15:32.555] Failure decoding NMP rsp: Invalid response: EOF
packet=
00000000  03 00 00 00 00 01 b8 01                           |........|
```

And this is the data captured by a tcpdump:
```
060941416f44414141414141473441586a410a
```

It looks fine right? and with an ending newline '0a'

kunohel, 2023-04-27:
I just tried @Frank-Buss tool and it also throws an error while parsing the reponse.

```
13:36:50 [ERROR] Error: EOF while parsing a value
```

nordicjm, 2023-04-27:
Sounds like a bug with your client @Frank-Buss ? Not really sure what to suggest, not had any of these issues even with the go application on nordic devices

kunohel, 2023-04-27:
I guess something must have been introduced since it was working fine with Zephyr v3.1.0. I will wait and see if any from NXP reacts on this issue :-)

danieldegrasse, 2023-05-04:
@mmahadevan108 Any ideas here? I can reproduce the issue on an RT1060 EVK.

The USB CDC subsystem appears to write 19 bytes of data, and I do see 19 bytes transferred in a wireshark capture:
```
0000  06 09 41 41 6f 42 41 41  41 41 41 41 46 43 41 42   ··AAoBAA AAAAFCAB
0010  74 4e 0a                                           tN·
```

However, MCUMGR reports only receiving 18 bytes of data:

```
DEBU[2023-05-03 22:53:20.284] Rx serial:
00000000  06 09 41 41 6f 42 41 41  41 41 41 41 46 43 41 42  |..AAoBAAAAAAFCAB|
00000010  74 4e                                             |tN|
DEBU[2023-05-03 22:53:20.284] Decoded input:
00000000  01 00 00 00 00 01 42 00                           |......B.|
DEBU[2023-05-03 22:53:20.284] rx nmp response: 00000000  01 00 00 00 00 01 42 00
```

I cannot reproduce this issue on a K64 board, so I suspect the problem may be related to our USB driver.

Frank-Buss, 2023-05-04:
> Sounds like a bug with your client @Frank-Buss ? Not really sure what to suggest, not had any of these issues even with the go application on nordic devices

I don't think it is a bug in my tool, because the Go program reports the same EOF problem. It looks like the CBOR encoded part after the binary header is missing. Here is a full example what the Go program returns with the "image list" program (note, the final 0x0a is not listed as "Rx serial", but it is there, otherwise it would hang forever, which I had once as well, because of a bug in the STM32 USB driver) :
```
DEBU[2023-05-04 06:28:06.165] Using connection profile: name=acm0 type=serial connstring=dev=/dev/ttyACM0,baud=115200,mtu=512 
DEBU[2023-05-04 06:28:06.166] {add-nmp-listener} [serial_sesn.go:213] seq=16 
DEBU[2023-05-04 06:28:06.166] Encoded &{NmpBase:{hdr:{Op:0 Flags:0 Len:0 Group:1 Seq:16 Id:0}}} to:
00000000  a0                                                |.| 
DEBU[2023-05-04 06:28:06.166] Encoded:
00000000  00 00 00 01 00 01 10 00  a0                       |.........| 
DEBU[2023-05-04 06:28:06.166] Tx NMP request: 00000000  00 00 00 01 00 01 10 00  a0                       |.........| 
DEBU[2023-05-04 06:28:06.166] Base64 encoding request:
00000000  00 00 00 01 00 01 10 00  a0                       |.........| 
DEBU[2023-05-04 06:28:06.166] Tx serial
00000000  06 09                                             |..| 
DEBU[2023-05-04 06:28:06.166] Tx serial
00000000  41 41 73 41 41 41 41 42  41 41 45 51 41 4b 44 46  |AAsAAAABAAEQAKDF|
00000010  6e 51 3d 3d                                       |nQ==| 
DEBU[2023-05-04 06:28:06.166] Tx serial
00000000  0a                                                |.| 
DEBU[2023-05-04 06:28:06.168] Rx serial:
00000000  06 09 41 42 55 42 41 41  41 4c 41 41 45 51 41 4c  |..ABUBAAALAAEQAL|
00000010  39 6d 61 57 31 68 5a 32  56 7a 6e 2f 2f 2f 57 46  |9maW1hZ2Vzn///WF|
00000020  59 3d                                             |Y=| 
DEBU[2023-05-04 06:28:06.168] Decoded input:
00000000  01 00 00 0b 00 01 10 00  bf 66 69 6d 61 67 65 73  |.........fimages|
00000010  9f ff ff                                          |...| 
DEBU[2023-05-04 06:28:06.168] rx nmp response: 00000000  01 00 00 0b 00 01 10 00  bf 66 69 6d 61 67 65 73  |.........fimages|
00000010  9f ff ff                                          |...| 
DEBU[2023-05-04 06:28:06.168] Received nmp rsp: &{NmpBase:{hdr:{Op:1 Flags:0 Len:11 Group:1 Seq:16 Id:0}} Rc:0 Images:[] SplitStatus:N/A} 
DEBU[2023-05-04 06:28:06.168] {remove-nmp-listener} [serial_sesn.go:213] seq=16 
Images:
Split status: N/A (0)
```

Frank-Buss, 2023-05-04:
PS: my tool has also a "-v" switch for verbose output. Looks like this:
```
frank@hal9000:~$ mcumgr-client -v -d /dev/ttyACM0 list
mcumgr-client 0.0.1, Copyright © 2023 Vouch.io LLC

04:49:29 [INFO] send image list request
04:49:29 [DEBUG] (1) mcumgr_client::transfer: request header: NmpHdr { op: Read, flags: 0, len: 1, group: Image, seq: 72, id: 0 }
04:49:29 [DEBUG] (1) mcumgr_client::transfer: serialized: 0000000100014800a0
04:49:29 [DEBUG] (1) mcumgr_client::transfer: encoded with packet length and checksum: 000b0000000100014800a032f2
04:49:29 [DEBUG] (1) mcumgr_client::transfer: encoded: AAsAAAABAAFIAKAy8g==
04:49:29 [DEBUG] (1) mcumgr_client::transfer: result string: ABUBAAALAAFIAL9maW1hZ2Vzn///H2g=
04:49:29 [DEBUG] (1) mcumgr_client::transfer: response header: NmpHdr { op: ReadRsp, flags: 0, len: 11, group: Image, seq: 72, id: 0 }
04:49:29 [DEBUG] (1) mcumgr_client::transfer: cbor: bf66696d616765739fffff
04:49:29 [INFO] response: {
  "images": []
}
``` 
But would be nice to also show the transmitted and received bytes, I added an issue for it, should be easy to add:
https://github.com/vouch-opensource/mcumgr-client/issues/5

danieldegrasse, 2023-05-04:
Some more info about this failure: I can get the SMR_SVR sample `samples/subsys/mgmt/mcumgr/smp_svr` to work correctly with the RT1060 over CDC ACM, but MCUBoot with mcumgr fails. In the working sample, I see the ACM driver class sends 202 bytes as a response. For the failing case only 19 bytes are sent. This makes me suspect the RX side may have an issue. However, the data sent by mcumgr and the data received by MCUBoot seem to be the same:

host side (RT1060 target):
```
DEBU[2023-05-04 10:49:09.832] Tx serial
00000000  06 09                                             |..|
DEBU[2023-05-04 10:49:09.832] Tx serial
00000000  41 41 73 41 41 41 41 42  41 41 46 43 41 4b 44 31  |AAsAAAABAAFCAKD1|
00000010  4d 77 3d 3d                                       |Mw==|
DEBU[2023-05-04 10:49:09.832] Tx serial
00000000  0a  
```

device side (RT1060):
```
D: dev 0x6000ebe0 size 1 rx_ringbuf space 1024
RX: 06 09 41 41 73 41 41 41 41 42 41 41 46 43 41 4B 44 31 4D 77 3D 3D 0A
D: Got 19 bytes from ringbuffer send to ep 82
```

On the K64, the initial data received is similar:

host side (K64 target):
```
DEBU[2023-05-04 12:09:42.907] Tx serial
00000000  06 09                                             |..|
DEBU[2023-05-04 12:09:42.907] Tx serial
00000000  41 41 73 41 41 41 41 42  41 41 46 43 41 4b 44 31  |AAsAAAABAAFCAKD1|
00000010  4d 77 3d 3d                                       |Mw==|
DEBU[2023-05-04 12:09:42.907] Tx serial
00000000  0a
```

device side (K64):
```
D: dev 0xb664 size 1 rx_ringbuf space 1024
RX: 06 09 41 41 73 41 41 41 41 42 41 41 46 43 41 4B 44 31 4D 77 3D 3D 0A
D: Got 202 bytes from ringbuffer send to ep 83
```

@nordicjm Any insight as to why MCUBoot might respond differently on the RT1060 versus the K64?

nordicjm, 2023-05-05:
> but MCUBoot with mcumgr fails

I assume by this you mean mcuboot with serial recovery, and not mcuboot with mcumgr enabled, because the latter would be bad.

> @nordicjm Any insight as to why MCUBoot might respond differently on the RT1060 versus the K64?

I would assume a driver bug, there's no issue on nordic boards, and I don't remember seeing issues on stm32 when I last tested (which was some months ago)

danieldegrasse, 2023-05-23:
@nordicjm I've looked at this some more, and don't believe it is a driver bug- it seems like there is a bug in MCUBoot.

After some debugging, it appears that the ZCBOR structure set up by the MCUBoot is invalid. Specifically, this call to `zcbor_new_encode_state`: https://github.com/zephyrproject-rtos/mcuboot/blob/2878eb4e32fabc14bc391c3221cbf285064d9db4/boot/boot_serial/src/boot_serial.c#L135:

```
zcbor_new_encode_state(cbor_state, 2, (uint8_t *)bs_obuf, (size_t)bs_obuf + sizeof(bs_obuf), 0);
```

Appears to be using the ZCBOR API incorrectly. It seems that the API call should be: 
```
zcbor_new_encode_state(cbor_state, 2, (uint8_t *)bs_obuf, sizeof(bs_obuf), 0);
```
as the 4th parameter is expected to be the payload length: https://github.com/zephyrproject-rtos/zcbor/blob/67fd8bb88d3136738661fa8bb5f9989103f4599e/include/zcbor_encode.h#L272

When debugging on the RT1050, `bs_obuf + sizeof(bsobuf)` comes out to be `0x800114d`. The pointer `bs_obuf` is at `0x8001143`.

The implementation of [zcbor_new_encode_state](https://github.com/zephyrproject-rtos/zcbor/blob/67fd8bb88d3136738661fa8bb5f9989103f4599e/src/zcbor_common.c#L126) does the following:

```
state_array[0].payload = payload;
state_array[0].payload_end = payload + payload_len;
state_array[0].elem_count = elem_count;
```

So for a platform like the RT1050 where the SRAM address space is at a high address, `payload` + `payload_len` will cause an integer overflow. On the FRDM-K64, the SRAM range is at 0x2000_0000, so roughly doubling the address like occurs here will not cause an overflow.

We can see the result of this, where `payload_end` comes out to be `0x2291` on the RT1050. This becomes an issue when constructing a ZCBOR object, as checks like the following will now fail:
```
#define ZCBOR_CHECK_PAYLOAD() \
	ZCBOR_ERR_IF(state->payload >= state->payload_end, ZCBOR_ERR_NO_PAYLOAD)
```

This results in the RT1050 responding with invalid data to the request from mcumgr, as the ZCBOR structure itself used to encode image data is invalid, and therefore never has image info stored within it.

closed 2023-05-30 by danieldegrasse