zephyrproject-rtos/zephyr #57285: tests/kernel/timer/timer_behavior fails on the stm32 boards

FRASTM, 2023-04-26:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

FRASTM, 2023-04-26:
Could this error be similar to https://github.com/zephyrproject-rtos/zephyr/issues/56589 ?

FRASTM, 2023-04-26:
branch https://github.com/FRASTM/zephyr/tree/timer_behavior is an adaptation to PASS the testcase

teburd, 2023-05-05:
Missing a tick alignment when starting the timer? The short timer (one time) seems to be the issue here

FRASTM, 2023-07-19:
not fixed

erwango, 2023-09-25:
@teburd Any insight to progress on this ?

teburd, 2023-09-25:
There's many possibilities, but I'd guess we're missing a tick alignment on the first timer

so something like k_sleep(K_TICKS(2)); just before starting the test case might align the first scheduled timer to the tick better avoiding the one timeout occuring shorter than the rest.

Would you mind trying to add a k_sleep(K_TICKS(2)); just before sample_collection_fn() in do_test_using() in timer_behavior/jitter_drift.c and see if this fixes the issue?

erwango, 2023-09-26:
ok, I think @FRASTM can have a try when available.

FRASTM, 2023-09-26:
I confirm this is fixing the error

FRASTM, 2023-10-03:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/63210

closed 2023-10-03 by FRASTM