zephyrproject-rtos/zephyr #57288: tests/kernel/timer/timer_behavior fails on test_jitter_drift_timer_startdelay for some stm32 boards

FRASTM, 2023-04-26:
Executing the tests/kernel/timer/timer_behavior/ fails

failure occurs on the timer_jitter_drift.test_jitter_drift_timer_startdelay
nucleo_wb55rg
stm32l562e_dk

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_wb55rg tests/kernel/timer/timer_behavior/
2. west flash
3. See error

**Expected behavior**
Test PASSED

**Impact**
CI failure

**Logs and console output**
```

```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA d2917610a5ba2f190c7e8cfa49db93fab0e2f99e

**Additional context**
same assertion as reported by the https://github.com/zephyrproject-rtos/zephyr/issues/56589

FRASTM, 2023-04-26:
branch https://github.com/FRASTM/zephyr/tree/timer_behavior is an adaptation to PASS the testcase

galak, 2023-05-02:
@teburd

teburd, 2023-05-05:
This seems to be the same as #57285 where one timer runs short (probably the first one)

FRASTM, 2023-07-19:
not fixed

FRASTM, 2023-07-25:
Now (SHA1  d2e89eb512) seen with _nucleo_f429zi_ and _nucleo_l152re_
Reported also by https://github.com/zephyrproject-rtos/zephyr/issues/58729

teburd, 2023-09-26:
#63086 as a possible fix

FRASTM, 2023-10-03:
I confirm https://github.com/zephyrproject-rtos/zephyr/pull/63086  is fixing this error

FRASTM, 2023-10-03:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/63210

closed 2023-10-03 by FRASTM