zephyrproject-rtos/zephyr #57312: Regression: regulator enable GPIO glitch during boot

silbe, 2023-04-26:
**Describe the bug**

Since commit 539d4aa6e661c9100c866d9ae0b416fc5d14303f ("drivers: regulator: add common init enable API") by @gmarull fixed regulators that are enabled by the boot loader and marked in the Device Tree as `regulator-boot-on` are turned off for a very short time and turned back on again immediately. This usually violates timing restrictions, causing hardware to malfunction. It can also cause brown-outs.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a fixed regulator with enable GPIO in Device Tree. E.g.:
    ```
	oscen: oscen {
		compatible = "regulator-fixed-sync", "regulator-fixed";
		label = "oscen";
		regulator-name = "oscen";
		enable-gpios = <&gpioh 1 GPIO_ACTIVE_HIGH>;
		regulator-boot-on;
	};
    ```
2. Enable the GPIO in the boot loader.
3. Boot a Zephyr based application via the boot loader.

**Expected behavior**
Enable GPIOs of fixed regulators with `regulator-boot-on` (or `regulator-always-on` for that matter) are kept enabled during boot until they are disabled because they are not needed; there are no glitches on the enable GPIO.

One could argue that `regulator-boot-on` shouldn't even be necessary ("This property is intended to only be used for regulators where software cannot read the state of the regulator.") as the state of the enable GPIO can be read by software, but that's a separate topic.

**Impact**
Glitches can cause random malfunction. In my case the external oscillator stopped working (fortunately reproducibly), breaking half the system.

**Logs and console output**
Can be provided on request.

**Environment (please complete the following information):**
Should not have an impact on the bug.

**Additional context**
The DuT is an Arduino Portenta H7 board; see `boards/arm/arduino_portenta_h7/arduino_portenta_h7_m7.dts`. `regulator-fixed` is used because `fixed-clock` in Zephyr does not support enable GPIOs.

gmarull, 2023-04-27:
Thanks for reporting @silbe. Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/57320 fixes your problem?

silbe, 2023-04-27:
> Thanks for reporting @silbe. Can you check if #57320 fixes your problem?

It does. Thanks for the fast fix!

PR #57320 (2023-04-28): drivers: regulator: fixed: refactor init code
In some cases, the enable pin may be already enabled by a previous stage, e.g. bootloader. Therefore, it is not desirable to disable the pin, as it could cause malfunctioning of the device. Refactor init procedure so that we pick the right GPIO flags during the first configuration stage.

Fixes #57312
closed 2023-04-28 by carlescufi