zephyrproject-rtos/zephyr #57315: hci_usb device recognition fails on the nucleo_wb55rg board.

2songh, 2023-04-27:
**Is your feature request related to a problem? Please describe.**
Route: zephyr/samples/bluetooth/hci_usb
Board: nucleo_wb55rg
I built this sample program and flashed it to the board.

The device is recognized normally but is marked as unavailable.
- **This device cannot be started. (Code 10)**
- **This is an unsupported request.**

**Describe the solution you'd like**
I want to use it as a bluetooth dongle and test it.

**Additional context**
[image]
Is this nucleo_wb55rg board not supported?

nordicjm, 2023-04-27:
This is expected behaviour on windows as it does not support devices bluetooth LE only radios and requires classic to be at least supported.

2songh, 2023-04-27:
> This is expected behaviour on windows as it does not support devices bluetooth LE only radios and requires classic to be at least supported.

Are you saying that this function is not available for Windows?
Or does the board itself not support USB dongle function?

nordicjm, 2023-04-27:
A windows issue, try the device on linux. @jfischer-no can't seem to find the issue where this was discovered with bluetooth LE only devices not working on windows, do you remember where it was?

2songh, 2023-04-27:
> A windows issue, try the device on linux. @jfischer-no can't seem to find the issue where this was discovered with bluetooth LE only devices not working on windows, do you remember where it was?

I don't know about that.
Windows seems to have a lot of bad issues.

Any solution to this problem?
Is it only available on Linux?

2songh, 2023-04-27:
> Here it is: #23866

Thank you for answer.
It was a great help in understanding the problem.

nordicjm, 2023-04-28:
Would suggest using `bluetoothctl` from the command line instead to see if it works

2songh, 2023-04-28:
> `bluetoothctl`작동하는지 확인하기 위해 대신 명령줄에서 사용하는 것이 좋습니다.

[image]

I used 'bluetoothctl'.
The action seems to be working normally.