zephyrproject-rtos/zephyr #57323: RFC: introduce the STM32 system wakeup pins to exit low power modes

FRASTM, 2023-04-27:
## Introduction
The stm32 mcu has several external wakeup pins (GPIO) that are able to trigger a wakeup from Standby or Shutdown low power modes.
The WakeUp pins are fixed GPIO pins. 
Each WakeUp pin is programmed previous to entering a low power standby/shutdown and will exit the system on edge-detection.
The stm32 can identify the source of WakeUp.

### Problem description
Depending on the stm32 serie, configuration of the WakeUp pins differs:
- the number of wakeup pin
- GPIO pin  position, including a multiplexer 4 -->1
- the trigger polarity : rising/falling edge
- pull-up/down 
- the configuration (control) and status of each wakeUp pin is made through the Power (PWR) instance of each stm32.

### Proposed change
Adding a new devicetree binding for each wakeUp pin (of each stm32 device)
Adding a new devicetree binding for using the WakeUp pins
Adding a software sequence to enable and configure each WakeUp pin

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.

gdb3-ampaworks, 2024-02-09:
One unresolved question is easy to answer: Identifying the wakeup source from shutdown mode.

It doesn't seem possible -- the WKUP flag registers are reset when exiting shutdown mode. So they're essentially useless...

https://community.st.com/t5/stm32-mcus-products/stm32l476-wakeup-flag-not-set-when-exit-from-shutdown-after/m-p/309420

Let me know if this isn't correct. This is the behavior I see on STM32L4P5 anyways.

ajarmouni-st, 2024-02-09:
> One unresolved question is easy to answer: Identifying the wakeup source from shutdown mode.
>
> It doesn't seem possible -- the WKUP flag registers are reset when exiting shutdown mode. So they're [...]

Yes, does seem so. The best I was able to do was this:
https://github.com/zephyrproject-rtos/zephyr/blob/56720088cd57d18e828f98fb22a59f458a0375da/samples/boards/stm32/power_mgmt/wkup_pins_gpio/src/main.c#L39-L43

closed 2024-06-05 by dleach02