zephyrproject-rtos/zephyr #57324: tests: subsys: settings: file: system.settings.file.raw fails

katgiadla, 2023-04-27:
**Describe the bug**
The tests/subsys/fs/littlefs/filesystem.littlefs.custom and tests/subsys/fs/littlefs/filesystem.littlefs.default fails.

Observed for nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:

1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/settings/file -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-04-26T23:12:03.573Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-3164-ga0ad7b7752e6 ***[0m
[2023-04-26T23:12:03.573Z] [0mRunning TESTSUITE settings_config[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_commit[0m
[2023-04-26T23:12:03.573Z] [0mPASS - test_config_commit in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_empty_lookups[0m
[2023-04-26T23:12:03.573Z] [0mPASS - test_config_empty_lookups in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_getset_int[0m
[2023-04-26T23:12:03.573Z] [0mPASS - test_config_getset_int in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_getset_int64[0m
[2023-04-26T23:12:03.573Z] [0mPASS - test_config_getset_int64 in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_getset_unknown[0m
[2023-04-26T23:12:03.573Z] [0mPASS - test_config_getset_unknown in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mTESTSUITE settings_config succeeded[0m
[2023-04-26T23:12:03.573Z] [0mRunning TESTSUITE settings_config_fs[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mAssumption failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/settings/file/src/settings_setup_littlefs.c:33: config_setup_fs: (rc == 0 is false)[0m
[2023-04-26T23:12:03.573Z] [0merasing flash area [-5][0m
[2023-04-26T23:12:03.573Z] [0mI: LittleFS version 2.5, disk version 2.0[0m
[2023-04-26T23:12:03.573Z] [0mI: FS at mx25r6435f@0:0x0 is 16 0x1000-byte blocks with 512 cycle[0m
[2023-04-26T23:12:03.573Z] [0mI: sizes: rd 16 ; pr 16 ; ca 64 ; la 32[0m
[2023-04-26T23:12:03.573Z] [0mE: ../../../../../../../../../zephyr/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}[0m
[2023-04-26T23:12:03.573Z] [0mW: can't mount (LFS -84); formatting[0m
[2023-04-26T23:12:03.573Z] [0mI: /fs mounted[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_compress_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_compress_file in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_empty_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_empty_file in 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_multiple_in_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_multiple_in_file in 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_save_in_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_save_in_file in 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_save_one_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_save_one_file in 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mSTART - test_config_small_file[0m
[2023-04-26T23:12:03.573Z] [0mSKIP - test_config_small_file in 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mTESTSUITE settings_config_fs succeeded[0m
[2023-04-26T23:12:03.573Z] [0m------ TESTSUITE SUMMARY START ------[0m
[2023-04-26T23:12:03.573Z] [0mSUITE PASS - 100.00% [settings_config]: pass = 5, fail = 0, skip = 0, total = 5 duration = 0.005 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- PASS - [settings_config.test_config_commit] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- PASS - [settings_config.test_config_empty_lookups] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- PASS - [settings_config.test_config_getset_int] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- PASS - [settings_config.test_config_getset_int64] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- PASS - [settings_config.test_config_getset_unknown] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0mSUITE SKIP -   0.00% [settings_config_fs]: pass = 0, fail = 0, skip = 6, total = 6 duration = 0.002 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_compress_file] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_empty_file] duration = 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_multiple_in_file] duration = 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_save_in_file] duration = 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_save_one_file] duration = 0.000 seconds[0m
[2023-04-26T23:12:03.573Z] [0m- SKIP - [settings_config_fs.test_config_small_file] duration = 0.001 seconds[0m
[2023-04-26T23:12:03.573Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-04-26T23:12:03.573Z] [0m===================================================================[0m
[2023-04-26T23:12:03.573Z] [0mRunID: d1bfa857dc700f5eb8791bafafc4a26f[0m
[2023-04-26T23:12:03.573Z] [0mPROJECT EXECUTION FAILED[0m
[2023-04-26T23:12:03.573Z] [0m[0m
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.15.2
- Commit SHA or Version used: zephyr-v3.3.0-3164-ga0ad7b7752e6

katgiadla, 2023-06-12:
Tests are executed on `Zephyr-sdk 0.16.1`, on `nrf52840dk_nrf52840`  and this issue still occur. Below logs from tests:

```
2023-06-12 10:26:25,354 - twister - ERROR - *** Booting Zephyr OS build v3.4.0-rc3-1-ga17fc4ff2b71 ***
Running TESTSUITE settings_config
===================================================================
START - test_config_commit
PASS - test_config_c
TESTSUITE settings_config succeeded
Running TESTSUITE settings_config_fs
===================================================================
Assumption failed at WEST_TOPDIR/zephyr/tests/subsys/settings/file/src/settings_setup_littlefs.c:33: config_setup_fs: (rc == 0 is false)
erasing flash area [-5]
I: LittleFS verk ersion 2.0
I: FS at mx25r6435f@0:0x0 is 16 0x1000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
I: /fs mounted
START - test_config_compress_file
SKIP - test_config_compress_file in 0.001 seconds
===================================================================
START - test_config_empty_file
SKIP - test_config_empty_file in 0.000 seconds
===================================================================
START - test_config_multiple_in_file
SKIP - test_config_monds
===================================================================
START - test_config_save_in_file
SKIP - test_config_save_in_file in 0.000 s============
START - test_config_save_one_file
SKIP - test_config_save_one_file in 0.000 seconds
===================================================================
START - test_config_small_file
SKIP - test_config_small_file in 0.000 seconds
===================================================================
TESTSUITE settings_config_fs succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [settings_config]: pass = 5, fail = 0, skip = 0, total = 5 duration = 0.005 seconds
- PASS - [settings_config.test_config_commit] duration = 0.001 seconds
- PASS - [settings_config.test_config_empty_lookups] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_int64] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_int] duration = 0.001 seconds
- PASS - [settings_config.test_config_getset_unknown] duration = 0.001 seconds
SUITE SKIP -   0.00% [settings_config_fs]: pass = 0, fail = 0, skip = 6, total = 6 duration = 0.001 seconds
- SKIP - [settings_config_fs.test_config_compress_file] duration = 0.001 seconds
- SKIP - [settings_config_fs.test_config_empty_file] duration = 0.000 seconds
- SKIP - [settings_config_fs.test_config_multiple_ion = 0.000 seconds
- SKIP - [settings_config_fs.test_config_save_in_file] duration = 0.000 seconds
- SKIP - [settings_config_fs.test_config_save_one_file] duration = 0.000 seconds
- SKIP - [settings_config_fs.test_config_small_file] duration = 0.000 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: c6237b8178fe8fd81439d03d28c47759
PROJECT EXECUTION FAILED
```

katgiadla, 2023-07-20:
Problem still occurs.

jhedberg, 2023-09-26:
@katgiadla is the issue still there with the latest tree? @de-nordic are you still not able to reproduce this?

de-nordic, 2023-09-26:
I have taken look at the problem and it seems that this is not with littlefs, it just manifest here.
The problem is with write lock being enabled to quickly after QSPI NOR erase command is issued.

katgiadla, 2023-10-16:
Issue still occurs. Test scenario is renamed as `settings.file.raw`.

PR #67383 (2024-02-05): tests/settings/file: increase main stack size
Increase ZTEST and MAIN stack to 4k.

Fixes #57324
commit 5f31c15b (2024-02-05): tests/settings/file: increase main stack size

Increase ZTEST and MAIN stack to 4k.

Fixes #57324
closed 2024-02-05 by henrikbrixandersen