zephyrproject-rtos/zephyr #57325: SPI2 DMA configuration isses STM32WB55

pebeoo, 2023-04-27:
**Describe the bug**
In our project we are using DMA2 to handle SPI2 transfers. With following configuration transmission works:
```
&spi2 {
    dmas = <&dmamux1 10 9 0x00440
            &dmamux1 11 8 0x00480>;
    dma-names = "tx", "rx";
};
```

but when we want to use dma2 notation, transmission does not work properly:
```
&spi2 {
  dmas = <&dma2 4 9 0x00440
          &dma2 5 8 0x00480>;
    dma-names = "tx", "rx";
};
```

**To Reproduce**
Set SPI dmas configuration to dma2 channel using dma notation.

**Expected behavior**
Possibility to setup SPI on DMA2 with dma notation in dts files.

**Environment (please complete the following information):**
OS: Linux
Toolchain Zephyr SDK
Commit SHA or Version used: 3.3.99

FRASTM, 2023-04-27:
Because the stm32wb has a **DMAMUX** in front of DMA1 and DMA2, you cannot access the DMA directly.
but through DMAMUX channels: 
-  DMAMUX channels 0 to 6 are connected to DMA1 channels 1 to 7
-  DMAMUX channels 7 to 13 are connected to DMA2 channels 1 to 7
--> DMA1, DMA2, DMAMUX have all 	`{ status = "okay"; }`

With the dts 
```
&spi2 {
    dmas = <&dmamux1 10 9 0x00440
            &dmamux1 11 8 0x00480>;
    dma-names = "tx", "rx";
};
```

channels 10 and 11 are on the DMA2

pebeoo, 2023-04-27:
Thank you for reply, but I am little bit confused. In other thread there was suggestion to use access by &dma1 and &dma2:
https://github.com/zephyrproject-rtos/zephyr/issues/55754#issuecomment-1468128986. So I assumed that this approach is valid for all drivers.

pebeoo, 2023-04-28:
Thank you for the clarification. In this case issue could be closed.

closed 2023-04-28 by FRASTM