zephyrproject-rtos/zephyr #57326: tests: kernel: timer: timer_behavior: kernel.timer.timer fails

katgiadla, 2023-04-27:
**Describe the bug**
The kernel.timer.timer fails. For board `nrf5340dk_nrf5340_cpuapp` fails and for rest of below, it behaves unstable.

Observed for:
* nrf5340dk_nrf5340_cpuapp
* nrf5340dk_nrf5340_cpuapp_ns
* nrf9160dk_nrf9160
* nrf5340dk_nrf5340_cpunet
* nrf52dk_nrf52832
* nrf9160dk_nrf9160_ns

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-3164-ga0ad7b7752e6 ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 10 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 8192
expected period drift: 98.632812 us
period duration statistics for 10000 samples (0 rollovers):
expected: 1000 us,       32.768000 cycles
min:      1098.632812 us, 36 cycles
max:      1220.703125 us, 40 cycles
mean:     1098.645020 us, 36.000400 cycles
variance: 1.489967 us, 0.001600 cycles
stddev:   1.220642 us, 0.039998 cycles
timer start cycle 9169, end cycle 369173,
total time 10986450.195312 us, expected time 10000000.000000 us,
expected time drift 986328.125000 us, difference 122.070313 us
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:242: do_test_using: (max_us <= max_us_bound is false)
Longest timer period too long (off by more than expected 10%)
FAIL - test_jitter_drift_timer_period in 11.196 seconds
===================================================================
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 10 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 8192
expected period drift: 98.632812 us
period duration statistics for 10000 samples (0 rollovers):
expected: 1000 us,       32.768000 cycles
min:      1098.632812 us, 36 cycles
max:      1098.632812 us, 36 cycles
mean:     1098.632812 us, 36.000000 cycles
variance: 0.000000 us, 0.000000 cycles
stddev:   0.000000 us, 0.000000 cycles
timer start cycle 376532, end cycle 736532,
total time 10986328.125000 us, expected time 10000000.000000 us,
expected time drift 986328.125000 us, difference 0.000000 us
PASS - test_jitter_drift_timer_startdelay in 11.134 seconds
===================================================================
TESTSUITE timer_jitter_drift failed.
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 8192Hz, Expecting 20480 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 0
Faking busy work, remaining timers is 4, timer callbacks 689
Faking busy work, remaining timers is 4, timer callbacks 1377
Faking busy work, remaining timers is 4, timer callbacks 2066
Faking busy work, remaining timers is 4, timer callbacks 2755
Faking busy work, remaining timers is 4, timer callbacks 3444
Faking busy work, remaining timers is 4, timer callbacks 4133
Faking busy work, remaining timers is 4, timer callbacks 4822
Faking busy work, remaining timers is 4, timer callbacks 5511
Faking busy work, remaining timers is 4, timer callbacks 6200
Faking busy work, remaining timers is 4, timer callbacks 6889
Faking busy work, remaining timers is 4, timer callbacks 7578
Faking busy work, remaining timers is 4, timer callbacks 8268
Faking busy work, remaining timers is 4, timer callbacks 8957
Faking busy work, remaining timers is 4, timer callbacks 9646
Faking busy work, remaining timers is 4, timer callbacks 10335
Faking busy work, remaining timers is 4, timer callbacks 11024
Faking busy work, remaining timers is 4, timer callbacks 11713
Faking busy work, remaining timers is 4, timer callbacks 12402
Faking busy work, remaining timers is 4, timer callbacks 13092
Faking busy work, remaining timers is 4, timer callbacks 13781
Faking busy work, remaining timers is 4, timer callbacks 14470
Faking busy work, remaining timers is 4, timer callbacks 15160
Faking busy work, remaining timers is 4, timer callbacks 15849
Faking busy work, remaining timers is 4, timer callbacks 16539
Faking busy work, remaining timers is 4, timer callbacks 17228
Faking busy work, remaining timers is 4, timer callbacks 17918
Faking busy work, remaining timers is 4, timer callbacks 18607
Faking busy work, remaining timers is 4, timer callbacks 19295
Faking busy work, remaining timers is 4, timer callbacks 19981
One Tick Timer Train Done, took 10070 ms, busy loop ran 30 times
Perfect delta 16 cycles or 488 us
Timer 0 max delta 16 cycles or 488 us, 0 late callbacks (0.0%)
Timer 1 max delta 16 cycles or 488 us, 0 late callbacks (0.0%)
Timer 2 max delta 16 cycles or 488 us, 0 late callbacks (0.0%)
Timer 3 max delta 16 cycles or 488 us, 0 late callbacks (0.0%)
!! The busy loop didn't run as much as expected.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
PASS - test_one_tick_timer_train in 10.122 seconds
===================================================================
TESTSUITE timer_tick_train succeeded
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  50.00% [timer_jitter_drift]: pass = 1, fail = 1, skip = 0, total = 2 duration = 22.330 seconds
- FAIL - [timer_jitter_drift.test_jitter_drift_timer_period] duration = 11.196 seconds
- PASS - [timer_jitter_drift.test_jitter_drift_timer_startdelay] duration = 11.134 seconds
SUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.122 seconds
- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.122 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: de60ff5cd37853b5a87a7f000902655e
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
* OS: Ubuntu 20.04.1 LTS
* Toolchain Zephyr SDK 0.15.2
* Commit SHA or Version used: zephyr-v3.3.0-3164-ga0ad7b7752e6

nordicjm, 2023-04-27:
Did you run this with the DK UART workaround Kconfig disabled?

FRASTM, 2023-04-27:
Similar issue found with stm32 boards : 
https://github.com/zephyrproject-rtos/zephyr/issues/57285
https://github.com/zephyrproject-rtos/zephyr/issues/57288

jgl-meta, 2023-05-02:
@teburd

katgiadla, 2023-05-04:
Similar problem occurred on `nrf52840dk_nrf52840`:

```
[2023-05-02T20:22:56.258Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-3341-gef3d01d16fd4 ***[0m
[2023-05-02T20:22:56.258Z] [0mRunning TESTSUITE timer_jitter_drift[0m
[2023-05-02T20:22:56.258Z] [0m===================================================================[0m
[2023-05-02T20:22:56.258Z] [0mSTART - test_jitter_drift_timer_period[0m
[2023-05-02T20:22:56.258Z] [0mperiodic timer behavior test using built-in restart mechanism[0m
[2023-05-02T20:22:56.258Z] [0mcollecting time samples for approx 10 seconds[0m
[2023-05-02T20:22:56.258Z] [0mperiodic timer samples gathered, calculating statistics[0m
[2023-05-02T20:22:56.258Z] [0mtimer clock rate 32768, kernel tick rate 8192[0m
[2023-05-02T20:22:56.258Z] [0mexpected period drift: 98.632812 us[0m
[2023-05-02T20:22:56.258Z] [0mperiod duration statistics for 10000 samples (0 rollovers):[0m
[2023-05-02T20:22:56.258Z] [0mexpected: 1000 us,       	32.768000 cycles[0m
[2023-05-02T20:22:56.258Z] [0mmin:      1098.632812 us, 	36 cycles[0m
[2023-05-02T20:22:56.258Z] [0mmax:      1098.632812 us, 	36 cycles[0m
[2023-05-02T20:22:56.258Z] [0mmean:     1098.632812 us, 	36.000000 cycles[0m
[2023-05-02T20:22:56.258Z] [0mvariance: 0.000000 us, 	0.000000 cycles[0m
[2023-05-02T20:22:56.258Z] [0mstddev:   0.000000 us, 	0.000000 cycles[0m
[2023-05-02T20:22:56.258Z] [0mtimer start cycle 13701, end cycle 373701,[0m
[2023-05-02T20:22:56.258Z] [0mtotal time 10986328.125000 us, expected time 10000000.000000 us,[0m
[2023-05-02T20:22:56.258Z] [0mexpected time drift 986328.125000 us, difference 0.000000 us[0m
[2023-05-02T20:22:56.258Z] [0mPASS - test_jitter_drift_timer_period in 11.135 seconds[0m
[2023-05-02T20:22:56.258Z] [0m===================================================================[0m
[2023-05-02T20:22:56.258Z] [0mSTART - test_jitter_drift_timer_startdelay[0m
[2023-05-02T20:22:56.258Z] [0mperiodic timer behavior test using explicit start with delay[0m
[2023-05-02T20:22:56.258Z] [0mcollecting time samples for approx 10 seconds[0m
[2023-05-02T20:22:56.258Z] [0mperiodic timer samples gathered, calculating statistics[0m
[2023-05-02T20:22:56.258Z] [0mtimer clock rate 32768, kernel tick rate 8192[0m
[2023-05-02T20:22:56.258Z] [0mexpected period drift: 98.632812 us[0m
[2023-05-02T20:22:56.258Z] [0mperiod duration statistics for 10000 samples (0 rollovers):[0m
[2023-05-02T20:22:56.258Z] [0mexpected: 1000 us,       	32.768000 cycles[0m
[2023-05-02T20:22:56.258Z] [0mmin:      1098.632812 us, 	36 cycles[0m
[2023-05-02T20:22:56.259Z] [0mmax:      1220.703125 us, 	40 cycles[0m
[2023-05-02T20:22:56.259Z] [0mmean:     1098.645020 us, 	36.000400 cycles[0m
[2023-05-02T20:22:56.259Z] [0mvariance: 1.489967 us, 	0.001600 cycles[0m
[2023-05-02T20:22:56.259Z] [0mstddev:   1.220642 us, 	0.039998 cycles[0m
[2023-05-02T20:22:56.259Z] [0mtimer start cycle 379048, end cycle 739052,[0m
[2023-05-02T20:22:56.259Z] [0mtotal time 10986450.195312 us, expected time 10000000.000000 us,[0m
[2023-05-02T20:22:56.259Z] [0mexpected time drift 986328.125000 us, difference 122.070313 us[0m
[2023-05-02T20:22:56.259Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:242: do_test_using: (max_us <= max_us_bound is false)[0m
[2023-05-02T20:22:56.259Z] [0mLongest timer period too long (off by more than expected 10%)[0m
[2023-05-02T20:22:56.259Z] [0mFAIL - test_jitter_drift_timer_startdelay in 11.195 seconds[0m
[2023-05-02T20:22:56.259Z] [0m===================================================================[0m
[2023-05-02T20:22:56.259Z] [0mTESTSUITE timer_jitter_drift failed.[0m
[2023-05-02T20:22:56.259Z] [0mRunning TESTSUITE timer_tick_train[0m
[2023-05-02T20:22:56.259Z] [0m===================================================================[0m
[2023-05-02T20:22:56.259Z] [0mSTART - test_one_tick_timer_train[0m
[2023-05-02T20:22:56.259Z] [0mInitializing 4 Timers, Tick Rate 8192Hz, Expecting 20480 callbacks in 11000 ms[0m
[2023-05-02T20:22:56.259Z] [0mStarting Timers with Skews[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 0[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 645[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 1290[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 1935[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 2580[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 3225[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 3870[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 4515[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 5160[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 5805[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 6451[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 7096[0m
[2023-05-02T20:22:56.259Z] [0mFaking busy work, remaining timers is 4, timer callbacks 7741[0m
[... cut ...]
[2023-05-02T20:22:56.259Z] [0mSUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.110 seconds[0m
[2023-05-02T20:22:56.259Z] [0m- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.110 seconds[0m
[2023-05-02T20:22:56.259Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-05-02T20:22:56.259Z] [0m===================================================================[0m
[2023-05-02T20:22:56.259Z] [0mRunID: dc7ae3d26a811fd14ec77379f98db2da[0m
[2023-05-02T20:22:56.259Z] [0mPROJECT EXECUTION FAILED[0m
[2023-05-02T20:22:56.259Z] [0m[0m
```

katgiadla, 2023-05-05:
Issue occured on `nrf9160dk_nrf9160_ns`:

```
[2023-05-05T03:17:26.868Z] [0mERROR   -  *** Booting Zephyr OS build v3.2.99-ncs2-5067-gb54c8baeeafb ***[0m
[2023-05-05T03:17:26.868Z] [0mRunning TESTSUITE timer_jitter_drift[0m
[2023-05-05T03:17:26.868Z] [0m===================================================================[0m
[2023-05-05T03:17:26.868Z] [0mSTART - test_jitter_drift[0m
[2023-05-05T03:17:26.868Z] [0mperiodic timer behavior test (approx 10 seconds)[0m
[2023-05-05T03:17:26.868Z] [0mperiodic timer samples gathered, calculating statistics[0m
[2023-05-05T03:17:26.868Z] [0mtimer clock rate 32768, kernel tick rate 8192[0m
[2023-05-05T03:17:26.868Z] [0mexpected period drift: 98.632812 us[0m
[2023-05-05T03:17:26.868Z] [0mperiod duration statistics for 10000 samples (0 rollovers):[0m
[2023-05-05T03:17:26.868Z] [0mexpected: 1000 us,       	32.768000 cycles[0m
[2023-05-05T03:17:26.868Z] [0mmin:      1098.632812 us, 	36 cycles[0m
[2023-05-05T03:17:26.868Z] [0mmax:      1220.703125 us, 	40 cycles[0m
[2023-05-05T03:17:26.868Z] [0mmean:     1098.645020 us, 	36.000400 cycles[0m
[2023-05-05T03:17:26.868Z] [0mvariance: 1.489967 us, 	0.001600 cycles[0m
[2023-05-05T03:17:26.868Z] [0mstddev:   1.220642 us, 	0.039998 cycles[0m
[2023-05-05T03:17:26.868Z] [0mtimer start cycle 16329, end cycle 376333,[0m
[2023-05-05T03:17:26.868Z] [0mtotal time 10986450.195312 us, expected time 10000000.000000 us,[0m
[2023-05-05T03:17:26.868Z] [0mexpected time drift 986328.125000 us, difference 122.070313 us[0m
[2023-05-05T03:17:26.868Z] [0mAssertion failed at ../../../../../../../../../ncs/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:199: timer_jitter_drift_test_jitter_drift: (max_us <= max_us_bound is false)[0m
[2023-05-05T03:17:26.868Z] [0mLongest timer period too long (off by more than expected 10%)[0m
[2023-05-05T03:17:26.868Z] [0mFAIL - test_jitter_drift in 11.212 seconds[0m
[2023-05-05T03:17:26.868Z] [0m===================================================================[0m
[2023-05-05T03:17:26.868Z] [0mTESTSUITE timer_jitter_drift failed.[0m
[2023-05-05T03:17:26.868Z] [0mRunning TESTSUITE timer_tick_train[0m
[2023-05-05T03:17:26.868Z] [0m===================================================================[0m
[2023-05-05T03:17:26.868Z] [0mSTART - test_one_tick_timer_train[0m
[2023-05-05T03:17:26.868Z] [0mInitializing 4 Timers, Tick Rate 8192Hz, Expecting 20480 callbacks in 11000 ms[0m
[2023-05-05T03:17:26.868Z] [0mStarting Timers with Skews[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 0[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 676[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 1352[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 2028[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 2704[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 3380[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 4056[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 4731[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 5407[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 6083[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 6759[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 7435[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 8111[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 8787[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 9463[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 10139[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 10815[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 11492[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 12168[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 12844[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 13521[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 14197[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 14873[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 15549[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 16225[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 16901[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 17577[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 18253[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 18930[0m
[2023-05-05T03:17:26.868Z] [0mFaking busy work, remaining timers is 4, timer callbacks 19606[0m
[... cut ...]
[2023-05-05T03:17:26.868Z] [0mSUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.234 seconds[0m
[2023-05-05T03:17:26.868Z] [0m- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.234 seconds[0m
[2023-05-05T03:17:26.868Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-05-05T03:17:26.868Z] [0m===================================================================[0m
[2023-05-05T03:17:26.868Z] [0mRunID: b92dfa528cdfa77343677a4b39cbcfb6[0m
[2023-05-05T03:17:26.868Z] [0mPROJECT EXECUTION FAILED[0m
[2023-05-05T03:17:26.868Z] [0m[0m
```

PR #59009 (2023-06-12): tests: timer_behavior: Use bigger drift tolerance for nRF RTC timer
Use 13% instead of the default 10% when the nRF RTC timer is used so that the allowed drift is at least one tick long (~122 us in this case).

Fixes #57326.
closed 2023-06-12 by nashif