zephyrproject-rtos/zephyr #57327: tests: drivers: flash: common: drivers.flash.common.default fails

katgiadla, 2023-04-27:
**Describe the bug**
The `drivers.flash.common.default` fails.

Observed for:
* nrf5340dk_nrf5340_cpuapp

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340_cpuapp connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/drivers/flash/common -p nrf5340dk_nrf5340_cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-04-26T21:47:47.954Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-3164-ga0ad7b7752e6 ***[0m
[2023-04-26T21:47:47.954Z] [0mRunning TESTSUITE flash_driver[0m
[2023-04-26T21:47:47.954Z] [0m===================================================================[0m
[2023-04-26T21:47:47.954Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/flash/common/src/main.c:101: flash_driver_setup: (rc not equal to 0)[0m
[2023-04-26T21:47:47.954Z] [0mFlash memory not properly erased[0m
[2023-04-26T21:47:47.954Z] [0mSTART - test_read_unaligned_address[0m
[2023-04-26T21:47:47.954Z] [0mFAIL - test_read_unaligned_address in 0.001 seconds[0m
[2023-04-26T21:47:47.954Z] [0m===================================================================[0m
[2023-04-26T21:47:47.954Z] [0mTESTSUITE flash_driver failed.[0m
[2023-04-26T21:47:47.954Z] [0m------ TESTSUITE SUMMARY START ------[0m
[2023-04-26T21:47:47.954Z] [0mSUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.001 seconds[0m
[2023-04-26T21:47:47.954Z] [0m- FAIL - [flash_driver.test_read_unaligned_address] duration = 0.001 seconds[0m
[2023-04-26T21:47:47.954Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-04-26T21:47:47.954Z] [0m===================================================================[0m
[2023-04-26T21:47:47.954Z] [0mRunID: d4ced2688b001370fec3c648b65075d3[0m
[2023-04-26T21:47:47.954Z] [0mPROJECT EXECUTION FAILED[0m
[2023-04-26T21:47:47.954Z] [0m[0m
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.15.2
- Commit SHA or Version used: zephyr-v3.3.0-3164-ga0ad7b7752e6

katgiadla, 2023-05-04:
The same problem occured for `drivers.flash.common.soc` on `nrf52840dk_nrf52840`:

```
[2023-05-02T21:12:33.637Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-3341-gef3d01d16fd4 ***[0m
[2023-05-02T21:12:33.637Z] [0mRunning TESTSUITE flash_driver[0m
[2023-05-02T21:12:33.637Z] [0m===================================================================[0m
[2023-05-02T21:12:33.637Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/drivers/flash/common/src/main.c:101: flash_driver_setup: (rc not equal to 0)[0m
[2023-05-02T21:12:33.637Z] [0mFlash memory not properly erased[0m
[2023-05-02T21:12:33.637Z] [0mSTART - test_read_unaligned_address[0m
[2023-05-02T21:12:33.637Z] [0mFAIL - test_read_unaligned_address in 0.000 seconds[0m
[2023-05-02T21:12:33.637Z] [0m===================================================================[0m
[2023-05-02T21:12:33.637Z] [0mTESTSUITE flash_driver failed.[0m
[2023-05-02T21:12:33.637Z] [0m------ TESTSUITE SUMMARY START ------[0m
[2023-05-02T21:12:33.637Z] [0mSUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.000 seconds[0m
[2023-05-02T21:12:33.637Z] [0m- FAIL - [flash_driver.test_read_unaligned_address] duration = 0.000 seconds[0m
[2023-05-02T21:12:33.637Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-05-02T21:12:33.637Z] [0m===================================================================[0m
[2023-05-02T21:12:33.637Z] [0mRunID: 5797357d5652efede69458caa06ab62e[0m
[2023-05-02T21:12:33.637Z] [0mPROJECT EXECUTION FAILED[0m
[2023-05-02T21:12:33.637Z] [0m[0m
```

de-nordic, 2023-06-07:
I have invoked
```
./scripts/twister -T tests/drivers/flash/common -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
```
on Zepyr @ b6d8989236147cb32da09f802b61aef70fb31b78 , with sdk 0.16.1 and the problem does not seem to occur.

katgiadla, 2023-06-09:
Issue is fixed.

closed 2023-06-09 by katgiadla