zephyrproject-rtos/zephyr #57356: testcase.yaml - twister filtering of tests based on DTS address sizing / DTS node prop values

aaronemassey, 2023-04-27:
**Is your enhancement proposal related to a problem? Please describe.**

Currently we have to special case testing various platforms based on dts address sizing. https://github.com/zephyrproject-rtos/zephyr/pull/55103#issuecomment-1449050118 shows the clear example. This is specifically needed when we attempt to validate a test against all native_posix/qemu platforms. Another example https://github.com/zephyrproject-rtos/zephyr/pull/56666#discussion_r1179505773.

**Describe the solution you'd like**
In a `testcase.yaml` we should be able to filter based on device-tree properties value. This would allow us to directly case by dts bus address sizing.

**Describe alternatives you've considered**
Status Quo:
Pro: No work to be done
Con: We keep having to special-case tests causing a lot of boiler-plate which may have to be updated with new qemu.

aaronemassey, 2023-04-27:
I'll take a quick look at it myself, but I'm not as familiar with twisters code.

galak, 2023-04-28:
Can you give an example of what the filter would kinda look like in the testcase.yaml?

aaronemassey, 2023-05-02:
> Can you give an example of what the filter would kinda look like in the testcase.yaml?

I'd like to make a filter like so:

```
tests:
  # section.subsection
  drivers.sbs_gauge_new_api.emulated:
    tags: drivers fuel_gauge
    filter: >
      dt_node_path_prop("/soc", "#address-cells") > and
      dt_compat_enabled("sbs,sbs-gauge-new-api") and
      (CONFIG_QEMU_TARGET or CONFIG_BOARD_NATIVE_POSIX)
```

I'm willing to iterate on it a bit myself but wanted to motivate the actual work/PR before doing so. Or if someone who could do it in 5 minutes wanted to do it instead it probably taking me closer to an hour or two as I've never made a change to `kconfigfunctions.py`

Let me know if you want me to give it a try first or if you think what I'm asking for is mistaken.