zephyrproject-rtos/zephyr #57366: sysbuild: Value propagation (e.g. mcuboot) is not working

nordicjm, 2023-04-28:
**Describe the bug**
It seems that value propagation for sysbuild is not working, for example if a project is configured then the sysbuild configuration is changed to have mcuboot use an ecdsa-p256 key, mcuboot is not configured for this. The `mcuboot_sysbuild_cache.txt` file then shows conflicting and wrong information:
```
mcuboot_CONFIG_BOOT_SIGNATURE_KEY_FILE:STRING="/tmp/aa/bootloader/mcuboot/root-rsa-2048.pem"
mcuboot_CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256:STRING=y
mcuboot_CONFIG_BOOT_SIGNATURE_TYPE_RSA:STRING=y
```

**To Reproduce**
Configure application with sysbuild and mcuboot, then change key type in sysbuild configuration

**Expected behavior**
Values to be correct

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: f12ffaea1a9105d0cff74ea50ecde0549c680740

tejlmand, 2023-04-28:
I believe this will be solved by one of those:
https://github.com/zephyrproject-rtos/zephyr/pull/55871
https://github.com/zephyrproject-rtos/zephyr/pull/55616

so we need to decide the approach to use.

tejlmand, 2023-04-28:
actually, those settings are sysbuild controlled, and user should adjust them using sysbuild config.
In such case sysbuild is the expected to pass the correct setting to image builds.

tejlmand, 2023-04-28:
this is already covered in this enhancement issue as part of work going forward: https://github.com/zephyrproject-rtos/zephyr/issues/53126

nordicjm, 2023-06-08:
@tejlmand Not resolved on main, configuring a project when changing mcuboot to work in ecdsa-p256 mode using `sysbuild_menuconfig` does not propagate value to mcuboot and it remains in RSA mode, though the key filename is updated successfully.

PR #59050 (2023-06-08): sysbuild: Fix value propagation of signature type to MCUboot
Fixes an issue whereby the type of the signature was not passed to MCUboot.

Fixes #57366
closed 2023-06-08 by nashif