zephyrproject-rtos/zephyr #57373: Zephyr instrumentation subsystem

gromero, 2023-04-28:
## TL:DR;

New Zephyr subsystem for profiling and tracing, proof of concept: https://github.com/zephyrproject-rtos/zephyr/pull/57371

CLI tool demo:
https://asciinema.org/a/581092

Perfetto output example:

[image]

## Introduction

This RFC proposes a new Zephyr subsystem named `instrumentation` meant for
profiling and tracing. The new 'instrumentation' subsys leverages the compiler
code instrumentation feature available on compilers like GCC and LLVM, enabled
via the flag `-finstrument-functions`, to collect profile data, function
entry/exit traces, and thread scheduling info in a non-intrusive way.

### Problem description

Most profiling and tracing features in Zephyr today either depend on proprietary
and expensive tooling, or have significant platform limitations. For instance,
Segger's SystemView, which requires J-Link, or gprof, which only supports a few
platforms, often emulated, like native_posix, and native_posix_64 platforms,
with no timestamp data associated with events. Also, although `tracing` subsys
has numerous hooks available in its APIs, these hooks are static and so can't be
easily set to trace arbitrary functions in kernel or application code.

By leveraging the compiler code instrumentation feature it is possible to have a
platform, tooling, and arch-neutral tracing and profiling subsystem which allows
collecting traces at arbitrary code locations, both at compile time and runtime,
and at the same time, using the very same mechanism, collect profiling data for
the traced code.

Once the traces and profiling data are collected, various transport layers can
be used to transfer data from the target to the host, similary to the transport
backends available in the `tracing` subsys (UART, USB, etc).

Data can then the viewed in the host using a CLI tool, which can display traces
and profing statistics in the text terminal or export it to various formats so
data can be visualized on GUI tools, like Perfetto TraceViewer. The same CLI
tool allows for setting the functions to be traced at runtime (it can also be
set via Kconfig, at compile time).

### Proposed change

Implement a new Zephyr subsystem named `instrumentation` and an associated CLI
tool that allows setting, collecting, displaying, and exporting tracing and
profiling data captured by this new subsystem.

## Detailed RFC

Compilers like GCC and LLVM can generate instrumentation calls for entry and
exit to functions if flag `-finstrument-functions` is passed to the compiler.
When this feature is enabled, just after function entry and just before function
exit, the following functions, respectively, are called with the address of the
current function and its call site:

```
void __cyg_profile_func_enter(void *callee, void *call_site);
void __cyg_profile_func_exit(void *callee, void *call_site);
```

The instrumentation is also done for functions expanded inline in other
functions. A function may be given the attribute `no_instrument_function`, in
which case this instrumentation is not done for the function. It's also possible
to exclude from instrumentation all functions in a compilation unit (source
file) by setting `-finstrument-functions-exclude-file-list=file,file,...`, or
also exclude functions by their name setting
[... 279 lines cut ...]

- Calculating delta in profiling for recursive calls can easily lead to wrong
  delta values because the number of recursive calls exceeds the number of slots
  in the timestamp stack (usually 16 slots). Maybe some heuristics can be used
  to avoid pushing further timestamps in recursion.
- If the number of functions profiled exceeds the max number of functions for
  which delta can be accumulated, some kind of heuristic can be used to discard
  the least relevant ones, for instance, function with low delta values.
- About 50% (20 bytes) of the event record size is spend in the thread name. It
  can be compacted further by using a lookup table and just recording the lookup
  index.
- Due to limitations in the CTF 1.8 a variant id can't be used to select a field
  type, so it's difficult to represent 32-bit vs 64-bit binary formats in a
  single `metadata` file. The same happens for expressing big-endian vs
  little-endian formats. Hence multiple `metadata` files would exist in CTF 1.8.
  This problem is solved in CTF 2.0, but currently there isn't a library
  implementation for it (it's a WIP).
- `trigger` and `stopper` mechanism can be enhanced, for instance, be activated
   only a certain amount of times and when callee is called from specific
   callers.
- It's possible to enhance the tracing by providing additional statistics. For
  example: if the trace buffer isn't enough to capture all the events for the
  location selected, print how much memory it would be necessary to capture all
  the events ideally, so the user can adjust the buffer size based on this
  estimation.

cfriedt, 2023-05-02:
> Also, although tracing subsys has numerous hooks available in its APIs, these hooks are static and so can't be
easily set to trace arbitrary functions in kernel or application code.

💯
Very much support adding this upstream.  At Meta we took a very similar approach, but never upstreamed it, and that has always been a bit of a regret. I'll see if I can get some interested parties to comment here. 

> Once the traces and profiling data are collected, various transport layers can be used to transfer data from the target to the host, similary to the transport backends available in the tracing subsys (UART, USB, etc).

Perfect - this is exactly what I was looking for!

Thanks for putting this together @gromero and @microbuilder

vlotech, 2023-05-15:
This is looking great! 
We used to debug and trace the same with eCos and it's internal tools which solved issues in hours instead of weeks.

microbuilder, 2024-01-10:
@gromero Any update on getting this over the finish line?

TaiJuWu, 2024-02-06:
Is it ready to merge?

gromero, 2024-03-06:
> Is it ready to merge?

@TaiJuWu @microbuilder Unfortunately, not yet. I need to remove the variable length support in the ring buffer (use Zephyr API instead) and test it on a VM as requested by @carlocaione in Prague. I didn't give up. It's been just hard to find the cycles I need to address it. Otherwise, I deem it done.

gromero, 2025-10-07:
https://github.com/zephyrproject-rtos/zephyr/pull/91595 revolved this RFC, so I'm closing this issue. Thanks a lot for all the efforts, comments, suggestions, and reviews!

closed 2025-10-07 by gromero