zephyrproject-rtos/zephyr #57376: STM32 DMAMUX and DMA share initialisation level and priority

bp-amitchone, 2023-04-28:
As mentioned in the title, the [DMAMUX driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/dma/dmamux_stm32.c#L425) and the [DMA driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/dma/dma_stm32.c#L725) share the initialisation priority `CONFIG_DMA_INIT_PRIORITY`. The issue with this is that `dmamux_stm32_init` can possibly return `-ENODEV` due to `if (device_is_ready(DEVICE_DT_GET(DT_NODELABEL(dma1))) == false)` and thus the DMA will not be properly initialised - my understanding being that if drivers share the same initialisation level and priority then the order in which they are initialised is ambiguous and, as such, if the DMAMUX is initialised before the DMA, then the DMAMUX will fail to initialise properly. 

This is present in the [current version of main](https://github.com/zephyrproject-rtos/zephyr/commit/14406cc664a15419ee6591297ba7c7a6cc6f06c3).

**To Reproduce**
To verify that the DMAMUX initialisation fails it is possible to manually specify a higher priority for the DMAMUX than the DMA such that the DMAMUX initialisation function is called first.

**Expected behavior**
DMA should be initialised before DMAMUX.

**Impact**
Very minor - this is easily worked around for now.

erwango, 2023-05-03:
@jgl-meta Not sure why you set prio to medium here:
- @bp-amitchone explicitly stated Impact "Very minor".
- Problem is not systematic, STM32 only, linked to only one driver, ..

FRASTM, 2023-06-22:
This very low probable as both drivers dma_stm32.c and dmamux_stm32.c have the same CONFIG_DMA_INIT_PRIORITY (=40)
We could still add a specific config DMAMUX_STM32_INIT_PRIORITY  = 41 
and add a _build_assert_ to be sure the DMA device is initialized before the DMAMUX device.

closed 2023-06-23 by carlescufi