zephyrproject-rtos/zephyr #57405: mps2_an385 Test Failure

jgl-meta, 2023-04-29:
**Describe the bug**
The ztests for net.socket.tls.preempt are failing for mps2_an385 it looked like the tests may be timing out due to a failure to allocate a lock due to running out of memory. These tests have been disabled and need to be reenabled.

**To Reproduce**
Steps to reproduce the behavior:
1. On linux setup a network bridge and the net-tools using https://docs.zephyrproject.org/3.1.0/connectivity/networking/native_posix_setup.html
2. Run the test set mentioned above with twister or west
3. Currently one test set will pass and one test set will fail  
4. See error in the json report.

**Expected behavior**
The test should pass.

**Impact**
If the out of memory error is in fact happening, the way it seems to happen for me due to something other than user error, it is perplexing as to why this would happen since these boards have 4M of memory.

jgl-meta, 2023-05-05:
@stephanosio I am still running into the same issue despite increasing the stack size and memory size in the test. Specifically, I am seeing this error,
`
Running TESTSUITE net_socket_tls
===================================================================
START - test_so_protocol
 PASS - test_so_protocol in 1.014 seconds
===================================================================
START - test_so_type
 PASS - test_so_type in 1.010 seconds
===================================================================
START - test_v4_dtls_sendmsg
 PASS - test_v4_dtls_sendmsg in 0.065 seconds
===================================================================
START - test_v4_msg_trunc
 PASS - test_v4_msg_trunc in 0.031 seconds
===================================================================
START - test_v4_msg_waitall
 PASS - test_v4_msg_waitall in 0.516 seconds
===================================================================
START - test_v6_dtls_sendmsg
E: could not allocate kernel object, out of memory
ASSERTION FAIL [*lock != ((void *)0)] @ WEST_TOPDIR/zephyr/lib/libc/newlib/libc-hooks.c:385
        recursive lock allocation failed
E: r0/a1:  0x00000004  r1/a2:  0x00000181  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00025bc5
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x00025bde
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20021ce8 (test_v6_dtls_sendmsg)
E: Halting system
</failure></testcase></testsuite></testsuites>
`

Is this user error? I am new to debugging network issues in Zephyr. Is there a good way to attach a debugger to this? When running `west debug` I seeing a message that it is not supported?

jhedberg, 2023-09-26:
@jgl-meta are you still working on this? Should it be closed?

jgl-meta, 2023-09-26:
> @jgl-meta are you still working on this? Should it be closed?

I haven't had cycles for a little while to circle back and look at it. But it should still be open. I will circle back as soon as I can.

carlescufi, 2023-10-04:
@jgl-meta tests for this platform were disabled here: https://github.com/zephyrproject-rtos/zephyr/pull/63220 due to https://github.com/zephyrproject-rtos/zephyr/issues/48608. Should we close this issue as a duplicate of #48608?

jgl-meta, 2023-10-04:
> @jgl-meta tests for this platform were disabled here: #63220 due to #48608. Should we close this issue as a duplicate of #48608?

It indeed does look like it.

Closing as duplicate.