zephyrproject-rtos/zephyr #57408: The samples/basic/button which replaced samples/boards/pine64_pinetime did not work after build and flashing a PineTime DK

ljunquera, 2023-04-29:
I was looking to run a Hello World sample for PineTime DK and saw references to a `samples/boards/pine64_pinetime` which would make for a good starting point. I didn't see it in the samples so I searched and found [samples: boards: pinetime: Remove duplicate sample](https://github.com/zephyrproject-rtos/zephyr/commit/92e317c158b3ea690baed5180a937ab82551c675):

> samples/boards/pine64_pinetime has the same functionality as the more
> general samples/basic/button application with the board specific
> driver `pinetime-key-out`.

So I created a new application in VS Code using the nRF Connect Add-In and used the `samples/basic/button` with the `PineTime DevKit0` board, built  and flashed the PineTime DK, but saw nothing on the screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new application in VS Code based on the `samples/basic/button`
2. Add a build configuration for `pinetime_devkit0`
3. build
4. flash, I used Open OCD and a ST-Link v2:
```
openocd \
     -c 'source [find interface/stlink.cfg]' \
     -c 'transport select hla_swd' \
     -c 'source [find target/nrf52.cfg]' \
     -c 'init' \
     -c 'halt' \
     -c 'nrf5 mass_erase' \
     -c 'program zephyr.hex verify' \
     -c 'reset' \
     -c 'exit'
```

**Expected behavior**
Expected to see a button. I used the peripheral sample, which worked, on the pinetime_devkit0, so I know the process works.

**Impact**
It's hard to get started with a working base example as outlined in the [YouTube video tutorial](https://www.youtube.com/watch?v=qbbQlSLO0V8&list=PLx_tBuQ_KSqEt7NK-H7Lu78lT2OijwIMl&index=2) which recommends using an existing example as a starting point.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr, VSCode + nRF Connect Add-In
 - ST-Link v2, OpenOCD
 - PineTime DK

henrikbrixandersen, 2023-04-30:
Please post the build failure output.

ljunquera, 2023-04-30:
@henrikbrixandersen, I didn't see that as an option.

henrikbrixandersen, 2023-05-01:
> @henrikbrixandersen, I didn't see that as an option.

You can post it here, on the issue report.

ljunquera, 2023-05-04:
I read about [Applying a Label](https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/managing-labels#applying-a-label) and I'm not sure I have triage access.

Is this a backlight issue? It seems like this should be fairly straight forward.

henrikbrixandersen, 2023-05-04:
@ljunquera Please post the command used to build and the complete output including all error messages detailing the issue, you are seeing.

ljunquera, 2023-05-04:
Ok. Build command was this:
[image]

I don't see any error messages. I flash the PineTime using the ST-Link and OpenOCD and nothing happens on the PineTime. It's just a blank screen.

henrikbrixandersen, 2023-05-08:
> Ok. Build command was this: [image]
> 
> It builds fine. I don't see any error messages. I flash the PineTime using the ST-Link and OpenOCD and nothing happens on the PineTime. It's just a blank screen.

The sample in `samples/basic/button` does not have any GUI. It solely uses a hardware push-button and prints messages to the console. Try connecting to the console of the PineTime and see the output there.

closed 2023-07-23 by github-actions[bot]