zephyrproject-rtos/zephyr #57412: arch: arm: aarch32: cortex_m: fault: ARMv6-M illegal instruction fetch triggers lockup before any fault debug output

watsonkm, 2023-04-30:
**Describe the bug**
When attempting to fetch an instruction from an address on an Cortex-M CPU in an MPU region with the XN attribute, a lockup is triggered before the HardFault handler can output any debug information.

As an example, running the MPU test in the samples directory outputs this when attempting an invalid read:

```
<err> os: ***** HARD FAULT *****
<err> os: r0/a1:  0x0800a54c  r1/a2:  0x00000008  r2/a3:  0x08003
<err> os: r3/a4:  0x24000000 r12/ip:  0x00000040 r14/lr:  0x0800d
<err> os:  xpsr:  0x01000000
<err> os: Faulting instruction address (r15/pc): 0x08000486
<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
<err> os: Current thread: 0x200006a8 (shell_uart)
<err> os: Halting system
``` 

A similar output happens on an invalid write. When attempting an invalid instruction fetch however, the processor locks up before any output occurs.

This bug occurs as the hard fault handler `hard_fault` in `zephyr/arch/arm/core/aarch32/cortex_m/fault.c` employs a workaround implemented in `z_arm_is_synchronous_svc` to determine if the HardFault originated from a SVC instruction. To do so, `z_arm_is_synchronous_svc:710` (GitHub link to line [here](https://github.com/zephyrproject-rtos/zephyr/blob/12ed08a2fb02515b2469f3e7d0f91ae4b03ea83e/arch/arm/core/aarch32/cortex_m/fault.c#L710)) dereferences the ESF PC:

```c
...

#if defined(CONFIG_ARMV6_M_ARMV8_M_BASELINE) && !defined(CONFIG_ARMV8_M_BASELINE)
	/* Note: ARMv6-M does not support CCR.BFHFNMIGN so this accesstriggered
	 * could generate a fault if the pc was invalid.
	 */
	uint16_t fault_insn = *(ret_addr - 1);
#else

...
```
The synchronous SVC detection change was introduced in #18712, and a fix to the bug described in this issue for Cortex-M CPUs that have distinct HardFaults and BusFaults (ARMv7-M and ARMv8-M) was introduced in #48487, along with the above comment indicating this same bug.

**To Reproduce**
To reproduce this behavior, follow the following steps (as outlined in the [MPU test documentation](https://docs.zephyrproject.org/latest/samples/arch/mpu/mpu_test/README.html) after #57104 was merged):

- Connect a ARMv6-M development board over USB (I used the Nucleo G071RB for my tests)
- Start the MPU test under gdb via `west debug`
- Execute `mpu run` over UART (i.e. via `minicom`)
- Interrupt gdb with Ctrl-C
- Display the program counter by using `l` in gdb

The program counter should be listed as `0xfffffffe`, indicating a lockup. Note in particular that there is no output over the console.

**Expected behavior**
The expected behavior, given that read/write MPU failures trigger HardFaults, is that an invalid instruction fetch should trigger a HardFault as well.

**Impact**
This bug could significantly confuse Zephyr users who aren't aware that MPU invalid instruction fetches trigger lockups, as there is no debug output to work from. Moreover, this behavior for ARMv6-M does not line up with that of any other Cortex-M architecture, potentially leading to additional confusion. I only discovered this after a few days and some lucky discoveries in the Zephyr source and the ARMv6-M Architecture Reference Manual, so this bug could cause timeline delays in applicatoins leveraging ARMv6-M MPU support in Zephyr.  

**Logs and console output**
The full outputs of the MPU test as run on my development laptop with a Nucleo G071RB are here:

```
uart:~$ mpu read
<err> os: ***** HARD FAULT *****
<err> os: r0/a1:  0x0800a54c  r1/a2:  0x00000008  r2/a3:  0x08003
<err> os: r3/a4:  0x24000000 r12/ip:  0x00000040 r14/lr:  0x0800d
<err> os:  xpsr:  0x01000000
<err> os: Faulting instruction address (r15/pc): 0x08000486
<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
<err> os: Current thread: 0x200006a8 (shell_uart)
<err> os: Halting system
```

```
mpu write
<err> os: ***** HARD FAULT *****
<err> os: r0/a1:  0x00000000  r1/a2:  0x0000000e  r2/a3:  0x0000e
<err> os: r3/a4:  0x0badc0de r12/ip:  0x00000000 r14/lr:  0x08009
<err> os:  xpsr:  0x61000000
<err> os: Faulting instruction address (r15/pc): 0x0800046a
<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
<err> os: Current thread: 0x200006a8 (shell_uart)
<err> os: Halting system
```

```
uart:~$ mpu run
<no output>
```

**Environment**
The following environment was used for the above tests:
 - OS: Ubuntu 22.04
 - Zephyr SDK
 - Zephyr commit `c2ec3d49cd2f952e0ead3dc7415d6cc9845a0db8`

**Proposed solutions**
As supported in the above comment, there is no way I know of to temporarially disable HardFaults on ARMv6-M CPUs.

One alternative could be to, for ARMv6-M CPUs only, disable the MPU before the PC dereference and then enable it immediately after. This could constitute a security risk though, given the HardFault handler is then able to bypass the MPU to read an address provided from user mode.

Another method could be to have the HardFault handler, for ARMv6-M CPUs only, read out the current MPU regions to a priori determine if dereferencing the ESF PC would result in a fault. This would likely add significant delay to the handler, and is a touch hacky, but since HardFaults are treated as generally unrecoverable in ARMv6-M, it could be a slightly better option.

Given I'm relatively new to Zephyr, I'd personally like to hear what developers' thoughts are both on the security implications of disabling the MPU for a single read in the HardFault handler, as well as whether there are any better solutions I could try to help implement.

(Thank you so much!)

watsonkm, 2023-07-02:
Would this issue be something worth diving further into? If not, I'm okay with it being closed.

microbuilder, 2023-07-02:
> Would this issue be something worth diving further into? If not, I'm okay with it being closed.

Hi, sorry. I'd very much like to fix this, and I'll look at a solution this week now that the release is finished and ZDS is over.

watsonkm, 2023-07-03:
Sounds great! No worries at all, and let me know if there's any way I can help.

microbuilder, 2023-07-05:
The problem here is clear (and what you identified): `ret_addr` is `0x20000000` in this test case, and decrementing it one instruction to determine if it's `SVC` (formerly `SWI` in older arches) puts `ret_addr` at an invalid address, which causes the secondary fault, locking the core up:

```c
uint16_t fault_insn = *(ret_addr - 1);
```

I was talking about this with a colleague today, but I'll have to think about the best way to handle this for v6-M, so it might take a day or two to put a PR together. Thanks for your patience.

microbuilder, 2023-07-06:
Memory range lookup may be solvable with the changes proposed in this PR to determine not only the valid range, but MPU properties for that range (executable, etc.): https://github.com/zephyrproject-rtos/zephyr/pull/60049

ithinuel, 2023-07-12:
Hello, would https://github.com/zephyrproject-rtos/zephyr/pull/60282 be an acceptable solution to this issue?

microbuilder, 2023-07-12:
> Hello, would #60282 be an acceptable solution to this issue?

This should solve the issue here of `0x20000000 - 2` leading to an invalid `pc`, yes. Thanks for the PR.

microbuilder, 2023-07-12:
@watsonkm With https://github.com/zephyrproject-rtos/zephyr/pull/60282 I now get the output below. Do you mind testing yourself? Thanks!

```
uart:~$ mpu run
[00:00:31.405,346] <err> os: ***** HARD FAULT *****
[00:00:31.415,016] <err> os: r0/a1:  0x0001ea94  r1/a2:  0x00000001  r2/a3:  0x2000132c
[00:00:31.416,232] <err> os: r3/a4:  0x20000000 r12/ip:  0x000002f8 r14/lr:  0x00012ec5
[00:00:31.417,168] <err> os:  xpsr:  0x00000000
[00:00:31.417,913] <err> os: Faulting instruction address (r15/pc): 0x20000000
[00:00:31.419,363] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:31.421,438] <err> os: Current thread: 0x20000458 (shell_uart)
[00:00:31.423,293] <err> os: Halting system
```

PR #60282 (2023-07-13): arch: arm: cortex_m: fault: Improve handling of fault escalation on armv6-m
The current implementation causes a lockup of the core when the exception originates from an invalid/unreachable pc. This fix first verifies on armv6-m and armv8-m.base that pc was in an expected runnable region, namely:

- .text
- .ramfunc
- .itcm

Fixes: #57412
closed 2023-07-13 by carlescufi