zephyrproject-rtos/zephyr #57423: Relocating libraries issues(zephyr_code_relocate)

hongshui3000, 2023-05-01:
1.If the serial port library is out of the tree, this method will not work properly
The file path that appears is the superposition of the two paths

```
zephyr_code_relocate(LIBRARY drivers__serial LOCATION SRAM2)
```

log

```
E:\work\MYTEST\IOT\zephyr\scripts\build\gen_relocate_app.py:490: 
UserWarning: No files in string: E:/work/MYTEST/IOT/zephyr/drivers/serial/E:/work/MYTEST/IOT/sdk/drivers/serial/uart_openc.c;
E:/work/MYTEST/IOT/zephyr/drivers/serial/E:/work/MYTEST/IOT/sdk/drivers/serial/uarth_openc.c found
  warnings.warn("No files in string: "+file_list+" found")
```

2.Libraries generated by external projects cannot use this method to reallocate mem
for example
```
Exteranl lib Cmakefile：
ExternalProject_Add(
  mytest
  SOURCE_DIR ${APPLICATION_SOURCE_DIR}/mytwst
  INSTALL_COMMAND ""      # This particular build system has no install command
  CMAKE_CACHE_ARGS -DBOARD:STRING=${BOARD_REMOTE}
  BUILD_BYPRODUCTS "${MYTEST_ZEPHYR_DIR}/${KERNEL_BIN_NAME}"
  # NB: Do we need to pass on more CMake variables?
  BUILD_ALWAYS True
)

....
zephyr_library_import(MYTES ${mytest_lib_dir}/test.lib)

my app：
zephyr_code_relocate(LIBRARY MYTES LOCATION SRAM2)

```

tejlmand, 2023-06-14:
> 1.If the serial port library is out of the tree, this method will not work properly
The file path that appears is the superposition of the two paths

This looks like a bug that must be investigated.

> 2.Libraries generated by external projects cannot use this method to reallocate mem
for example

and this is not an issue.
The possibility to relocate libraries are only supported (and intended to support) CMake libraries created in the current CMake invocation, not imported libraries.

Should be checked if some docs must be updated.

hongshui3000, 2023-06-15:
There is also a noinit data segment that needs to be enhanced, which cannot be formally processed through this   zephyr_code_relocate(LIBRARY drivers__serial LOCATION SRAM2)  noinit data segment, will not be put into SRAM2

hongshui3000, 2023-06-15:
> and this is not an issue. The possibility to relocate libraries are only supported (and intended to support) CMake libraries created in the current CMake invocation, not imported libraries.
> 
> Should be checked if some docs must be updated.

Will the sdk-nrfxlib library work? If using zephyr_code_relocate

hongshui3000, 2023-06-17:
@nordicjm 
[2/15] Generating code_relocation.c, include/generated/linker_relocate.ld
E:\work\MYTEST\IOT\zephyr\scripts\build\gen_relocate_app.py:530: UserWarning: File: E:/work/MYTEST/IOT/zephyr/drivers/serial/E:/work/MYTEST/IOT/ZCAN-sdk/drivers/serial/uart_openc.c Not found
  warnings.warn("File: "+file_glob+" Not found")
E:\work\MYTEST\IOT\zephyr\scripts\build\gen_relocate_app.py:536: UserWarning: No files in string: E:/work/MYTEST/IOT/zephyr/drivers/serial/E:/work/MYTEST/IOT/ZCAN-sdk/drivers/serial/uart_openc.c found
  warnings.warn("No files in string: "+file_list+" found")

nordicjm, 2023-06-19:
@hongshui3000 are you able to upload a minimal reproduction of this issue as a repo or zip?

nordicjm, 2023-06-19:
I've tried this by combining the `samples/application_development/out_of_tree_driver` and `samples/application_development/code_relocation_nocopy` samples and tried on windows and it builds fine for me though this is not acting as a serial driver. I've put the code on https://github.com/nordicjm/zephyr/tree/testbug - compare that to how you are setting your project up as it does not look to be a zephyr issue

hongshui3000, 2023-06-19:
> @hongshui3000 are you able to upload a minimal reproduction of this issue as a repo or zip?

My sdk is modified according to this, have you tested it on this?
https://github.com/zephyrproject-rtos/example-application/tree/main/drivers

nordicjm, 2023-06-19:
> > @hongshui3000 are you able to upload a minimal reproduction of this issue as a repo or zip?
> 
> My sdk is modified according to this, have you tested it on this? https://github.com/zephyrproject-rtos/example-application/tree/main/drivers

Can you upload a minimal reproduction of this issue?

hongshui3000, 2023-06-19:
> > > @hongshui3000 are you able to upload a minimal reproduction of this issue as a repo or zip?
> > 
> > 
> > My sdk is modified according to this, have you tested it on this? https://github.com/zephyrproject-rtos/example-application/tree/main/drivers
> 
> Can you upload a minimal reproduction of this issue?

I'll give it a try, it might take a while

nordicjm, 2023-06-30:
@dcpleung I'm on the verge of transferring this to you. I have (accidentally) replicated this on linux, and it does seem that some paths randomly have the path in twice e.g. "/tmp/bb/zephyr/modules/hal_nordic/nrf_802154//tmp/bb/zephyr/modules/hal_nordic/nrf_802154/serialization/platform/nrf_802154_spinel_response_notifier.c|RAM_DATA", any thoughts on what is happening here?

nordicjm, 2023-06-30:
@hongshui3000 you cannot, at present, use absolute paths for source files when relocating files, the paths need to be relative from the cmake file due to how prepending source directories works with generator expressions.

manuargue, 2023-07-14:
Hi @nordicjm I still can reproduce a similar issue on Windows 10 / Zephyr SDK 0.16.0 @ 76663444c1faa8618ddf0cef735b77ffffce4700.
```
west build -p always -b qemu_cortex_m3 tests/application_development/code_relocation -T tests.application_development.code_relocation.no_itcm
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/user/zephyr-github/zephyr/tests/application_development/code_relocation      
-- CMake version: 3.25.1
-- Found Python3: C:/Users/user/AppData/Local/Programs/Python/Python311/python.exe (found suitable exact version "3.11.1") found components: Interpreter
-- Cache files will be written to: C:/Users/user/zephyr-github/zephyr/.cache
-- Zephyr version: 3.4.99 (C:/Users/user/zephyr-github/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.14.0")
-- Board: qemu_cortex_m3
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.0 (C:/Users/user/tools/zephyr-sdk-0.16.0)
-- Found toolchain: zephyr 0.16.0 (C:/Users/user/tools/zephyr-sdk-0.16.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/user/zephyr-github/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts   
-- Generated zephyr.dts: C:/Users/user/zephyr-github/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/user/zephyr-github/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/user/zephyr-github/zephyr/build/zephyr/dts.cmake
Loaded configuration 'C:/Users/user/zephyr-github/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig'
Merged configuration 'C:/Users/user/zephyr-github/zephyr/tests/application_development/code_relocation/prj.conf'
Configuration saved to 'C:/Users/user/zephyr-github/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/user/zephyr-github/zephyr/build/zephyr/include/generated/autoconf.h'

-- Found GnuLd: c:/users/user/tools/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/user/tools/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe     
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/user/zephyr-github/zephyr/build
←[92m-- west build: building application
[1/145] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (C:/Users/user/zephyr-github/zephyr), build: zephyr-v3.4.0-1121-g76663444c1fa
[135/145] Generating code_relocation.c, include/generated/linker_relocate.ld
C:\Users\user\zephyr-github\zephyr\scripts\build\gen_relocate_app.py:530: UserWarning: File: C:/Users/user/zephyr-github/zephyr/tests/application_development/code_relocation/test_lib/C:/Users/user/zephyr-github/zephyr/tests/application_development/code_relocation/test_lib/test_lib2.c Not found
  warnings.warn("File: "+file_glob+" Not found")
[145/145] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           SRAM2:        5056 B        32 KB     15.43%
           FLASH:       26362 B       256 KB     10.06%
             RAM:        6662 B        64 KB     10.17%
        IDT_LIST:          0 GB         2 KB      0.00%
```
Using relative path for `test_lib2.c` as was done prior https://github.com/zephyrproject-rtos/zephyr/pull/59872/files#diff-661b08eb70b62efc89b72faa81d20ca0c0ff5e9b4c5875bc67853ef120f65e19R5 works fine.

The issue also happens when using other libraries relocation, for example, just for the sake of testing adding to the bottom of https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/application_development/code_relocation/CMakeLists.txt:
```cmake
zephyr_code_relocate(LIBRARY zephyr LOCATION SRAM2)
```
outputs:
```
west build -p always -b qemu_cortex_m3 tests/application_development/code_relocation -T tests.application_development.code_relocation.no_itcm
-- west build: making build dir C:\Users\user\zephyr-github\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/user/zephyr-github/zephyr/tests/application_development/code_relocation      
-- CMake version: 3.25.1
-- Found Python3: C:/Users/user/AppData/Local/Programs/Python/Python311/python.exe (found suitable exact version "3.11.1") found components: Interpreter
-- Cache files will be written to: C:/Users/user/zephyr-github/zephyr/.cache
-- Zephyr version: 3.4.99 (C:/Users/user/zephyr-github/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.14.0")
-- Board: qemu_cortex_m3
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.0 (C:/Users/user/tools/zephyr-sdk-0.16.0)
-- Found toolchain: zephyr 0.16.0 (C:/Users/user/tools/zephyr-sdk-0.16.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/user/zephyr-github/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
[... 55 lines cut ...]
  warnings.warn("File: "+file_glob+" Not found")
[... cut ...] .c;C:/Users/user/zephyr-github/zephyr/C:/Users/user/zephyr-github/zephyr/subsys/tracing/tracing_none.c found
  warnings.warn("No files in string: "+file_list+" found")
[145/145] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           SRAM2:        5056 B        32 KB     15.43%
           FLASH:       26362 B       256 KB     10.06%
             RAM:        6662 B        64 KB     10.17%
        IDT_LIST:          0 GB         2 KB      0.00%
```

PR #59872 (2023-07-10): cmake: Fix code relocation functions with absolute paths
Fixes an issue whereby absolute paths in libraries would have duplicate paths in the name, causing issues in that they would not be relocated.

Fixes #57423