zephyrproject-rtos/zephyr #57448: Network MGMT and LLDP stuck within iface_event_handler upon NET_EVENT_IF_DOWN handling

GeorgeCGV, 2023-05-01:
The kernel is stuck within https://github.com/zephyrproject-rtos/zephyr/blob/ef3d01d16fd462805565d83443db7a7bd38d306d/subsys/net/ip/net_mgmt.c#L179-L183
https://github.com/zephyrproject-rtos/zephyr/blob/ef3d01d16fd462805565d83443db7a7bd38d306d/subsys/net/l2/ethernet/lldp/lldp.c#L308-L312
handling `NET_EVENT_IF_DOWN` event:
https://github.com/zephyrproject-rtos/zephyr/blob/ef3d01d16fd462805565d83443db7a7bd38d306d/subsys/net/l2/ethernet/lldp/lldp.c#L245-L251

I use STM32H730 target with a custom L2 ethernet driver that defines 2 interfaces.
Each interface handles link status with `net_eth_carrier_on` and `net_eth_carrier_off` within offloaded IRQ handler.

The application is configured to send LLDP packages via both interfaces.

Everything works fine at startup when the cable is connected at the board's startup (_link present at boot process_) or when the cable is connected after the startup (_link present established after boot process_).
However, as soon as (_in previously mentioned cases_) the cable is detached the link goes down and LDDP begins to handle `NET_EVENT_IF_DOWN` the whole firmware gets stuck and is not able to finish handling the event.

It looks like each `cb->node` has `next` pointing to itself "infinitely": https://github.com/zephyrproject-rtos/zephyr/blob/ef3d01d16fd462805565d83443db7a7bd38d306d/subsys/net/ip/net_mgmt.c#L183

I am not very familiar with that code part. Therefore, open to suggestions.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a board with several L2 ethernet interfaces that able to handle link state change
2. Configure LLDP to send LLDPDU on each interface
3. Start the board so that link is established
4. Observe LLDP frames
5. Disconnect the cable
6. Observe the freeze/stuck condition

**Expected behavior**
The `NET_EVENT_IF_DOWN` event shall be correctly handled. Ideally, the LLDP frames sending shall start as soon as the link is established again.

**Impact**
Showstopper. The application can't recover after the ethernet cable is unplugged.

**Logs and console output**
No errors are observed. No log output is produced with enabled `NET_LOG` and `NET_MGMT_EVENT_LOG_LEVEL_DBG`.

**Environment (please complete the following information):**
 - Toolchain: arm-none-eabi 2021.07
 - Zephyr [a0ad7b](https://github.com/zephyrproject-rtos/zephyr/commit/a0ad7b)

GeorgeCGV, 2023-05-02:
Should be reproducible with a single interface as long as it is able to trigger `NET_EVENT_IF_DOWN`.
The issue seems to be with the event handling.

@rlubos please let me know if you require any additional information.

GeorgeCGV, 2023-05-04:
@rlubos after additional investigation the problem is found.

When `CONFIG_NET_CONFIG_AUTO_INIT` then `net_lldp_init` is going to be called in `net_post_init`. Therefore, there is no need to initialize it manually in the application. Unfortunately, that is exactly the situation that I had.

Maybe it could be beneficial to mention that in the `net_lldp_init` comment.

There is another question related to the LLDP and networking stack...
The LLDP subscribes to `NET_EVENT_IF_UP` and `NET_EVENT_IF_DOWN` to control tx work. However, at start-up when it iterates over interfaces it checks if `NET_IF_UP` flag is set, and starts LLDP TX in that case. That flag is set by `net_if` within `net_if_post_init` if `NET_IF_NO_AUTO_START` is not set. That means, when a carrier is off the LLDP will attempt to send the packages.

Shouldn't carrier off put iface into the down state? Or what is the idea?

GeorgeCGV, 2023-05-09:
@rlubos experimented a bit more. There is a problem, at least in my case with eth driver that manages carrier status.

The application has 2 network interfaces. Both are managed as discussed before:

> net_if_carrier_off in the iface_init and then control carrier with net_eth_carrier_<off|on>

Both interfaces are down at init, then it is enough for any of them to be up before `net_lldp_init` gets is executed.

The `lldp_start` is called at init time if iface is up from `net_lldp_init` during iface iteration. It also subscribes to iface up and down events.

The problem happens when iface is up before `net_lldp_init`. That results in double `lldp_start` execution:

1. Because iface is UP when it iterates each iface using `net_if_foreach` 
2. Receives UP event from net manager

That starts more than one timer for ex work. The result is lldp getting stuck within slist iteration in `lldp_tx_timeout`.

It works as expected as soon as iface iteration is removed from `net_lldp_init` to get rid of the first `lldp_start` call.

rlubos, 2023-05-09:
It seems to me that `lldp_start()` should check whether it has already been configured (started) on an interface. Maybe `tx_timer_start` could be used to verify if LLDP is already running (it's set to uptime on start).

GeorgeCGV, 2023-05-09:
Yes. Adding `tx_timer_start != 0` check into `lldp_start_timer` and reset of the `tx_timer_start` to `0` within down handling fixes the problem.

rlubos, 2023-05-09:
@GeorgeCGV Can you send a PR with the change you've tested?

GeorgeCGV, 2023-05-09:
@rlubos sure, but have to wait until GitHub is fully operational again

closed 2023-05-10 by fabiobaltieri