zephyrproject-rtos/zephyr #57464: Function websocket_recv_msg() of WebSocket subsystem ignores the timeout parameter

c4explosive, 2023-05-02:
**Describe the bug**
Using the function "websocket_recv_msg", this function accept various parameters like the [documentation says](https://docs.zephyrproject.org/latest/connectivity/networking/api/websocket.html#c.websocket_recv_msg). However, the timeout parameter apparently is ignored or is not respected by the function, so the routine is called forever like the parameter was "SYS_FOREVER_MS" or with infinite timeout.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   I'm using ESP32 WROVER dev board 

 - What have you tried to diagnose or workaround this issue?
   I tried different values in timeout parameters and gave the same result. Only the value 0 give different behavior. So, using 0 and checking if the function returns EAGAIN, i put this in a loop and check with a condition if the the time passed is greater than the desired timeout. Something like this:

 ```
 ...
 int64_t time = k_uptime_get();
 int64_t timeout = 20000;  //20 seconds
 while( k_uptime_get() - time < timeout)
{
    int ret = websocket_recv_msg(sck, buf, sizeof buf, &m_type, &reamaining, 0);
    if( ret == -EAGAIN)
       continue;
    //do something with the buffer
}
//timeout expired here
...
```
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
   No.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a websocket using the API like described in the [documentation](https://docs.zephyrproject.org/latest/connectivity/networking/api/websocket.html#websocket-transport).
2. Call the function websocket_recv_msg() with any timeout parameter value (except 0 or SYS_FOREVER_MS)
3. If the other end (or device) do not send any message, the function will hang forever and the code below this call will never be called, even if the timeout is expired.

**Expected behavior**
The functions should return after the timeout is expired, giving the return value -EAGAIN.

**Impact**
If the function hangs forever, the function will stuck the entire routine and some functionality of the board who use the Web Socket subsystem may be disabled and restart will be needed for enable again.

**Logs and console output**
_No aplicable_

**Environment (please complete the following information):**
 - OS: Archlinux
 - Toolchain Zephyr SDK 3.2.99
 - Commit SHA or Version used:  4f3d5a10c43711c91e3e5bf6d721c2c1703ee661

**Additional context**
_No aplicable_

rlubos, 2023-05-24:
@c4explosive Please check https://github.com/zephyrproject-rtos/zephyr/pull/58240

c4explosive, 2023-05-25:
Confirmed, i applied the changes made in #58240 (temporally/local) and tested the code using the websocket_recv_msg() function with three different values for the timeout parameter: 3000 ms, 5000 ms and 1 minute; **all works as expected**. Thanks so much. ¿Do i need to close this issue?.

rlubos, 2023-05-25:
Glad to hear that. No need to close the issue - Github will close it automatically when the PR gets merged.

PR #58240 (2023-05-31): net: websocket: Implement websocket_recv_msg timeout
Although websocket_recv_msg function accepts timeout parameter, the functionality was rather limited, allowing only to either work in non-blocking manner, or to block indefinitely. Any timeout value other than -1 (forever) ended up in non-blocking operation.

This PR fixes this by implementing a basic timeout mechanism, built on top of poll(). For now on, only timeout of 0 will result in non-blocking operation, any other timeout will make the function block for the specified amount of time.

Fixes #57464
closed 2023-05-31 by carlescufi