zephyrproject-rtos/zephyr #57471: LE Audio: HAS optional notification for Hearing Aid Features

Thalley, 2023-05-02:
**Is your enhancement proposal related to a problem? Please describe.**
The Hearing Aid Features characteristic is optionally notifiable, which should be reflected in the implementation. It should be configurable by a Kconfig whether to support this or not (for both the server and client implementations). 

**Describe the solution you'd like**
Implement optional support for notifications for the Hearing Aid Features characteristic. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

fredrikdanebjer, 2023-05-11:
This should be addressed in https://github.com/zephyrproject-rtos/zephyr/pull/57785

Thalley, 2023-05-11:
> This should be addressed in #57785

@fredrikdanebjer Please use the closing keywords in Github to link PRs and issues: https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword :) That way the issues are automatically closed when the PR fixing the issue is merged

PR #57785 (2023-05-26): Bluetooth: Audio: Make HAS optional notify characteristics optional
Added Kconfig options to make HAS characteristics that has notify as optional property selectable and thus optional. These settings are global, meaning that a chosen notify property will be used for all registered has instances.

closes #57471