zephyrproject-rtos/zephyr #57496: Bluetooth: Controller: BLE active scan with PA/LNA crash

carlst, 2023-05-02:
**Describe the bug**
Running `samples/bluetooth/hci_usb` using  `generic-fem-two-ctrl-pins` 
- crashes if PA/LNA enabled and BLE **active** scanning used
- normal operation if PA/LNA bypassed
- normal operation if PA/LNA enabled but passive scanning (or advertising) used

Please also mention any information which could help others to understand
the problem you're facing:
 - using custom board with Fanstel BT840XE and `fem.overlay`
 -  `fem.overlay` defines 2 pins (for PA Tx and LNA Rx), settle time, and gain
 -  `fem.overlay` also defines 2 PA/LNA configuration pins (statically set at startup)
 - crash occurs if PA/LNA enabled (by configuration pins at startup), active scan performed
 - using Segger Embedded Studio (SES)
 - SES displays: 
`Mis-aligned memory read: Address: 0x0000006B, NumBytes: 4, Alignment: 4 (Word-aligned)
Mis-aligned memory read: Address: 0x0000006F, NumBytes: 4, Alignment: 4 (Word-aligned)
Mis-aligned memory read: Address: 0x00000073, NumBytes: 4, Alignment: 4 (Word-aligned)
Mis-aligned memory read: Address: 0x00000077, NumBytes: 4, Alignment: 4 (Word-aligned)
Mis-aligned memory read: Address: 0x0000007B, NumBytes: 4, Alignment: 4 (Word-aligned)`
 
- SES crashes without a stack trace. All registers are `0xffffffff`
- SES does not stop if hardware breakpoint (read) enabled
 - Zephyr 3.3.0

**To Reproduce**
Steps to reproduce the behavior:
1. Build `samples/bluetooth/hci_usb` modified `src/main.c`  to statically set 2 PA/LNA configuration pins
2. Download and run application

**Expected behavior**
Would expect active scan to operate similar to passive scan.

**Impact**
Low Priority for Zephyr (but showstopper for this custom project)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Zephyr SDK 0.15.2
 - Zephyr 3.3.0

**Additional context**
BT840XE requires 2 additional FEM pins to configure PA/LNA. 1st pin controls operation (PA/LNA on or bypass). 2nd pin controls Tx power (low or high). Both values are set statically at start-up. Only Tx power low is used in tests.

cvinayak, 2023-05-03:
Is there in anyway you can have a UART setup in your custom board, so that if there is a controller or OS assertion it can help debug.

At this point, I suspect program stack overflow in one of the thread in hci_usb, but a console log would help. 

Add the below to get program stack usage once you have an UART configured:
```
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_USE_PRINTK=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_NAME=y
```

carlst, 2023-05-04:
The custom board does not provide a UART, but the thread analyzer output is available through RTT (attached file). Thank you for the recommendation.

**However, the crash stopped occurring today; active scan with PA/LNA now works fine.**

The only explanations I can offer are:
- one or more nearby Advertising devices are not operating (the test area is in a multi-office environment where occupants vary day-day)
- the build became corrupted (although the image was built completely from scratch and tested several times)

I did run several fast-rate scannable Advertisers (in addition to multiple of these that appear in the normal office environment) and this worked fine.

I will continue to test, but unless the problem reappears, this issue can be closed.

[57496_hci_usb_active_scan.log](https://github.com/zephyrproject-rtos/zephyr/files/11392041/57496_hci_usb_active_scan.log)

carlescufi, 2023-05-04:
> I will continue to test, but unless the problem reappears, this issue can be closed.

Thanks for coming back to us. Would you please update this issue once you've had a chance to test further?

carlst, 2023-05-04:
@cvinayak I let hci_usb run overnight without problems. Further, one reduced stack size was used for that testing:

    CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y
    CONFIG_BT_HCI_TX_STACK_SIZE=768

Since the issue cannot be reproduced, I am closing it.
Thank you again for the advice and information.

closed 2023-05-04 by carlst