zephyrproject-rtos/zephyr #57499: STM32H743 USB (High Speed) CDC ACM Can't receive data

cbrake, 2023-05-02:
**Describe the bug**

We have a custom STM32H743 board configured for High-Speed USB CDC ACM Uart. The device will enumerate and we can receive data from the devices, but can't send data. 

We have tried it on a NUCLEO-H743ZI2 (full speed) and it works fine there, so it may be related to the high speed with external ULPI PHY. 

The Custom High-Speed USB board does work with TinyUSB running under FreeRTOS, so the hardware has been proven with that software. We are in the process of transitioning from FreeRTOS to Zephyr.

**To Reproduce**

Steps to reproduce the behavior:

We ran the `samples/subsys/usb/cdc_acm` on the custom board and it does not echo data back. We have modified the test to send some data from the device to the host and that works fine, but it does not receive data.

We never see the UART (USB CDC ACM) callback functions called on the device to indicate data is received.

**Expected behavior**

We need data flowing from host -> device.

**Impact**

What impact does this issue have on your progress: showstopper

**Logs and console output**

Will post USB and other traces in additional comments.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: ef3d01d16fd462805565d83443db7a7bd38d306d (recent Zephyr HEAD)

**Additional context**

See DTS and config:

[custom_board_conifg.zip](https://github.com/zephyrproject-rtos/zephyr/files/11378069/custom_board_conifg.zip)

cbrake, 2023-05-02:
The device enumerates fine. Below is a comparison of the Zephyr device (mini) compared to a working Arduino device:

[image]

[image]

cbrake, 2023-05-02:
We enabled debug messages in the Linux driver and see the same messages:

`echo "module cdc_acm +p" > /proc/dynamic_debug/control`

# Arduino board connected:

[image]

# Custom STM32H743 board connected:

[image]

cbrake, 2023-05-02:
Here is a log on the zephyr device side. We are transmitting a short message every 10 minutes.

```
*** Booting Zephyr OS build zephyr-v3.3.0-3366-gd433ec25bf5c ***
Set up button at gpio@58022400 pin 3
CDC Uart ready
[00:00:00.558,000] <inf> z_comm: Wait for DTR
[00:00:00.704,000] <dbg> usb_cdc_acm: cdc_acm_dev_status_cb: cfg 0x240017d0 status 1
[00:00:00.704,000] <dbg> usb_cdc_acm: cdc_acm_do_cb: Device reset detected
[00:00:00.765,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
--- 9 messages dropped ---
[00:00:00.765,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0000
[00:00:00.765,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.765,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
--- 16 messages dropped ---
[00:00:00.765,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:00.776,000] <dbg> usb_cdc_acm: cdc_acm_dev_status_cb: cfg 0x240017d0 status 1
[00:00:00.776,000] <dbg> usb_cdc_acm: cdc_acm_do_cb: Device reset detected
[00:00:00.838,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.838,000] <dbg> usb_device: custom_handler: bRequest 0x05, wIndex 0x0000
[00:00:00.838,000] <dbg> usb_device: usb_handle_std_device_req: Set Address 3 request
[00:00:00.838,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.855,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.855,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0000
[00:00:00.856,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0000
[00:00:00.856,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.856,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.858,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:00.858,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.858,000] <dbg> usb_device: custom_handler: bRequest 0x06, wIndex 0x0409
[00:00:00.858,000] <dbg> usb_device: usb_get_descriptor: Get Descriptor request
[00:00:00.858,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.858,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:00.900,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.900,000] <dbg> usb_device: custom_handler: bRequest 0x09, wIndex 0x0000
[00:00:00.900,000] <dbg> usb_device: usb_set_configuration: Set Configuration 1 request
[00:00:00.900,000] <dbg> usb_device: set_endpoint: Set endpoint 0x81 type 3 MPS 16
[00:00:00.900,000] <dbg> usb_device: set_endpoint: Set endpoint 0x82 type 2 MPS 512
[00:00:00.900,000] <dbg> usb_device: set_endpoint: Set endpoint 0x1 type 2 MPS 512
[00:00:00.900,000] <dbg> usb_cdc_acm: cdc_acm_dev_status_cb: cfg 0x240017d0 status 3
[00:00:00.900,000] <inf> usb_cdc_acm: Device configured
[00:00:00.900,000] <dbg> usb_cdc_acm: cdc_acm_read_cb: ep 1 size 0 dev_data 0x24000840 rx_ringbuf space 1024
[00:00:00.900,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: Transfer start, ep 0x%02x, data %p, dlen %zd" argument:2
[00:00:00.900,000] <dbg> usb_transfer: usb_transfer: Transfer start, ep 0x01, data 0x2400088c, dlen 512
[00:00:00.900,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:00.901,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:00.901,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:00.901,000] <dbg> usb_device: class_handler: bRequest 0x20, wIndex 0x0000
[00:00:00.901,000] <dbg> usb_cdc_acm: cdc_acm_class_handle_req: CDC_SET_LINE_CODING 9600 0 0 8
[00:00:00.901,000] <dbg> usb_device: usb_handle_control_transfer: >> usb_data_to_host(2)
[00:00:00.901,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:23.565,000] <dbg> usb_device: class_handler: bRequest 0x22, wIndex 0x0000
[00:00:23.565,000] <dbg> usb_cdc_acm: cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE 0x3
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:23.565,000] <dbg> usb_device: class_handler: bRequest 0x20, wIndex 0x0000
[00:00:23.565,000] <dbg> usb_cdc_acm: cdc_acm_class_handle_req: CDC_SET_LINE_CODING 115200 0 0 8
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: >> usb_data_to_host(2)
[00:00:23.565,000] <dbg> usb_device: usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:23.581,000] <inf> z_comm: DTR set
[00:00:23.619,000] <dbg> usb_cdc_acm: cdc_acm_int_in: CDC_IntIN EP[81]
[00:00:23.683,000] <dbg> usb_cdc_acm: cdc_acm_int_in: CDC_IntIN EP[81]
[00:00:23.783,000] <inf> z_comm: Baudrate detected: 115200
[00:00:23.783,000] <dbg> usb_cdc_acm: cdc_acm_fifo_fill: dev_data 0x24000840 len 11 tx_ringbuf space 1024
[00:00:23.783,000] <dbg> z_comm: z_comm_thread: uart_fifo_fill sent 11 bytes
[00:00:23.783,000] <dbg> usb_cdc_acm: tx_work_handler: Got 11 bytes from ringbuffer send to ep 82
[... cut ...]
[00:00:33.783,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: Transfer start, ep 0x%02x, data %p, dlen %zd" argument:2
[00:00:33.783,000] <dbg> usb_transfer: usb_transfer: Transfer start, ep 0x82, data 0x2400d563, dlen 11
[00:00:33.783,000] <dbg> usb_transfer: usb_transfer_work: Transfer done, ep 0x82, status 0, size 11
[00:00:33.783,000] <dbg> usb_cdc_acm: cdc_acm_write_cb: ep 82: written 11 bytes dev_data 0x24000840
[00:00:33.783,000] <dbg> usb_cdc_acm: cdc_acm_write_cb: tx_ringbuf is empty

```

cbrake, 2023-05-02:
Current test code. As mentioned above, we also tested `samples/subsys/usb/cdc_acm`.

``` c
#include "zephyr/sys/time_units.h"
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/drivers/uart.h>
#include <zephyr/logging/log.h>
#include <zephyr/sys/ring_buffer.h>
#include <stdio.h>

LOG_MODULE_REGISTER(z_comm, LOG_LEVEL_DBG);

/* size of stack area used by each thread */
#define STACKSIZE 1024

/* scheduling priority used by each thread */
#define PRIORITY 7

// static const struct device *dev = DEVICE_DT_GET(DT_NODELABEL(cdc_acm_uart0));

#define RING_BUF_SIZE 1024
uint8_t ring_buffer[RING_BUF_SIZE];

struct ring_buf ringbuf;

static void interrupt_handler(const struct device *dev, void *user_data)
{
	ARG_UNUSED(user_data);

	LOG_DBG("CLIFF: interrupt_handler");

	while (uart_irq_update(dev) && uart_irq_is_pending(dev)) {
		if (uart_irq_rx_ready(dev)) {
			int recv_len, rb_len;
			uint8_t buffer[64];
			size_t len = MIN(ring_buf_space_get(&ringbuf), sizeof(buffer));

			recv_len = uart_fifo_read(dev, buffer, len);
			if (recv_len < 0) {
				LOG_ERR("Failed to read UART FIFO");
				recv_len = 0;
			};

			rb_len = ring_buf_put(&ringbuf, buffer, recv_len);
			if (rb_len < recv_len) {
				LOG_ERR("Drop %u bytes", recv_len - rb_len);
			}

			LOG_DBG("tty fifo -> ringbuf %d bytes", rb_len);
			if (rb_len) {
				uart_irq_tx_enable(dev);
			}
		}

		if (uart_irq_tx_ready(dev)) {
			uint8_t buffer[64];
			int rb_len, send_len;

			rb_len = ring_buf_get(&ringbuf, buffer, sizeof(buffer));
			if (!rb_len) {
				LOG_DBG("Ring buffer empty, disable TX IRQ");
				uart_irq_tx_disable(dev);
				continue;
			}

			send_len = uart_fifo_fill(dev, buffer, rb_len);
			if (send_len < rb_len) {
				LOG_ERR("Drop %d bytes", rb_len - send_len);
			}

			LOG_DBG("ringbuf -> tty fifo %d bytes", send_len);
		}
	}
}

void z_comm_thread(void *arg1, void *arg2, void *arg3)
{
	int ret;
	uint32_t baudrate, dtr = 0U;
	const char testBuffer[] = "hi there\r\n";
	printf("z comm thread\n");
	const struct device *dev;
	dev = DEVICE_DT_GET_ONE(zephyr_cdc_acm_uart);

	while (1) {
		if (!device_is_ready(dev)) {
			printf("waiting for CDC device to be ready\n");
		} else {
			break;
		}
		k_msleep(1000);
	}

	printk("CDC Uart ready\n");

	ring_buf_init(&ringbuf, sizeof(ring_buffer), ring_buffer);

	LOG_INF("Wait for DTR");
	while (true) {
		uart_line_ctrl_get(dev, UART_LINE_CTRL_DTR, &dtr);
		if (dtr) {
			break;
		} else {
			// Give CPU resources to low priority threads.
			k_sleep(K_MSEC(100));
		}
	}

	LOG_INF("DTR set");

	/* They are optional, we use them to test the interrupt endpoint */
	ret = uart_line_ctrl_set(dev, UART_LINE_CTRL_DCD, 1);
	if (ret) {
		LOG_WRN("Failed to set DCD, ret code %d", ret);
	}

	ret = uart_line_ctrl_set(dev, UART_LINE_CTRL_DSR, 1);
	if (ret) {
		LOG_WRN("Failed to set DSR, ret code %d", ret);
	}

	/* Wait 100ms for the host to do all settings */
	k_msleep(100);

	ret = uart_line_ctrl_get(dev, UART_LINE_CTRL_BAUD_RATE, &baudrate);
	if (ret) {
		LOG_WRN("Failed to get baudrate, ret code %d", ret);
	} else {
		LOG_INF("Baudrate detected: %d", baudrate);
	}

	uart_irq_callback_set(dev, interrupt_handler);

	/* Enable rx interrupts */
	uart_irq_rx_enable(dev);

	while (1) {
		// does not work
		// int ret = usb_write(cdc_uart_dev, testBuffer, sizeof(testBuffer), NULL);
		// printf("CLIFF: usb_write returned %i\n", ret);

		// works
		int send_len = uart_fifo_fill(dev, testBuffer, sizeof(testBuffer));
		LOG_DBG("uart_fifo_fill sent %d bytes", send_len);

		k_msleep(10000);
	}
}

K_THREAD_DEFINE(z_comm, STACKSIZE, z_comm_thread, NULL, NULL, NULL, PRIORITY, 0, 0);

```

cbrake, 2023-05-02:
Here is a USB trace. Notes, we see the CDC IN Data which is the "hi there" message I'm printing from the above application. 

[image]

I see a 0-byte OUT packet once, but nothing else when I type in the terminal on the Linux host.

[image]

cbrake, 2023-05-03:
@jfischer-no thanks so much -- we are now getting rx callbacks on the device after I make that change.

jfischer-no, 2023-05-03:
> @jfischer-no thanks so much -- we are now getting rx callbacks on the device after I make that change.

See also @loicpoulain  comment https://github.com/zephyrproject-rtos/zephyr/issues/54162#issuecomment-1432158791, perhaps a simpler workaround is possible by changing the number of endpoints in your board's devicetree.

cbrake, 2023-06-07:
I'm also running into a write problem where if I write more than 200 bytes at a time, the system gets stuck in a USB IRQ handler. This is the current code that works:

``` c
                if (uart_irq_tx_ready(dev)) {
                        if (tx_buf_len <= 0) {
                                uart_irq_tx_disable(dev);
                                k_sem_give(&tx_sem);
                                // LOG_DBG("Z USB tx sent %d bytes", tx_buf_sent);
                                continue;
                        }

                        int cnt = MIN(tx_buf_len, 200);
                        int sent_bytes_cnt = uart_fifo_fill(dev, &tx_buf[tx_buf_sent], cnt);
                        tx_buf_sent += sent_bytes_cnt;
                        tx_buf_len -= sent_bytes_cnt;
                }
```

I've not tried @jfischer-no 's suggestion to change the number of endpoints yet.

cbrake, 2023-08-07:
I think this is still an issue as we have been running with a local change. I'm refreshing to latest Zephyr now and will retest. We also having some other issues that I will post more data on soon.

loicpoulain, 2023-08-22:
Wondering if you tested the new usb stack wich should be now supported with cdc_acm.

cbrake, 2023-08-22:
Updated information -- I'm seeing differences in connect behavior and number of errors based on different hardware versions and cables, so I suspect we are fighting a hardware issue with the latest revision. 

Regarding the new USB stack, no, currently using the classic stack. Have you used the new stack on STM32 devices? What are the advantages of using the new stack, or what is it trying to do better?

I tried using the new/drivers stack, and now getting this on the host:

```
[1742974.122035] usb 3-6: reset high-speed USB device number 3 using xhci_hcd
[1742974.570259] usb 3-6.2: new high-speed USB device number 29 using xhci_hcd
[1742979.736905] usb 3-6.2: device descriptor read/64, error -110
[1742995.306826] usb 3-6.2: device descriptor read/64, error -110
[1742995.490141] usb 3-6.2: new high-speed USB device number 30 using xhci_hcd
[1743000.640138] usb 3-6.2: device descriptor read/64, error -110
[1743016.213378] usb 3-6.2: device descriptor read/64, error -110
[1743016.320370] usb 3-6-port2: attempt power cycle
[1743016.913357] usb 3-6.2: new high-speed USB device number 31 using xhci_hcd
[1743022.113357] xhci_hcd 0000:2a:00.3: Timeout while waiting for setup device command
[1743027.449983] xhci_hcd 0000:2a:00.3: Timeout while waiting for setup device command
[1743027.656629] usb 3-6.2: device not accepting address 31, error -62
```

So probably have a  little more configuration work to do ...

I've been thinking of getting a STM32H7 board with high-speed USB for testing -- is the STM32H747I-DISCO a good option, or is there a better one?

loicpoulain, 2023-08-22:
No strong requirement to move to new device stack for your specific use case, but te legacy stack will probably be replaced at some point.

I have a STM32H723ZG on my side that I have tested with the new stack (I've not checked if it's same phy as yours), though I don't remember if I tested high speed mode. I will give it a try (probably early next week) and will try to help you on that.

Meanwhile yeah it would be good to test with same MCU based disco dev board. 
Also Maybe @erwango has STM32H747I-DISCO to test on and check if issue is reproduced.

erwango, 2023-08-23:
@Desvauxm-st Can you have a look ?

cbrake, 2023-08-23:
Thanks for the help!

STM32H747I-DISCO should arrive today -- I'll start with a clean Zephyr tree and re-verify all the issues in this thread and report back.

cbrake, 2023-08-24:
[STM32H747I-DISCO](https://www.st.com/en/evaluation-tools/stm32h747i-disco.html) set up and tested with cdc_acm example with latest Zephyr main:

`west build -b stm32h747i_disco_m7 samples/subsys/usb/cdc_acm && west flash`

The target does not receive any data from the host unless I apply the following patch, so it appears this is still an issue:

``` patch
commit 6e94847974671766d6ab549dc3e844faa9b5b167 (HEAD -> upstream-main)
Author: Cliff Brake <cbrake@bec-systems.com>
Date:   Thu May 4 12:26:17 2023 -0400

    fix for USB Rx problem

diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c
index 6fd0a3a833..7e9db5e913 100644
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -437,7 +437,11 @@ static int usb_dc_stm32_init(void)
        }
 #else /* USB_OTG_FS */
        /* TODO: make this dynamic (depending usage) */
+#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32_otghs) // USBOTG_HS
+       HAL_PCDEx_SetRxFiFo(&usb_dc_stm32_state.pcd, 0x200);
+#else // USBOTG_FS
        HAL_PCDEx_SetRxFiFo(&usb_dc_stm32_state.pcd, FIFO_EP_WORDS);
+#endif
        for (i = 0U; i < USB_NUM_BIDIR_ENDPOINTS; i++) {
                HAL_PCDEx_SetTxFiFo(&usb_dc_stm32_state.pcd, i,
                                    FIFO_EP_WORDS);
```

jfischer-no, 2023-08-30:
From https://github.com/zephyrproject-rtos/zephyr/issues/54162#issuecomment-1432158791
>Good catch ! But we can't change the ram size, it's not main memory but a dedicated 4kb ram for USB HS. Ideally we need to better adjust the allocated memory for the endpoints instead of just splitting it equally between all (possibly unused) endpoints. We can e.g limit fifo 0 (ep0) to 64 bytes.

@loicpoulain would it not be fixed by :arrow_up: ? I would not refer users to a new stack while it is still experimental. This bug needs to be fixed or mitigated.

Desvauxm-st, 2023-09-06:
@cbrake

hello

I try to check the modifications before  in the Board STM32H747i-Disco
 
1: without the modifications in drivers/usb/device/usb_dc_stm32.c:

In the console (COM44) the other USB in COM49 for me in my laptop is detected :

Device configured , DTR set , Baudrate detected: 115200
 
[image]
 

But in COM49 when i enter 12345 no echo are received==> not OK.
[image]

 

2: with the modifications you propose in drivers/usb/device/usb_dc_stm32.c line 437
[image]

 

In the console (COM44) the other USB in COM49 for me in my laptop is detected :

Device configured , DTR set , Baudrate detected: 115200
[image]

 

And now in COM49 when i enter 12345  echo are received : OK.
[image]

see after resume test in attachement:
[USB High Speed CDC ACM Can't receive data.docx](https://github.com/zephyrproject-rtos/zephyr/files/12538302/USB.High.Speed.CDC.ACM.Can.t.receive.data.docx)

Desvauxm-st, 2023-09-06:
when you call HAL_PCDEx_SetRxFiFo(&usb_dc_stm32_state.pcd, value);
the minimum into value is 160  ,lower than 160  the echo doesn't work

Desvauxm-st, 2023-09-12:
@cbrake
hi
Can you see the PR proposal to check :
drivers: usb: device: fix Rx FIFO min size #62530

erwango, 2023-09-25:
@cbrake Did you had time to try https://github.com/zephyrproject-rtos/zephyr/pull/62530 ?

cbrake, 2023-09-25:
No, not yet -- but planning to work on it this week -- will report back soon. Thanks!

I've been working on a test app: 

https://github.com/zephyrproject-rtos/zephyr/pull/62945

Still needs some cleanup, but has been useful for testing high throughput scenarios.

cbrake, 2023-09-28:
@erwango tested #62530 and it looks good with the new samples/subsys/usb/cdc_acm_perf/CMakeLists.txt (#62945) performance test.

@Desvauxm-st thanks for the PR!

Without the current zephyr main, I could not even get the cdc_acm echo test to work with the disco (high speed). It worked fine with Nucleo (full speed). I then applied #62530 and both the echo and perf samples ran fine.

I still have an issue with the new cdc_acm_perf test where I get occasional errors -- not sure if we are dropping a byte occasionally, or there is a transfer send that is not aligned to 4 bytes, as I'm sending incrementing uint32 values as a test payload.

Next, I'll test with our custom application ...

```
Read 5461592 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461416 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461192 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461392 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461616 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5460992 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461392 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461416 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461592 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461192 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461416 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461392 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461216 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461392 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
Read 5461392 bytes, Data rate: 43.69 Mb/sec, error cnt: 9016, leftover cnt: 12
```

cbrake, 2023-10-03:
One thing I have noticed is that I can only write about 200 bytes at a time to the uart fifo:

With the full speed (nucleo), I get tons of errors with the test app.

With high speed (disco), the MCU locks up.

https://github.com/cbrake/zephyr/blob/feature-sample-usb-cdc-amc-perf/samples/subsys/usb/cdc_acm_perf/src/main.c#L126

Is this expected?

cbrake, 2023-10-03:
> Next, I'll test with our custom application ...

testing with our custom application is going well (with the 200-byte write limit).

erwango, 2023-12-04:
@cbrake https://github.com/zephyrproject-rtos/zephyr/pull/62530 has been merged. Can we close current issue ?

erwango, 2024-03-18:
CLosing as #62530 is merged since some time now. Feel free to re-open if required

closed 2024-03-18 by erwango

cbrake, 2024-03-18:
Sorry I missed your question -- thanks to all for getting this merged.