zephyrproject-rtos/zephyr #57512: Qemu emulation: kernel time and hw timers are not synchronized

erlingrj, 2023-05-03:
**Describe the bug**
I am experiencing strange behaviour when using the Counter device with `qemu_cortex_m0`.  I have experienced two, possible related, bugs.

1. Kernel time and Counter device is not advancing at correct rate.
The kernel ticks are supposedly with 1MHz and the Counter device at 16MHz, still I am only seeing a 8x different in advanced time as measured with a program like this:

```
uint32_t k_now, c_now, k_now1, c_now1;
k_now = k_cycle_get_32();
res = counter_get_value(counter_dev, &c_now);
printk("start: Counter=%u, Kernel=%u\n", c_now, k_now);

k_now1 = k_cycle_get_32();
res = counter_get_value(counter_dev, &c_now1);
printk("Counter=%u Kernel=%u rate=%u\n", c_now1, k_now1, (c_now1-c_now)/(k_now1- k_now));
```

2. Counter device is not emulated properly when the system is sleeping.
I am running a very simple program where a callback is registered with overflows of the counter which prints the kernel time of that event:

```
uint32_t prev_overflow=0;
static void  overflow(const struct device *dev, void *user_data) {
    uint32_t now = k_cycle_get_32();
    printk("Overflow @ %u Delta=%u\n", now/1000, now-prev_overflow);
    prev_overflow=now;
}

int main() {
    // Configure an overflow callback
    top_cfg.callback = overflow;
    top_cfg.ticks = counter_get_max_top_value(counter_dev);
    res = counter_set_top_value(counter_dev, &top_cfg);
    if (res != 0) {
        printk("error");
        return 1;
    }
    
    // Start timer
    counter_start(counter_dev);
    while(true) {} 
    return 0;
}
```

The overflow callback is called regularily every 8ms (measured using `k_cycle_get_32`).  This is wrong, but consistent with the what we observed in (1). The simulation speed is also reasonably fast.

However, if I insert a sleep in that while loop I get really strange behaviour.  By adding a small sleep of 1ms I know only get overflows invoked every 20 ms. If I increase the sleep til 100ms I get 110ms between each overflow.  If I go to sleep for `K_FOREVER` then we get no overflows at all. Also, the simulation speed drops by maybe 10x

It seems as if the emulated Timer peripheral is not synchronized to the kernel time when we go to sleep. Or maybe the interrupts from the Timer are, for some reason, dropped when we enter sleep.

carlescufi, 2023-05-10:
This should not be assigned to @anangl. Leaving it unassigned for now.

nashif, 2023-05-10:
@carlescufi it was assigned to @anangl  because this qemu uses nrf timer AFAIK.

carlescufi, 2023-05-31:
This part here needs investigating:
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/qemu_cortex_m0/nrf_timer_timer.c#L238-L239

carlocaione, 2023-05-31:
@erlingrj can you paste the whole reproducer you are using for the first bug?

carlocaione, 2023-06-01:
@erlingrj can you try again with the fix in the https://github.com/zephyrproject-rtos/zephyr/pull/58753 PR?

I cannot precisely reproduce your use case because I'm not entirely clear which counter you are using given that by default the `qemu_cortex_m0` board is exposing none.

microbuilder, 2023-06-01:
> This part here needs investigating: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/qemu_cortex_m0/nrf_timer_timer.c#L238-L239

@carlescufi  `qemu_cortex_m0` looks to be using the nRF51822 for emulation beneath the hood ... is someone at Nordic actively maintaining that upstream in QEMU to maybe give some context to the 1MHz comment?

carlocaione, 2023-07-05:
This was hopefully fixed by #58753