zephyrproject-rtos/zephyr #57516: Safety development concept (for IEC61508 certification)

romkell, 2023-05-03:
## Introduction

The quality and finally safety (IEC61508 SIL3 certification) and
security development of Zephyr OS requires
- to follow a or several processes 
- to have a tool environment installed to complete tasks and support the processes.

This RFC defines a rough concept how the safety committee intends to setup, both,
tools and processes to do safety development.

Serveral areas, in example requirements management, are not restricted
to safety and shall be setup at the best of our current knowledge to suite
for the full extend of the ecosystem.

Nevertheless, in order to be able to finish with the safety certifcation task,
it must be accepted that the focus of the safety committee and WG and therefore
this RFC is safety.

Please feel invited
- to suggest improvement
- to disclose gaps
- to ask questions when unclear
- to state your opinion

### Problem description

An overall concept how to do safety is not (fully) defined, described and known to the safety WG
and TSC.
It is necessary to achieve a common understanding.

### Proposed change

Create an overall concept by defining processes and a tool environment that allows
to get 'main' in a 'safety certifiable ready' state of which an auditable can
be branched with basically no gaps to reach certification.
Doing safety on auditable and forward merge all code changes, all tests, requirements etc.
has shown to not work.

We start with an rough concept and iteratively detail and implement the different areas.

## Detailed RFC

#### Proposed change (Detailed)

#### Overview

[image]

#### Requirements Management

For safety (IEC61508) the following specifications need to be written:

##### Requirements specifications
- User Story (US) - not requested by 61508 but makes requirements more understandable
- Software Requirements Specification (SRS)
- Software Architecture & Interface Specification (SAIS)
- Software Module Design Specification (SMDS)			

Requirements management(SRS, SAIS, STS) is done using
- *Doorstop* (https://doorstop.readthedocs.io/en/latest/,https://github.com/doorstop-dev/doorstop)
- *Doorhole* (https://github.com/sevendays/doorhole) may serve as a GUI frontend for convenience

Storage location is zephyrproject-rtos/reqmgmt repo (https://github.com/zephyrproject-rtos/reqmgmt).

##### Test specifications
- Software Test Specification (STS)
- Software Integration Test Specification (SITS)
- Software Module Test Specification (SMTS)       			

Remark: possibly replace term 'module' as in 61508 by 'component' as in 50128 or 'unit' as in unit test for SMDS and SMTS, since easily confused with git (sub-)module.
[... 144 lines cut ...]
	   - /sched_queue
     - /queue
       - /_m_a_test
       - /_m_u_test
     - /semaphore
       - /_m_a_test
       - /_m_u_test
     - /sched
       - /_m_a_test (former schedule_api test?)
       - /_m_u_test
   

### Dependencies

Introducing a system for quality to achieve a safety certification will introduce changes in various areas.
It has to be an iterative step by step process in which the direction very likely will have to be corrected
now and then.

### Concerns and Unresolved Questions

Is part of this RFC's discussion to adjust the concept.

## Alternatives

Is part of this RFC's discussion to adjust the concept.

nicpappler, 2023-05-04:
That's a lot to discuss in a single RFC - how about we split this into several sub RFCs?

gregshue, 2023-05-04:
> @gregshue - regarding the security and safety claims - as already discussed, there are no security claims attached to the safety claims. Both safety and security need a certain level of systematic capability. We aim for the systematic capability of SIL 3 (IEC 61508:2010), that also is enough for security, but does not make specific claims regarding security features, mechanisms, etc.

I repeated the question here to capture the topic within this issue. The question originated from the following statement in Embedded Software Development for Safety-Critical Systems, Second Edition (Chris Hobbs, CRC Press, 2020), p 10,

> Safety is interlinked with security - the behavior of an insecure device is effectively unpredictable once a hacker has found a vulnerability.

This may be another subject appropriate for a topic RFC.

RobertoBagnara, 2023-05-04:
Regarding the choice of Doorstop, I think it is not a good one,
for the following reasons.

Doorstop is arguably unmaintained – its creator only sends Python
support patches from time to time. For any advanced patches from other
people, there are long unfinished threads hanging around with no
progress being made.

About "Doorstop uses single yaml files per requirement", this is not
so good granularity, if one considers that everything in the industry
is kept in documents (requirement specifications, test specifications,
verification control document, etc.). StrictDoc intentionally packages
requirements in .sdoc document files because people create and read
the whole documents and requirement sections. Maintaining separate
files without first-class document support is cumbersome. Take the
"level:" field for example
https://github.com/doorstop-dev/doorstop/blob/develop/reqs/REQ002.yml#LL4
– it is so inconvenient to manage it on the single req's YAML file
level.

Merge conflict argument works a little better with Doorstop, but I could
argue that if there is a conflict, this already means that the engineers
working on requirements went out of sync. To get a merge conflict in an
.sdoc file, you really have to work on the same or closely neighboring
requirements or someone has to start moving sections around, which
already indicates that people work without any synchronization.

Additionally, StrictDoc has made and is making a lot of progress recently:
see the GitHub project and compare it with the one for DoorStop.
In particular, StrictDoc has at least an 80%-working bridge to ReqIF export
and import.  I personally used it to successfully import Polarion-generated
requirements in ReqIF format. Doorstop doesn't have ReqIF support.

RobertoBagnara, 2023-05-04:
The following sentence misses one point: "Even if the open source SCA does not have the same quality as a commercial grade solutions, it will be suitable to detect and this way fix 80 - 90% of the violations in a fast development cycle."
It is not just that cppcheck has lots of false negatives, it also has lots of false positives.  The danger is that developers that have not been properly trained on the coding guidelines will not be able to recognize the false positives and will believe they have to to something about them.  In my experience, this always causes a decrease in code quality.

gregshue, 2023-05-04:
We need a requirements management system keeping with the composability and extensibility of Zephyr.

1. It needs to support a single "document" (e.g., StRS, SyRS, SyAID, SRS, SAID, etc.) that has requirements spread across multiple repositories in the workspace.
2. It needs to support "variation management" - tracking moderately different requirement sets for closely related products.

I have attempted to use Doorstop and found it at least unable to link to requirements in other repositories. I didn't look into variation management or aggregating requirements from multiple repositories.

 @RobertoBagnara How well does StrictDoc support those needs?

sebhub, 2023-05-05:
Zephyr is not the first project for which a pre-qualification according to safety related standards is available. We did something similar for the [RTEMS]( https://www.rtems.org/) real-time operating system. The pre-qualification for RTEMS was performed using software development standards from the [European Cooperation for Space Standardization](https://ecss.nl/): ECSS-E-ST-40C and ECSS-Q-ST-80C Rev.1. We took also other standards such as IEC 61508,
ISO 26262, DO-178, DO-330, DO-333, and Galileo Software Standard (GSWS) into account. The work started in 2020 and the first results are available on a ESA website: https://rtems-qual.io.esa.int/. The [technical note QT-109](https://ftp.rtems.org/pub/rtems/people/sebh/QT-109.pdf) contains some background information. The latest documentation of the RTEMS pre-qualification process is available here: https://docs.rtems.org/branches/master/eng/req/index.html. A key observation is that the standards require a high degree of traceability from requirements to source code and tests and vice versa.

Interestingly, the proposal here has some overlaps with what we did. I can highly recommend the EARS syntax for requirements.

We also use Doxygen to make the link from the specification to the source code and vice versa. In a first pass, Doxygen is invoked to produce a tagfile. This tagfile is used to link requirements to source code. Then a file is generated which contains Doxygen comments for source code elements such as functions. The generated comments augment the existing Doxygen documentation and provide links to the specification.

For the requirements (or specification in general) we also started with Doorstop. However, while working with it, it turned out that some customization was required: https://docs.rtems.org/branches/master/eng/req/tooling.html. The RTEMS specification contains not just requirements, so we added a hierarchical type system for specification items: https://docs.rtems.org/branches/master/eng/req/items.html. These types are also specified by specification items: https://github.com/RTEMS/rtems-central/tree/master/spec/spec. It can be easily customized to meet project-specific requirements.

Real-time operating systems such as Zephyr or RTEMS are just a collection of state machines and data structures. For RTEMS the majority of the function specification is done by the specification of state machines. We use a specialized specification item for this: https://docs.rtems.org/branches/master/eng/req/howto.html#action-requirements. It contains the specification along with the corresponding validation test code. It needs a bit of learning to be able to write these files (example: https://github.com/RTEMS/rtems-central/blob/master/spec/c/req/posix-memalign.yml), however, it turned out to be very useful for the functional specification of RTEMS. We reach 100% branch and code coverage for the pre-qualified feature set of RTEMS through validation tests. Since we enumerate all state transitions, this yields more or less MC/DC coverage automatically.

Performance requirements are stated in terms of runtime measurement requests. The measurements are done using the [RTEMS Test Framework](https://docs.rtems.org/branches/master/eng/test-framework.html). This test framework would also work with Zephyr.

We use Python modules for the tooling (BSD-2-Clause license): https://github.com/RTEMS/rtems-central/tree/master/rtemsspec.

Just let me know if you are interested in this work. Maybe it can be generalized so that it can be used also for Zephyr.

romkell, 2023-05-05:
@sebhub 
> Zephyr is not the first project for which a pre-qualification according to safety related standards is available. We did something similar for the [RTEMS](https://www.rtems.org/) real-time operating system.

Many thanks for reaching out to us.
We are aware of RTEMS in general, basically due to your evaluation of Doorstop and your decision to implement your own RQM tool for various reasons.
At least I have not looked into your detailed implementation and requirements of RTEMS, but appreciate your input.

The Zephyr OS code is existing and rewriting everything to yield MC/DC coverage will I assume effort-wise not work at least in the first place. It also is not demanded by IEC61508 SIL3, which we aim for.
Let me have a look at all the links you provided.

romkell, 2023-05-05:
@RobertoBagnara 
> It is not just that cppcheck has lots of false negatives, it also has lots of false positives. The danger is that developers that have not been properly trained on the coding guidelines will not be [...]

The final check is the SCA in the CI that matters. 
How do you suggest to provide fast SCA checks on a local developer machine with short cycle times with free access to the community?
Or how do you suggest the community shall do SCA runs? Just via CI?

RobertoBagnara, 2023-05-05:
> We need a requirements management system keeping with the composability and extensibility of Zephyr.
>
>     1. It needs to support a single "document" (e.g., StRS, SyRS, SyAID, SRS, SAID, etc.) that [...]

TL;DR: Currently, StrictDoc does not support this, but it could. In my experience, developers are very keen on receiving feedback and making StrictDoc more useful.  Rather, what is needed is a more detailed specification of how the feature-based requirements can be filtered.

Longer version:

The filtering has to happen based on some criteria.  Specific feature-based requirements are either:

- Placed in a specific document, like features/ABC.sdoc
- Have a specific field like FEATURE: ABC (makes most sense?)
- Tagged with a given tag, such as #ABC
- Have a specific requirement UID, like REQ-...-ABC-...
- Placed in a specific section.

Once the convention is established, one can think of implementing a query mechanism that takes a given document, filters in/out given requirements and then the resulting document is a needed subset.  We did implement a similar mechanism in a day.

This feature definitely requires a good specification and a good implementation, so that it is not hard-coded to only Zephyr community as a single user.

RobertoBagnara, 2023-05-05:
> @RobertoBagnara
>
>
> The final check is the SCA in the CI that matters.

If developers plagues by false positives worsen code quality, the final check will come too late.

> How do you suggest to provide fast SCA checks on a local developer machine with short cycle times with free access to the community? Or how do you suggest the community shall do SCA runs? Just via CI?

As far as I know, no free tool has half-decent support of the MISRA coding standards.  Said that, yes, a CI infrastructure allowing developers to analyze their own branches/forks/whatever could do the job.

gregshue, 2023-05-05:
> The filtering has to happen based on some criteria.

In the EARS syntax optional feature requirements are captured with the "Where <feature is included>," fragment. This condition can be expressed as either a series of Where conditions (implicitly all must be true), or as a single condition with explicit logical "and" conjunctions.

sebhub, 2023-05-05:
> We need a requirements management system keeping with the composability and extensibility of Zephyr.
> 
>     1. It needs to support a single "document" (e.g., StRS, SyRS, SyAID, SRS, SAID, etc.) that has requirements spread across multiple repositories in the workspace.

The RTEMS specification graph can be built using from any number of directories, for example:
* https://github.com/RTEMS/rtems-central/tree/master/spec-spec
* https://github.com/RTEMS/rtems-central/tree/master/spec-glossary
* https://github.com/RTEMS/rtems-central/tree/master/spec
* https://github.com/RTEMS/rtems/tree/master/spec

The tools visit a configurable list of base directories and search recursively for specification items. The items are added to an item cache which associates an UID with exactly one item. The UID relates to the path of the file relative to the base directory. Specification item are stored in YAML or JSON files.

> 
>     2. It needs to support "variation management" - tracking moderately different requirement sets for closely related products.

For RTEMS we solved this by adding an [enabled-by](https://docs.rtems.org/branches/master/eng/req/items.html#spectypeenabledbyexpression)  attribute to each specification item. For example, some requirements are restricted to SMP configurations: https://github.com/RTEMS/rtems-central/blob/master/spec/rtems/scheduler/req/get-processor-smp-only.yml. Other requirements depend on a particular SoC: https://github.com/RTEMS/rtems-central/blob/master/spec/bsp/sparc/leon3/req/errata-gr712rc-05.yml. These enabled-by attributes can be viewed as *Where* clauses in EARS.

gregshue, 2023-05-05:
@sebhub Thank you for the pointers. It looks to be very relevant content.

> The RTEMS specification graph can be built using any number of directories, ...

From the [RTEMS UID definitions](https://docs.rtems.org/branches/master/eng/req/req-for-req.html) it looks like RTEMS requires a (all?) specification(s) to be within a consistent, fixed directory tree. The Zephyr build system learns about the Zephyr module pathnames through the `ZEPHYR_MODULES` and `ZEPHYR_EXTRA_MODULES` variables. Any idea about the complexity of inserting a module reference parameter into the UIDs?

stanislaw, 2023-05-07:
> It needs to support a single "document" (e.g., StRS, SyRS, SyAID, SRS, SAID,
> etc.) that has requirements spread across multiple repositories in the
> workspace.

> It needs to support "variation management" - tracking moderately different
> requirement sets for closely related products.

> @RobertoBagnara How well does StrictDoc support those needs?

Hello everyone,

I am the developer of StrictDoc. I became aware of this thread and thought I
could answer some of the questions about StrictDoc directly.

StrictDoc is in active development, and we are always happy to get feedback from
the users and implement all improvements suggestions that match our roadmap.

I started building StrictDoc in 2019 to implement some features that I was
missing in Doorstop and the main use case I had in mind originally was the
production of requirements documentation for spacecraft avionics projects
according to the ECSS standards. The ECSS-E-ST-40C suggests that a documentation
package for software projects consists of documents like:

- Software System Specification (SSS)
- Software Requirements Specification (SRS)
- Interface Control Document (ICD)
- Software Design Document (SDD)
- ...and some other documents, each having a very detailed predefined template.

StrictDoc is based on a Document concept that consists of sections/chapters or
requirements, and sections can consists of child sections or requirements.
Besides including requirements, sections can have blocks of free text. Free text
blocks provide all necessary glue to help a user to understand the requirements.
A document can also contain only sections and free text. Example: An ECSS SW
User Manual document normally does not have requirements but only free text
structured with sections (chapters).

[An example of a typical SDoc file](https://github.com/strictdoc-project/strictdoc/blob/main/docs/strictdoc_20_requirements.sdoc)

StrictDoc deliberately does not cut the requirements to be one requirement per
one file because, to my original and still existing view, that breaks the idea
of a coherent requirements specification document that can be read and
understood as a whole.

After some time, I also implemented a Python library for ReqIF format, and it
was very easy to map the concept of StrictDoc's Document to ReqIF's
Specification and the concept of StrictDoc's Section to ReqIF's Spec Hierarchy.
The library is not coupled to StrictDoc and at some point I extracted it to be a
standalone building block: https://github.com/strictdoc-project/reqif.

> It needs to support a single "document" (e.g., StRS, SyRS, SyAID, SRS, SAID,
> etc.) that has requirements spread across multiple repositories in the
> workspace.

StrictDoc supports
[including Sections from other files](https://strictdoc-project.github.io/docs/strictdoc_01_user_guide.html#6.2.5-Include-files).
It should be possible to build a StrictDoc document from a number of
contributing sections stored in separate files. StrictDoc does not support
including single requirements but that should be doable. In that case, I would
like to see a strong rationale/use case for going back to Doorstop's "single
req - single file" approach.

StrictDoc is not connected to a single repository concept in any way. You can
run StrictDoc against a given folder and it will recursively find all \*.sdoc
files and collect them into a single documentation tree. Nested folders and
submodules are possible. The same should apply to finding the document fragments
but I admit I haven't tested this multi-repo case with section fragments yet.

@gregshue, could you elaborate a little more on which repositories do you have
in mind and how are you planning to maintain a single document that is spread
[... 67 lines cut ...]
  accurate estimates about the functionality missing in StrictDoc. Feel free to
  introduce a temporary email thread or a GitHub issue to clarify what's needed
  in a detailed discussion and then summarize the conclusions here.

- I can imagine a hybrid solution where StrictDoc could play a limited role and
  be used together with other tools such as RTEMS specs toolchain. In any case,
  consider using [ReqIF](https://github.com/strictdoc-project/reqif) for
  converting Zephyr requirements to ReqIF and vice versa. I want to believe that
  the tool is becoming quite a capable Python parser/unparser for ReqIF. Once in
  a few months, I still receive a ReqIF file with "a yet another edge case" from
  the users of DOORS, Polarion, etc but these reports are becoming rare.

- I have a basic description of how StrictDoc compares to some other tools
  [here](https://strictdoc-project.github.io/docs/strictdoc_02_faq.html#5-How-StrictDoc-compares-to-other-tools-).
  In particular, I didn't see anyone to mention
  [Sphinx-Needs](https://sphinx-needs.readthedocs.io/en/latest/) which is at
  least as capable as StrictDoc and is very powerful due to its full
  compatibility with the Sphinx plugin ecosystem.

- I would like to include several people whose work was either mentioned in this
  thread or is related to the topic of text-based requirements management:
  @jacebrowning (Doorstop), @danwos (Sphinx-Needs), @anmavrid (FRET).

Thank you for your attention, and I hope that my input adds value to the
discussion.

danwos, 2023-05-08:
Hi all,

thanks @stanislaw for mentioning [Sphinx-Needs](https://www.sphinx-needs.com/) and for the great, detailed answer.

Let me try to introduce Sphinx-Needs, which in my understanding can also be used to solve your documentation requirements for safety and security docs.

First of all, I'm the current maintainer and founder of Sphinx-Needs, which was created ~6 years ago based on the needs of safety-relevant documentation at an Automotive OEM company (following ISO26262).

**What is it?**

In my understanding, Sphinx-Needs is more of an ALM toolbox for docs-as-code-based documentation, which contains safety-relevant documentation but is also used for architecture docs, test run docs or simple FAQs.

And as each team is following slightly different processes, with different tool and CI setups, Sphinx-Needs is highly customizable to fulfill nearly every dream regarding customization, data validation, and final representation.

Sphinx-Needs is an extension for the documentation generator Sphinx.
It uses and supports natively all Sphinx data formats and features.
So defining a requirement in rst or md syntax, and adding tables, images, math formulas, and own written solutions, is given by default.
This also means Sphinx-Needs objects (like requirements) can be defined everywhere in the documentation: all in one file, one per file, and between other paragraphs. It's up to your own rules how the structuring should look like.

You are also free to decide, what kind of objects you need. Is it a customer requirement, or more an epic with a linked user story?
And all of these decisions can be validated by Python based statements ([needs-warnings](https://sphinx-needs.readthedocs.io/en/latest/configuration.html#needs-warnings)) or even more complex by pydantic models via [Sphinx-Modeling](https://sphinx-modeling.useblocks.com/), which is an extension on top of Sphinx-Needs. 

Another one is [Sphinx-Test-Report](https://sphinx-test-reports.readthedocs.io/en/latest/) to get test results into your docs and automatically link them with test cases.
The API and data model are completely open, so you are free to inject our own customizations.

With [needservices]}(https://sphinx-needs.readthedocs.io/en/latest/services/index.html) you can integrate external systems, for instance, to fetch GitHub issues and link them to requirements/specs.
Or you use the [Variant Handling](https://sphinx-needs.readthedocs.io/en/latest/services/index.html) to maintain different options for different targets/customers for the same requirement.
And [dynamic functions](https://sphinx-needs.readthedocs.io/en/latest/dynamic_functions.html) allows the automatic calculation of option values, like status or safety statements based on the linked elements.

**Where is it used?**

Sphinx-Needs is used in some bigger commercial Automotive ECU projects for SW development.
The biggest project I'm aware of is ~2.000 SW developers with a single documentation containing up to 130.000 Sphinx-Needs objects in ~22.000 rst/md files (Disclaimer: I and my team are providing commercial support for this).

Some projects are trying to introduce Sphinx-Needs for all steps in the V-Model:
[image]
Starting from below (SW dev part) and injecting more and more process steps from the left and right side (SW Arch, SW Testing, System Arch, ...). It's an ongoing process to make Sphinx-Needs available to support also the special requirements on the upper process steps.

**What else**
There is an online presentation available, which shows most features on some slides:
[Documentation without frustration presentation](https://useblocks.com/dwf/#/).

Sorry, that I wasn't able to get into your special project requirements and provide already some solutions.

But in my understanding, StrictDoc and Sphinx-Needs are both capable to deal with your requirements.
And if not,  we hopefully have found some new contributors ;)

gregshue, 2023-05-08:
@stanislaw Good questions.  I don't have a solution in mind, but I have a pretty clear idea of the problem.

Let's look at this from the perspective of a manufacturer/product developer:
- The IoT device fits within an overally solution that interfaces with natural people (HMI) and external automated agents (MMI, e.g., the electrical utility load-shed client). The natural people, the regulatory agencies, and the business developing the end product are all examples of Stakeholders. Stakeholders have needs that may be met by the system and impose some level of constraints on how the system meets those needs.
- The System consists of everything that is provided that meets those needs.  For an IoT device, the system includes the remote service(s) associated with the device, along with all related cloud/desktop/mobile-based applications. 
- Devices, mobile apps, and cloud services are examples of System Elements. Each has specific responsibilities and implements a subset of the System-external and/or System-Internal interfaces.
- A IoT Device (System Element) is composed of at least HW (e.g., mechanics, electronics, immutable boot code), configuration data (e.g., unit serial number), and typically has updatable embedded SW. Each of these are Sub-Elements.
- A Device FW image may contain complete binary images for subordinate devices (e.g., radios, GPUs, asymmetric CPU cores, peer MPUs). These images may be proprietary, commercial, or OSS.
Note that product developers may want a single platform that supports configuring and building multiple Sub-Elements (bootloader, app) within each device along with supporting a set of distinctly different devices in the System (e.g., sensor vs actuator, initiator vs responder, scanner vs printer vs. MFP). 
Zephyr is primed to be a reasonable framework for that platform. Part of the Zephyr solution is to support bootloader/app internal components and interfaces to be distributed across peer repositories. These repositories must coexist peacefully when viewed in aggregate in the same code tree, after filtering based on project configuration settings. 
Note that the end user gets to choose the topology of the aggregate code tree. They also get to create whatever document tree they want, and which portions to use Zephyr RM tool for. They should have the option to develop and trace their entire system in detail with the Zephyr RM tool.

So ...

> could you elaborate a little more on which repositories do you have in mind 

I'm thinking of any/all of them. We need a requirements solution that scales to:

1. up to all levels of requirements decomposition;
2. across all potential repository distributions (e.g., zephyrproject-rtos, downstream OSS, commercial, private);
3. down to being in the middle of a larger solution.

> how are you planning to maintain a single document that is spread across several repositories?

I don't have a proposal. I'm looking for suggestions too.

> I can guess that from these child repositories

We cannot restrict these to being child repositories.

> you will not be cherry-picking single requirements to your parent document but rather you will be including more coherent and self-contained sections with requirements and free text, is it so?

The solution needs to scale across both usages. 

> In the end, StrictDoc as a top-down system can produce Documents from smaller fragments,  and the RTEMS toolchain can produce Documents from composable specification items.

Fundamentally, a Zephyr application is a selection/configuration of composable subsystems & drivers (components) controlled via Kconfig & Devicetree. It is natural to model the RM for interfaces & components in the same way.

danwos, 2023-05-09:
@gregshue thanks for the detailed problem description.
[Sphinx-Needs ](https://sphinx-needs.readthedocs.io/en/latest/)together with [Sphinx-Collections](https://sphinx-collections.readthedocs.io/en/latest/)  have some concepts to deal with this kind of decentralized requirement distribution.

First of all, Sphinx-Needs has an exchange format, called ``needs.json``.
This can be used to import objects from other projects and have a representation of them in the new project. 
Or the objects from ``needs.json`` are just referenced, so that you can link objects to them, but the external objects have no representation in the new project. Links in this kind of referencing are later real HTML-links to the correct location of the referenced need in the external project.

A simple workflow looks like:
[image]

The above mechanism is good to share objects between Sphinx projects.
But it does not solve generic integration problems, where you have for instance standalone projects for components and need an integrated, overall system documentation, which shall include all the content (text, images, tables, ...) from the component projects.
For this, [Sphinx-Collections](https://sphinx-collections.readthedocs.io/en/latest/) was created, which integrates files from different projects into an overall project. Sources can be repositories, local folders, functions, and data provided by self-written functions.
Sphinx tags can be used to easily switch between different integration configurations.
A build call could look like: `sphinx-build -t system_view -b html input/ output/`

Some further links:
* [need.json details](https://sphinx-needs.readthedocs.io/en/latest/builders.html#needs)
* [needimport - filter and import objects from needs.json](https://sphinx-needs.readthedocs.io/en/latest/directives/needimport.html)
* [external_needs config - how to allow objects to link to external reqs and co.](https://sphinx-needs.readthedocs.io/en/latest/configuration.html#needs-external-needs)
* [permalinks - track/link objects, even if they have moved](https://sphinx-needs.readthedocs.io/en/latest/configuration.html#needs-external-needs)

I hope this already helps to get a feeling about some mechanisms of object-handling between different docs-as-code projects.

sebhub, 2023-05-09:
> @sebhub Thank you for the pointers. It looks to be very relevant content.
>
>
> From the [RTEMS UID definitions](https://docs.rtems.org/branches/master/eng/req/req-for-req.html) it looks like RTEMS [...]

Each specification item (for example a requirement) needs a unique identifier (UID). For the RTEMS specification, the UID is derived from the file system path to the file which stores the specification item (in YAML or JSON format). The base directory and the file format extension is removed from the path to form the UID. For example [spec/req/root.yml](https://github.com/RTEMS/rtems-central/blob/master/spec/req/root.yml) is loaded using the base directory `spec` which results in the UID `/req/root`. The specification items are loaded from the file system by recursively searching for files of a given format in a configurable list of base directories. The RTEMS specification and tooling is located in the [rtems-central](https://github.com/RTEMS/rtems-central) repository and the RTEMS sources added through a Git submodule.

The links in the specification item are unidirectional and have a role. This can be used to make the specification composable. For example, the requirement refinement relationship is stored in the specializing item and links to the more general requirement. So, the higher level parts of the specification do not need to know the specializations. You could for example have a higher level requirement like:
```
The system shall be extendable by modules.
```
Then in a specific module you may have this:
```
The network stack module shall extend the system to provide TCP/IP networking services.
```
When you build the documents for a given configuration, the links are traversed to compose documents specific to this configuration.

Another feature which helps to make the specification modular are proxy items. They can be used by higher level specification parts to refer to items which may be implemented in an architecture-specific or device-specific way. For example, the Gaisler IP library is available for SPARC and RISC-V systems. The generic part uses a proxy item [/dev/grlib/if/store-32](https://github.com/RTEMS/rtems-central/blob/master/spec/dev/grlib/if/store-32.yml) to refer to some method which stores a 32-bit value to a register. This method is provided by an architecture-specific implementation [/bsp/sparc/if/grlib-load-32](https://github.com/RTEMS/rtems-central/blob/master/spec/bsp/sparc/if/grlib-load-32.yml). For a particular configuration, the proxy items are resolved to a specific item.

Another feature is the hierarchical glossary of terms. A glossary of terms is quite important for larger projects. However, some terms may be context-specific. An example are state machines specific to a certain functionality such as timers. For this you can create a hierarchy of glossaries:

* https://github.com/RTEMS/rtems-central/blob/master/spec/glossary-general.yml

* https://github.com/RTEMS/rtems-central/blob/master/spec/rtems/timer/glossary/group.yml

The terms defined by the Timer Manager glossary are used in the state machine specifications, for example [/rtems/timer/req/reset](https://github.com/RTEMS/rtems-central/blob/master/spec/rtems/timer/req/reset.yml).

sebhub, 2023-05-09:
@stanislaw thanks for your good overview. Yes, the RTEMS specification approach is very bottom up. It was a reverse engineering activity and not the development of a new product. For general requirements engineering I would not store individual requirements in individual files. This is just way to cumbersome and does not help you to discover a good set of high level requirements, user stories, etc. I think these two approaches could be easily combined. In the end, the specification is just a directed graph where each node is identified by an UID. I also belief that each projects needs a bit of specific tooling to be productive. It is important to have simple and open data formats which can be effectively used with Git (like reST, StrictDoc, YAML).

tobiaskaestner, 2023-05-09:
IMHO, this issue is a big leap into the right direction for allowing product creators in safety-critical domains adopt zephyr for their use-cases. That said I felt a bit lost in the discussions as it seems that ideas at least around process and actual tools are mixed up. Not saying they do not relate to one another, obviously they do. Still, disentangling the threads might help to focus the discussion. 
* **process complexity** Regarding the process I am concerned that an unmanageable thing might be created which I have seen to happen too often in product developments. And soon enough maintenance cost will dominate everything and hinder innovation. From my analysis that's one of the main reasons organisations are stuck on using decades-old technologie stacks. Manually (that's where the cost for many companies sits) maintaining a set of requirements in a seperate repo (even if pulled in through west) means submitting PRs against two repos in case something needs to change and is governed by a safety-clause. Given the frequency of changes that go into the kernel folder alone (and which would clearly be considered governed by a safety-clause) I am concerned about the general acceptance. 
* **combinatorical complexity** Another thing that I have thought about for some time now is how to deal with the 'superset' character of Zephyr. One of the hallmarks of Zephyr is that every aspect of it can be configued at compile time. The downside of this is that it creates an infinite (infinite for all practial purposes) configuration space. In the end downstream projects have to expect certain safety claims to hold as they have to build their safety case based on these assumptions and claiming validity in all end every conceivable configuration I consider not feasible. That begs the question which options to include and which to freeze if you wanted to use Zephyr in a safety-critical context, i.e. you have to use the dumb scheduler, not more than 16 threads, at least 128K stack_size etc.). 
* **smart approach to design specification** Lastly, even then a huge amount of test cases need to be generated (and kept up-to-date) and looking into methods how test cases can be (re-)generated  from (semi-)formal specifications may hold some additional value in that they keep the maintenance cost at an acceptable level. Examples include [NASA's FRET](https://github.com/NASA-SW-VnV/fret) or [Lamport's TLA+](https://lamport.azurewebsites.net/tla/tla.html#:~:text=TLA%2B%20is%20a%20high%2Dlevel,precisely%20is%20with%20simple%20mathematics.)

To sum up before going to call for some tools and rush into action some thought experiments how the process will play out over some time may help falling into avoidable pitfalls

gregshue, 2023-05-10:
> some thought experiments how the process will play out over some time may help

Agreed, which is why I the first thing I contributed to #57703 are some needs from downstream users, particularly from product manufacturers. Please feel free to add others that you are aware of.

tobiaskaestner, 2023-05-15:
@sebhub I eventually managed to take a deeper look into what you guys are doing at RTEMS and it's really impressing. 

Personally, I think it would make a lot of sense if FOSS projects targeting the safety-critical domain would eventually align on a 'standard' approach towards requirements engineering and how to provide a safety case for the FOSS under consideration. Reason being that many notified bodies (those agencies eventually responsible to asses the safety case) will probably have had none to very little exposure to FOSS yet. And the less familiar things look the more scrutiny auditors will put forth and hence whatever could be presented as a 'standard' way may help (even more so if someone would actually testify that the process is sound and apt for FOSS projects)

The RTEMS process and the content you have already produced using it could well serve as a reference for other projects to follow. 

I wonder how much experience you have had with the process when it comes to adding updates to design elements that had been captured previously