zephyrproject-rtos/zephyr #57520: Compile warning with lib/posix/getopt

Frodevan, 2023-05-03:
After the changes by https://github.com/zephyrproject-rtos/zephyr/pull/48130, lib/posix/getopt/getopt_long.c will compile with a warning that **char* dash_prefix** may be uninitialized before use in function **parse_long_options**.

While the root concern of this warning will never happen on runtime due to conditional checks, the compiler is conservative and throws the warning never the less, causing problems if the use of this file is to be combined with the -Werror compile flag.

Possible solutions include to initialize this variable in the case it is not used, as was the case in the past, or do the conditional checks associated with it on compile-time instead of runtime.

cloudflare-wangler, 2023-05-04:
***@***.*** added you to their Edison Mail+ Family Plan

    
    
        
            ***@***.*** added you to their Edison Mail+ Family Plan
        
        
            
                Welcome to the family! Make sure your Back up & Sync is turned on,
                and access your new Edison Mail+ premium email features by
                connecting this email account to the Edison Mail app.
                You will continue to have free access as long as ***@***.***
                has an active Plus subscription.
            
        
        
            
                Maximize your email security, block spam calls, and more!
            
        
        
            
                Don't have the app?

cfriedt, 2023-05-04:
@Frodevan - can you please use the bug template for your summary?

Frodevan, 2023-05-04:
@cfriedt Done

koffes, 2023-05-04:
https://github.com/zephyrproject-rtos/zephyr/pull/57540

cfriedt, 2023-05-04:
Is there a `west` repro command that you can add to the summary?

Frodevan, 2023-05-04:
@cfriedt Allright, added it

cfriedt, 2023-05-04:
@Frodevan - can you provide a repro command that can be run in the zephyr repository by an arbitrary person who does not have access to your source code?

Frodevan, 2023-05-04:
@cfriedt ok, you should be able to do the following, assuming you have the Zephyr toolchain all set up and ready:

**step 1: Fetch code to be compiled and make sure it's at the stage we are at**
- Make an empty directory and open a command line interface there
- Run the following commands:
```
git clone https://github.com/koffes/sdk-nrf.git nrf
cd nrf
git checkout bis_acl_test
cd ..
```

**Step 2: Make sure zephyrptoject-rtos Zephyr and not the downstream fork is used**
- In the file **nrf/west.yml**, replace what's on line 61 and 62 with the following and save:
_Line 61:_ `      url: https://github.com/zephyrproject-rtos/zephyr.git`
_Line 62:_ `      revision: main`

**Step 3: Prepare and invoke build**
- In the cli (same folder as before), run:
```
west init -l nrf
west update
```
- Wait for everything to fetch, it takes a few minutes
- Copy this one file from the downstream fork into the **zephyr/modules/mbedtls** folder: https://github.com/nrfconnect/sdk-zephyr/blob/main/modules/mbedtls/Kconfig.psa
- In the cli:
```
west build -p -b nrf5340_audio_dk_nrf5340_cpuapp -d build_output_folder nrf/tests/bluetooth/iso -- -DCMAKE_C_FLAGS="-Werror"
```

Frodevan, 2023-05-04:
As of samples in Zephyr itself, I was not able to find any that sets CONFIG_GETOPT_LONG. If you compile any existing sample for this target with **-DCONFIG_GETOPT=y -DCONFIG_GETOPT_LONG=y -DCONFIG_SHELL_GETOPT=y** added to the end of the build-command, it should in theory invoke the bug.

cfriedt, 2023-05-05:
@Frodevan - probably a better repro recipe is here. And it seems we actually do have coverage for `CONFIG_GETOPT_LONG`. This would be executed in CI pretty regularly.
```
twister -i --build-only -p nrf5340_audio_dk_nrf5340_cpuapp -T tests/posix/getopt
# or the command below for 
west build -p always -b nrf5340_audio_dk_nrf5340_cpuapp tests/posix/getopt -- -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y
```

If you don't mind, please try to use the common formats above for reproducibility information next time. I do appreciate the bug report though - thank you!

What I find weird is this: normally, `twister` has historically elevated all warnings to errors and we have some Kconfig options that imply `-Werror` is enabled.

We have a few layers of abstraction here, and sorry for the safari, but `twister` normally [sets](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/runner.py#L328) `CONFIG_COMPILER_WARNINGS_AS_ERRORS=y`.

Then, in `CMakeLists.txt`, we have `zephyr_compile_options($<$<COMPILE_LANGUAGE:C>:$<TARGET_PROPERTY:compiler,warnings_as_errors>>)`. which translates to the following based on the toolchain:
```shell
~/zephyrproject/zephyr (main)]$ grep -R "warnings_as_errors" cmake/
cmake/compiler/arcmwdt/compiler_flags.cmake:set_property(TARGET compiler PROPERTY warnings_as_errors -Werror)
cmake/compiler/arcmwdt/compiler_flags.cmake:set_property(TARGET asm PROPERTY warnings_as_errors -Werror)
cmake/compiler/gcc/compiler_flags.cmake:set_property(TARGET compiler PROPERTY warnings_as_errors -Werror)
cmake/compiler/gcc/compiler_flags.cmake:set_property(TARGET asm PROPERTY warnings_as_errors -Werror -Wa,--fatal-warnings)
cmake/compiler/compiler_flags_template.cmake:set_property(TARGET compiler PROPERTY warnings_as_errors)
cmake/compiler/compiler_flags_template.cmake:set_property(TARGET asm PROPERTY warnings_as_errors)
cmake/linker/arcmwdt/linker_flags.cmake:set_property(TARGET linker PROPERTY warnings_as_errors -Wl,--fatal-warnings)
cmake/linker/ld/gcc/linker_flags.cmake:set_property(TARGET linker PROPERTY warnings_as_errors -Wl,--fatal-warnings)
cmake/linker/linker_flags_template.cmake:set_property(TARGET linker PROPERTY warnings_as_errors)
```

Just to verify that all of the above works as expected, I used [this suggestion from the docs](https://docs.zephyrproject.org/latest/develop/west/build-flash-debug.html#verbose-builds).

```shell
cd /home/cfriedt/zephyrproject/zephyr/build/zephyr && ccache /home/cfriedt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF5340_XXAA_APPLICATION -DNRF_SKIP_FICR_NS_COPY_TO_RAM -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/cfriedt/zephyrproject/zephyr/kernel/include -I/home/cfriedt/zephyrproject/zephyr/arch/arm/include -I/home/cfriedt/zephyrproject/zephyr/include -I/home/cfriedt/zephyrproject/zephyr/build/zephyr/include/generated -I/home/cfriedt/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53 -I/home/cfriedt/zephyrproject/zephyr/lib/posix/getopt/. -I/home/cfriedt/zephyrproject/zephyr/soc/arm/nordic_nrf/common/. -I/home/cfriedt/zephyrproject/zephyr/subsys/testsuite/include -I/home/cfriedt/zephyrproject/zephyr/subsys/testsuite/ztest/include -I/home/cfriedt/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/cfriedt/zephyrproject/modules/hal/nordic/nrfx -I/home/cfriedt/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/cfriedt/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/cfriedt/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -isystem /home/cfriedt/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/cfriedt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /home/cfriedt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Werror -Os -imacros /home/cfriedt/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs --sysroot=/home/cfriedt/zephyr-sdk-0.15.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/cfriedt/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/cfriedt/zephyrproject/zephyr/tests/posix/getopt=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/cfriedt/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/cfriedt/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -D_POSIX_THREADS -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/posix/getopt/getopt_long.c.obj -MF CMakeFiles/zephyr.dir/lib/posix/getopt/getopt_long.c.obj.d -o CMakeFiles/zephyr.dir/lib/posix/getopt/getopt_long.c.obj -c /home/cfriedt/zephyrproject/zephyr/lib/posix/getopt/getopt_long.c
```

So as far as I can tell, I'm unable to reproduce the actual problem.

Is there perhaps some static analyzer or other toolchain that is being used here?

Just wanted to investigate here, because as far as I can tell, this is not a bug.

cfriedt, 2023-05-05:
I'll leave the bug label for now. If you can provide any additional information, that would be great.

cfriedt, 2023-05-07:
> west build -p -b nrf5340_audio_dk_nrf5340_cpuapp -d build zephyr/samples/bluetooth/iso_broadcast/ -- -DCMAKE_C_FLAGS="-Werror" -DCONFIG_GETOPT=y -DCONFIG_GETOPT_LONG=y

@koffes just saw this.

Was able to reproduce with a slightly different command
```shell
west build \
    -b nrf5340_audio_dk_nrf5340_cpuapp \
    samples/bluetooth/iso_broadcast/ \
    -- \
    -DCMAKE_C_FLAGS="-Werror" \
    -DCONFIG_GETOPT=y \
    -DCONFIG_GETOPT_LONG=y
```

So I'll merge the fix. However, I need to investigate why `twister` does not seem to be escalating the `-Werror` at some point.

cfriedt, 2023-05-07:
@Thalley are you able to investigate why `tests/posix/getopt` passes in CI in spite of the uninitialized `current_dash` variable? For some reason, I'm actually unable to make the testsuite fail. It's possible that there is some aspect of `samples/bluetooth/iso_broadcast` that triggers an uncovered code path in the getopt code.

I guess that should probably be part of a new ticket.

jakub-uC, 2023-05-08:
@cfriedt 
First of all, I see the bug was introduced with a commit: 6f831d542a3b94629d3af1819ebe78a7fa957371.

Variable `current_dash` is used only for logging purposes. For the sake of saving memory and to be able to run the test on as many platforms as possible, the logger module was disabled in this test. Since the logger is disabled then the code where `current_dash` is used does not compile at all.
As a result, the `tests/posix/getopt` test does not generate a warning.

cfriedt, 2023-05-08:
> @cfriedt 
> 
> Variable `current_dash` is used only for logging purposes. For the sake of saving memory and to be able to run the test on as many platforms as possible, the logger module was disabled in this test. Since the logger is disabled then the code where `current_dash` is used does not compile at all.
> As a result, the `tests/posix/getopt` test does not generate a warning.

@jakub-uC - would it be possible to add a new permutation to `testcase.yaml` that enables the logger with a min ram filter?

jakub-uC, 2023-05-08:
Sure, I can do it.

closed 2023-05-09 by cfriedt