zephyrproject-rtos/zephyr #57523: Add RSOC/ASOC support to sbs_gauge

protterc, 2023-05-03:
## Introduction

Add support for reading absolute and relative state of charge from SBS batteries.

### Problem description

Currently the sbs_gauge driver only has provision for returning "state of charge" (SOC), so returns absolute state of charge (ASOC) when requested.  For developing battery charging and system integration purposes, also need to get access to relative state of charge (RSOC)

### Proposed change

Propose to add support for both RSOC and ASOC, and drop SOC from the SBS gauge driver.  Would leave SOC in the generic fuel gauge interface, so that other types of fuel gauges could continue to use it.

## Detailed RFC
1) Add FUEL_GAUGE_ABSOLUTE_STATE_OF_CHARGE and FUEL_GAUGE_RELATIVE_STATE_OF_CHARGE to list of symbols in fuel_gauge.h
2) Add support for the new symbols to sbs_gauge.c, pointing to the correct device registers for each value
3) Remove FUEL_GAUGE_STATE_OF_CHARGE support from sbs_gauge.c

### Proposed change (Detailed)

It's a small change, and mostly detailed above.  Only question would be step 3 - should we remove the generic SOC support?  On the one hand, it's not really descriptiive of what the function actually returns (ASOC), but there may be applications that currently use it that would be broken.

### Dependencies
This is a focused change that only affects the sbs_gauge functionallity

### Concerns and Unresolved Questions

Only the question about deprecating item 3

## Alternatives

aaronemassey, 2023-05-05:
Sounds like a good idea to me. I'd prefer we also make a change on the API level though and change `FUEL_GAUGE_STATE_OF_CHARGE` to `FUEL_GAUGE_ABS_STATE_OF_CHARGE` or similar.

It's already documented in the API as ASOC. This will break current up/downstream clients of the API but it'll be a compile-time breakage that is easily fixable via regex. We're also a bit more okay breaking downstream users right now as the API is explicitly listed as `experimental`.

protterc, 2023-05-19:
Resolved with https://github.com/zephyrproject-rtos/zephyr/pull/57824

closed 2023-05-19 by protterc