zephyrproject-rtos/zephyr #57526: uart_basic_api fails on FRDM K64F with power management enabled

edersondisouza, 2023-05-03:
**Describe the bug**
tests/drivers/uart/uart_basic_api fails on FRDM K64F with power management enabled

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p -b frdm_k64f tests/drivers/uart/uart_basic_api -DCONFIG_PM=y -DCONFIG_PM_DEVICE=y
2. [Open a terminal to collect serial output, e.g., `screen /dev/ttyACM0 115200`]
3. west flash

**Expected behavior**
On serial output, see tests completing successfully.

**Impact**
Fairly low, just happened to notice.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-3420-g7a946450e51e ***
Running TESTSUITE uart_basic_api
===================================================================
START - test_uart_config_get
 PASS - test_uart_config_get in 0.003 seconds
===================================================================
 PASS - test_uart_configure in 0.001 seconds
===================================================================
START - test_uart_fifo_fill
This is a FIFO test.
```
And it hangs there.

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.15.0
 - 7a946450e51e9b92c99c4326eb2f092e420ba64f

**Additional context**
Seems to have caused by commit 3f0fc7fe6b8ff6fc7ca6cccbd40c4fb7349f5fe3

decsny, 2023-06-02:
hi @edersondisouza , I performed this test on commit 44e691e20e0b323ad67227ee072bd4166c75d894 with the command `west build -p -b frdm_k64f tests/drivers/uart/uart_basic_api -DCONFIG_PM=y -DCONFIG_PM_DEVICE=y` and `west flash -r jlink` (my frdm k64 board has jlink debug firmware) and I was not able to reproduce this issue (the test passed and project successful with no hang). Do you still see the issue?

edersondisouza, 2023-06-05:
Hi @decsny, 

Indeed, there's no issue anymore. It seems to have been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57909.

Thanks!

closed 2023-06-05 by edersondisouza