zephyrproject-rtos/zephyr #57536: up_squared: sample: rtio sensor_batch_processing - test not responding

KamilxPaszkiet, 2023-05-04:
When the test starts on Dut it stops at this point:
```
DEBUG   - DEVICE: >>Checking Media Presence...... 
DEBUG   - DEVICE: >>Media Present...... 
DEBUG   - DEVICE: Downloading NBP file...
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Succeed to download NBP file.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x79c34000
DEBUG   - DEVICE: Zeroing 4816288 bytes of memory at 0x108000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 32768 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2114144 data bytes to 0x59fda0 from image offset 65536
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
```
This sample does not have any README.

**To Reproduce**
```
twister --device-testing --hardware-map $HOME/up2_double.map -T samples/subsys/rtio/sensor_batch_processing/ --no-skipped-report -vv
```
or
```
west build -p -b up_squared -T samples/subsys/rtio/sensor_batch_processing/sample.rtio.sensor_batch_processing
```

**Expected behavior**
The test should pass. 
We expect prints from log_inf, for example:
`LOG_INF("Submitting %d read requests", M);`
```
regex:
        - "(.*)Submitting (.*) read requests"
        - "(.*)Start processing (.*) samples"
        - "(.*)Finished processing (.*) samples"
```

- OS: Ubuntu for the controller
- Toolchain: Zephyr SDK 15.0
- Commit: c729e12beb

KamilxPaszkiet, 2023-07-25:
The isssue still exists

teburd, 2023-07-25:
Is this fixed with #59716 perhaps?

KamilxPaszkiet, 2023-07-27:
> Is this fixed with #59716 perhaps?

Unfortunately that doesn't fix the problem. I still have a reproduction... :(

jhedberg, 2023-11-22:
The sample is working fine for me with the latest tree. I don't have any test automation/harness to use `twister --device-testing` but simply building the sample and booting it from a USB drive works fine, i.e. I get the continuous output that the sample produces, e.g:
```
I: Submitting 4 read requests
D: Consumed completion event 0
D: Consumed completion event 1
D: Consumed completion event 2
D: Consumed completion event 3
I: Start processing 4 samples
D: Sample data:
D: c0 c1 c2 c3 c4 c5 c6 c7 |........
D: c8 c9 ca cb cc cd ce cf |........
D: Sample data:
D: d0 d1 d2 d3 d4 d5 d6 d7 |........
D: d8 d9 da db dc dd de df |........
D: Sample data:
D: e0 e1 e2 e3 e4 e5 e6 e7 |........
D: e8 e9 ea eb ec ed ee ef |........
D: Sample data:
D: f0 f1 f2 f3 f4 f5 f6 f7 |........
D: f8 f9 fa fb fc fd fe ff |........
I: Finished processing 4 samples
I: Submitting 4 read requests
D: Consumed completion event 0
D: Consumed completion event 1
D: Consumed completion event 2
D: Consumed completion event 3
I: Start processing 4 samples
D: Sample data:
D: 00 01 02 03 04 05 06 07 |........
D: 08 09 0a 0b 0c 0d 0e 0f |........
D: Sample data:
D: 10 11 12 13 14 15 16 17 |........
D: 18 19 1a 1b 1c 1d 1e 1f |........
D: Sample data:
D: 20 21 22 23 24 25 26 27 | !"#$%&'
D: 28 29 2a 2b 2c 2d 2e 2f |()*+,-./
D: Sample data:
D: 30 31 32 33 34 35 36 37 |01234567
D: 38 39 3a 3b 3c 3d 3e 3f |89:;<=>?
I: Finished processing 4 samples
```

closed 2023-11-22 by jhedberg