zephyrproject-rtos/zephyr #57559: Lost external interrupt for ADC data ready is non-recoverable

benediktibk, 2023-05-04:
**Describe the bug**
It might happen that the GPIO interrupt for a data ready signal for an ADC gets lost. This can occur for instance, if the data ready signal is connected to the MCU via an external port expander, like an MCP23s17. Due to the nature of such a setup, combining multiple possible input interrupt lines on one, it can happen that interrupts get lost.
In such a case the ADC driver will wait forever for the read operation to finish.

**Expected behavior**
It should be possible to recover from such a hardware error.

**Impact**
Makes the affected ADC instance completely unusable.

**Additional context**
I'm working already on a fix in #57558.

benediktibk, 2023-07-10:
@henrikbrixandersen This is still an open issue for me, and I am just waiting for feedback in the PR.

henrikbrixandersen, 2023-07-10:
> @henrikbrixandersen This is still an open issue for me, and I am just waiting for feedback in the PR.

Good. I have removed the stale label.

benediktibk, 2023-07-27:
PR #57558 is merged

closed 2023-07-27 by benediktibk