zephyrproject-rtos/zephyr #57579: CI: tests_net_tcp: RT11xx series: tests_net_tcp_simple failure and tests hangs

hakehuang, 2023-05-05:
**Describe the bug**
tests_net_tcp_simple

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt1160_evk_cm7 and mimxrt1170_evk_cm7
 - What have you tried to diagnose or workaround this issue?
NA
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
```
commit 788ba121373ba74ed197fa99fdc1bd92e9f15a42
Author: Yves Vandervennet <yves.vandervennet@nxp.com>
Date:   Tue Apr 4 13:12:54 2023 -0500

    nxp: hal: code update to reflect changes in SDK 2.13

    HAL API changes in ethernet and pwm
    SoC RT595 power management code change
    west.yml update
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/net/tcp
2.  west build -b mimxrt1170_evk_cm7 -- -DCONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=1000
3.  west flash --runner jlink --tool-opt='-SelectEmuBySN 000725371294' --  '--device=MIMXRT1176xxxA_M7' '--reset-after-load'
4. See error

**Expected behavior**
Test PASS

**Impact**
tcp driver

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.3.0-2976-gdf7f10422e45 ***
Running TESTSUITE net_tcp
===================================================================
START - test_client_closing_ipv6
 PASS - test_client_closing_ipv6 in 0.488 seconds
===================================================================
START - test_client_fin_wait_2_ipv4
 PASS - test_client_fin_wait_2_ipv4 in 0.394 seconds
===================================================================
START - test_client_invalid_rst

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp/src/main.c:205: test_sem_take: (false is false)
semaphore timed out (line 1283)
 FAIL - test_client_invalid_rst in 0.294 seconds
===================================================================
START - test_client_ipv4
 PASS - test_client_ipv4 in 0.478 seconds
===================================================================
START - test_client_ipv6
 PASS - test_client_ipv6 in 0.394 seconds
===================================================================
START - test_client_syn_resend
 PASS - test_client_syn_resend in 0.635 seconds
===================================================================
START - test_server_ipv4

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp/src/main.c:205: test_sem_take: (false is false)
semaphore timed out (line 792)
 FAIL - test_server_ipv4 in 0.278 seconds
===================================================================
START - test_server_ipv6
E: Identical connection handler 0x80002a4c already found.

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp/src/main.c:924: net_tcp_test_server_ipv6: (false is false)
Failed to set accept on net_context

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp/src/main.c:133: verify_flags: (false is false)
handle_server_test:650 flags mismatch (0x0010 vs 0x0012)
 FAIL - test_server_ipv6 in 0.142 seconds
===================================================================
START - test_server_out_of_order_data
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.3.0-3014-ge59e65dc75e4

hakehuang, 2023-05-05:
@dleach02 , @mmahadevan108 , @yvanderv  after merge with New HAL  into mainline, I see the RT11xx serise has an issue with ethernet changes, which is missed in PR board testing, false judeged in failure log triage review.

dleach02, 2023-05-17:
@yvanderv I'm assigning this one over to you as bisect points to your changes

yvanderv, 2023-06-09:
using a rt1170 EVK, I cannot reproduce with the instructions above. Same results for each network jack.

*** Booting Zephyr OS build zephyr-v3.3.0-2945-g788ba121373b ***
Running TESTSUITE net_tcp
===================================================================
START - test_client_closing_ipv6
 PASS - test_client_closing_ipv6 in 0.105 seconds
===================================================================
START - test_client_fin_wait_2_ipv4
 PASS - test_client_fin_wait_2_ipv4 in 0.105 seconds
===================================================================
START - test_client_invalid_rst
 PASS - test_client_invalid_rst in 0.257 seconds
===================================================================
START - test_client_ipv4
 PASS - test_client_ipv4 in 0.105 seconds
===================================================================
START - test_client_ipv6
 PASS - test_client_ipv6 in 0.103 seconds
===================================================================
START - test_client_syn_resend
 PASS - test_client_syn_resend in 0.352 seconds
===================================================================
START - test_server_ipv4
 PASS - test_server_ipv4 in 0.253 seconds
===================================================================
START - test_server_ipv6
 PASS - test_server_ipv6 in 0.253 seconds
===================================================================
START - test_server_out_of_order_data
 PASS - test_server_out_of_order_data in 2.060 seconds
===================================================================
START - test_server_with_options_ipv4
 PASS - test_server_with_options_ipv4 in 0.253 seconds
===================================================================
TESTSUITE net_tcp succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [net_tcp]: pass = 10, fail = 0, skip = 0, total = 10 duration = 3.846 seconds
 - PASS - [net_tcp.test_client_closing_ipv6] duration = 0.105 seconds
 - PASS - [net_tcp.test_client_fin_wait_2_ipv4] duration = 0.105 seconds
 - PASS - [net_tcp.test_client_invalid_rst] duration = 0.257 seconds
 - PASS - [net_tcp.test_client_ipv4] duration = 0.105 seconds
 - PASS - [net_tcp.test_client_ipv6] duration = 0.103 seconds
 - PASS - [net_tcp.test_client_syn_resend] duration = 0.352 seconds
 - PASS - [net_tcp.test_server_ipv4] duration = 0.253 seconds
 - PASS - [net_tcp.test_server_ipv6] duration = 0.253 seconds
 - PASS - [net_tcp.test_server_out_of_order_data] duration = 2.060 seconds
 - PASS - [net_tcp.test_server_with_options_ipv4] duration = 0.253 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL

yvanderv, 2023-06-09:
OK, some more comments: the issue cannot be reproduced with the latest HAL configuration. It can when the HAL is reset to 2930bcb6.

hakehuang, 2023-06-10:
> OK, some more comments: the issue cannot be reproduced with the latest HAL configuration. It can when the HAL is reset to 2930bcb6. The HAL commit making the test succeed is 6053bdcb

@yvanderv  thanks for the checking. it is PASS in rc2 regression test.

closed 2023-06-10 by hakehuang