zephyrproject-rtos/zephyr #57581: snippets: Possibility to include multiple overlay files in one snippet

maciejbaczmanski, 2023-05-05:
**Is your enhancement proposal related to a problem? Please describe.**
Right now, only one `DTC_OVERLAY_FILE` and `OVERLAY_CONFIG` can be added per snippet, which in some cases requires creating multiple snippets.
For example, while building low power mode application for power consumption measurements from nRF OpenThread CLI sample (https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/protocols/thread/overview/power_consumption.html#openthread-power-consumption) , requires adding: `-- -DOVERLAY_CONFIG="overlay-ci.conf;overlay-low_power.conf" -DDTC_OVERLAY_FILE="low_power.overlay"`. It would be nice to replace it with one snippet, e.g. `-S low_power`. 

**Describe the solution you'd like**
 Above problem could be solved by changing `snippets.py` file and `snippet-schema.yml` to accept list of strings instead of just one string and creating a snippet this way:
```
name: low_power
    append:
      OVERLAY_CONFIG:
        - overlay-ci.conf
        - overlay-low_power.conf
      DTC_OVERLAY_FILE:
        - low_power.overlay
```

**Describe alternatives you've considered**
We can create multiple snippets for now, but proposed enhancement could make the process easier.

@mbolivar-nordic @stephanosio @carlescufi

stephanosio, 2023-05-05:
While I see no problem with adding this feature, I have a question regarding your specific use case:

>  requires adding: -- -DOVERLAY_CONFIG="overlay-ci.conf;overlay-low_power.conf"

What is the purpose of `overlay-ci.conf` and why can't it be merged into `overlay-low_power.conf`?

maciejbaczmanski, 2023-05-05:
> While I see no problem with adding this feature, I have a question regarding your specific use case:
> 
> > requires adding: -- -DOVERLAY_CONFIG="overlay-ci.conf;overlay-low_power.conf"
> 
> What is the purpose of `overlay-ci.conf` and why can't it be merged into `overlay-low_power.conf`?

Hi @stephanosio. `overlay-ci.conf` changes shell appearance. It is used with some other configurations too, for example together with `overlay-logging.conf` in leader board's application from example given in the link above. So I think it would be nice to have it as a separate file to avoid duplicating the code.

stephanosio, 2023-05-05:
> `overlay-ci.conf` changes shell appearance. It is used with some other configurations too, for example together with `overlay-logging.conf` in leader board's application from example given in the link above.

Based on that description, `overlay-ci.conf` should be a separate snippet. Is there any reason that should not be the case?

maciejbaczmanski, 2023-05-05:
> Based on that description, `overlay-ci.conf` should be a separate snippet. Is there any reason that should not be the case?

I think it could, but:
-  `overlay-ci.conf` is rarely used on its own (only for one use case AFAIK) while commonly added to other use cases. IMO not requiring to add it as separate snippet for them could make building the SW more transparent as snippet could be created specifically for the use case.
- we can see that according to documentation: https://docs.zephyrproject.org/3.2.0/build/dts/howtos.html#set-devicetree-overlays now  file `boards/<BOARD>.overlay` is loaded only if `DTC_OVERLAY_FILE` is empty, so if user adds some other overlay (e.g `low_power.overlay`), board overlay won't be included. We were hoping that snippets could be a remedy for that (as some use cases require always including `boards/<BOARD>.overlay`, so we wanted to add it to their snippets as board specific ones), but if splitting the snippets for example into `CI` and `low_power`  would be necessary, we think that we shouldn't add board specific overlays to one of them, but create another snippet. Which would come to  at least three snippets per use case here and would not really differ from current `-- -DOVERLAY_CONFIG="overlay-ci.conf;overlay-low_power.conf" -DDTC_OVERLAY_FILE="low_power.overlay"` approach.