zephyrproject-rtos/zephyr #57585: arch: arm64: smp: secondary cpu cannot boot if cluster id !=0

vkorenblit, 2023-05-05:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/57253

<div type='discussions-op-text'>

<sup>Originally posted by **vkorenblit** April 21, 2023</sup>
Platform: ARM Fast Models 2xCortex-A55 + GIC400
Zephyr release 3.3.0
App: samples/hello_world

Config:
SMP=y
SMP_BOOT_DELAY=n
CPU_MPIDR_LSB=8
CPU_MPIDR_WIDTH=8
CONFIG_MP_MAX_NUM_CPUS=2

Hello,

I have a custom platform with 2 clusters of 2xCortex-A55 each, the first with cluster id = 0 and the 2nd one with cluster id = 1.

This gives the following MPIDR_EL1 for each core:
0x0000000081000000	Cluster 0 core 0
0x0000000081000100	Cluster 0 core 1

0x0000000081010000	Cluster 1 core 0
0x0000000081010100	Cluster 1 core 1

If I run Zephyr on cluster 0 with the following cpu nodes in dts everything works as expected:
                cpu@0 {
                        device_type = "cpu";
                        compatible = "arm,cortex-a55";
                        reg = <0x0>;
                };
                cpu@1 {
                        device_type = "cpu";
                        compatible = "arm,cortex-a55";
                        reg = <0x1>;
                };

Console logs:
Secondary CPU core 1 (MPID:0x1) is up
Hello World! Custom board

On the other hand, if cluster id != 0 I see the following problem: secondary cpu will wait until it sees its cpu_id (aff1) in BOOT_PARAM_MPID_OFFSET. However, primary cpu does not store cpu_id but cpu_mpid which comes from dts (arch_start_cpu). My point is that we are not comparing the same thing.

The only ugly workaround I have for now is declaring the 2 cpu nodes as follows:

		cpu@0 {
			device_type = "cpu";
			compatible = "arm,cortex-a55";
			reg = <0x10000>;  //this will make https://github.com/zephyrproject-rtos/zephyr/blob/71dfaff583b00c721ec2bbb451d7bf2a099a0a4f/arch/arm64/core/smp.c#L71 true
		};

		cpu@1 {
			device_type = "cpu";
			compatible = "arm,cortex-a55";
			reg = <0x1>;
		};

I think the main issue is that on one side we wait for our cpu number (only aff 1 in my opinion) to be stored but the primary cpu stores the value in reg property which "should correspond" to MPIDR_TO_CORE which has a mask for aff3 aff2 aff1 and aff0.

In dts bindings I didn't find a specification about what we should write in cpu reg property for arm64

This workaround is enough to boot but breaks the scheduling because broadcast_ipi from core 1 won't work as we'll send an SGI to target_mpidr 0x1 because it will think in the device tree that this core is not itself. This results in reading 1024 in GICC_IAR, which is actually not true.

I think the easiest solution should be setting the real MPIDR_EL1 only with affinity bits in dts and secondary cpus should wait for this value (but read from their own register) instead of cpu_id in reset.S.  Haven't tested yet...

Best regards,

Valentin

</div>

carlocaione, 2023-05-05:
tagging @povergoing

povergoing, 2023-05-06:
It should work with `cluster id != 0` per my understanding.
>On the other hand, if cluster id != 0 I see the following problem: secondary cpu will wait until it sees its cpu_id (aff1) in BOOT_PARAM_MPID_OFFSET. However, primary cpu does not store cpu_id but cpu_mpid which comes from dts (arch_start_cpu). My point is that we are not comparing the same thing.

Secondary cpu will wait until it sees its `MPIDR` not cpu_id, which is set (stored) by the primary core: https://github.com/zephyrproject-rtos/zephyr/blob/71dfaff583b00c721ec2bbb451d7bf2a099a0a4f/arch/arm64/core/smp.c#L93 And cpu_id is not `aff1`, instead, it is just a logic id determined by booting sequence.

Do you mean the following DTS doesn't work?
```
cpu@10000 {
	device_type = "cpu";
	compatible = "arm,cortex-a55";
	reg = <0x10000>; 
};

cpu@10100 {
	device_type = "cpu";
	compatible = "arm,cortex-a55";
	reg = <0x10100>;
};
```
Could you please check if there are only two (listed by DTS) cores booted? Or do you have trustfirmware with PSCI service? What I want to make clear is, in terms of FVP, for simplicity, we usually don't use trustfirmware + PSCI but just let all cores boot at the same time. Thus the core that wins the rat race at the very early boot stage becomes the primary core.
If that is the case you have met, when you let four cores boot (say 0x0, 0x100, 0x10000, 0x10100), when you only configure two of them in DTS (Cluster 1: 0x10000, 0x10100), it unlikely works. Because one of the first two (in cluster0) probably becomes the primary core but Zephyr doesn't know them at all.

What's more, I remember there are people who use Zephyr with `cluster id != 0`, and seems it works well.

vkorenblit, 2023-05-06:
Again, old local commit's fault. My apologies

vkorenblit, 2023-05-09:
@povergoing The other cluster (id 0) is not running zephyr, only cluster 1 with these 2 cores is running zephyr. I have just tried with clean zephyr and effectively no core becomes the primary and both are spinning because no one jumps to primary_core.

Remark: I am not using trusted firmware nor psci.

carlocaione, 2023-05-09:
@vkorenblit so should we re-open this issue?

vkorenblit, 2023-05-09:
Yes, even if I mixed up some local stuff I think there is still a problem. It seems no core succeeds after stlxr instruction

povergoing, 2023-05-09:
Let me have a look. Is there a quick way that I can reproduce the issue?

> Now I see that the developer who modified this to get it working with ARM Fast Models faced this issue but never opened an issue upstream.

I am curious about how they solve the issue? or just a workaround?

vkorenblit, 2023-05-09:
I think in order to reproduce you can generate a fast model with 2xA55, it should be enough. A workaround, after getting cpu_id we get only aff1 bits and then if 0 we jump to primary and we stay in the loop otherwise.

povergoing, 2023-05-10:
I cannot reproduce the issue using `FVP_Base_Cortex-A55x2` or `FVP_Base_Cortex-A55x4`, I also tried only booting core 2 and core 3 with core 0, and 1 powered down.
But the weird behavior sounds like a cache-relevant issue. Could you please check your FVP parameter `cache_state_modelled`? I also encountered a problem when I set `-C cache_state_modelled=1`, if this is the case, I will try to fix it.

vkorenblit, 2023-05-10:
ldr works fine, but then stlxr doesn't write to memory. I'm checking with ARM DS 2022.2.

I'm not using FVP but a custom model. I am running linux kernel on one cluster so I believe the model is fine

vkorenblit, 2023-05-10:
To boot both cores I just replaced your code by a read to aff1 (because I don't have threading so no aff0) and cpu interfaces 0 and 1 of gic-400 are connected to core 0 and 1 :

``
ubfx x4, x1, #8, #8
cbz x4, primary_core
``

However, this is enough only to boot. After that, I have replaced aff0 inside broadcast_ipi by aff1 when raising SGIs and
in intc_gic.c I have the following change:

cpu_mask |= BIT(cpu_mpid_list[i]); is now cpu_mask =|BIT(MPIDR_AFFLVL(cpu_mpid_list[i], 1));

All these changes allow me to get irqs, however the scheduling seems strange when receiving an irq, I am still debugging so I think there is some stuff missing.

vkorenblit, 2023-05-10:
I agree, but is this instruction working at early boot on real silicon? Is this maybe an issue with FM only?

povergoing, 2023-05-11:
> I agree, but is this instruction working at early boot on real silicon? Is this maybe an issue with FM only?

We have tested it on FPGA.

jhedberg, 2023-09-12:
Since there hasn't been any update since May I'll close this with the assumption that it's fixed or there's a good workaround. Feel free to reopen if this still needs to be tracked as an open issue.

closed 2023-09-12 by jhedberg