zephyrproject-rtos/zephyr #57594: drivers: counter: Potential error in documentation

fgrandel, 2023-05-05:
**Describe the bug**

The current counter documentation (counter.h, line 115ff, struct counter_alarm_cfg) says:

```
Absolute alarm cannot be set further in future than top_value
decremented by the guard period. Relative alarm ticks cannot
exceed current top value (see @ref counter_get_top_value).

```
Isn't it exactly the other way round? An absolute alarm can be anything in the range between zero and top while a relative alarm must be less than (or equal) top minus guard to be well-defined.

I think the error was introduced by 1baae96af6573fb20de37dc9a76729b3d8b524f0 - before that the documentation was based on a different API but essentially had it the other way round.

If you agree, I can provide a PR.

**Expected behavior**

I propose something along the lines of:

`An absolute alarm can be anything between zero and the current top value (see @ref counter_get_top_value). Relative alarm ticks cannot be set further in future than top_value decremented by the guard period.`

fgrandel, 2023-05-05:
I think I'm mistaken on that one - sorry for the noise. Closing. I got confused.

fgrandel, 2023-05-05:
Reopening with a proposal for an improved API documentation.

henrikbrixandersen, 2023-05-08:
@fgrandel Thank you for submitting this. Please submit a PR with a suggestion for updated documentation for this. Discussing proposed changes like this is much easier done in a PR.

fgrandel, 2023-05-08:
> Please submit a PR

@henrikbrixandersen Done

PR #57676 (2023-05-22): drivers: counter: improved guard period doc
The previous counter documentation said:

> Absolute alarm cannot be set further in future than top_value
> decremented by the guard period. Relative alarm ticks cannot
> exceed current top value (see @ref counter_get_top_value).

And further down:

> Absolute value cannot exceed: (now + top_value - guard_period)
> % top_value.

The first documentation snippet is not wrong but may be confusing. On casual reading one might assume that the absolute tick value should not be set above top minus guard. Users have to understand driver internals to follow t [... cut]
commit 693edd06 (2023-05-22): drivers: counter: improved guard period doc

The previous counter documentation said:

> Absolute alarm cannot be set further in future than top_value
> decremented by the guard period. Relative alarm ticks cannot
> exceed current top value (see @ref counter_get_top_value).

And further down:

> Absolute value cannot exceed: (now + top_value - guard_period)
> % top_value.

The first documentation snippet is not wrong but may be confusing.
On casual reading one might assume that the absolute tick value
should not be set above top minus guard. Users have to understand
driver internals to follow  [... cut]
closed 2023-05-22 by carlescufi