zephyrproject-rtos/zephyr #57631: PWM Capture on STM32 doesn't function on Channel 2

heinwessels, 2023-05-06:
**Describe the bug**
The current implementation of PWM capture on the STM32 only works for timer channel 1, while it's designed to work for channel 1 and 2. When channel 2 is selected the capture gets stuck with the ISR constantly being triggered forever.

Some symptoms I observed, although it's hard to say which values correct:
- The `CNT` register never gets higher than about 20, it's always reset by Slave Mode Controller almost immediatelly.
- The two capture channels (1 and 2) captures these small counter values, usually `0` or `1`.

**To Reproduce**
Steps to reproduce the behavio on Nucleo H743zi. 
1. Apply the following patch to change the `pwm_loopback` test to use Timer 5 Channel 2 for PWM Capture. **Note:** The same happens when there's a valid input on the capture pin as well.
```patch
diff --git a/tests/drivers/pwm/pwm_loopback/boards/nucleo_h743zi.overlay b/tests/drivers/pwm/pwm_loopback/boards/nucleo_h743zi.overlay
index 5007b962ec..275f6d6398 100644
--- a/tests/drivers/pwm/pwm_loopback/boards/nucleo_h743zi.overlay
+++ b/tests/drivers/pwm/pwm_loopback/boards/nucleo_h743zi.overlay
@@ -11,7 +11,7 @@
                compatible = "test-pwm-loopback";
                /* first index must be a 32-Bit timer */
                pwms = <&pwm2 4 0 PWM_POLARITY_NORMAL>,
-                       <&pwm5 1 0 PWM_POLARITY_NORMAL>;
+                       <&pwm5 2 0 PWM_POLARITY_NORMAL>;
        };
 };
 
@@ -29,7 +29,7 @@
        status = "okay";
        pwm5: pwm {
                status = "okay";
-               pinctrl-0 = <&tim5_ch1_pa0>; /* CN11 PIN28 */
+               pinctrl-0 = <&tim5_ch2_pa1>; /* CN11 PIN28 */
                pinctrl-names = "default";
        };
 };
```
2. Connect the pwm capture pin (A1) to ground.
3. `west build -p always -b nucleo_h743zi tests/drivers/pwm/pwm_loopback/`
4. `west flash`
5. See following output, which is a result of the ISR being called repeatedly
```
*** Booting Zephyr OS build zephyr-v3.3.0-3460-gb10817ba25b2 ***
Running TESTSUITE pwm_loopback
===================================================================
START - test_capture_busy
E: counter overflow during PWM capture

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/pwm/pwm_loopback/src/test_pwm_loopback.c:322: p)
pwm capture not busy (err 0)
 FAIL - test_capture_busy in 0.021 seconds
===================================================================
START - test_capture_timeout
E: counter overflow during PWM capture
```
**Note:** This is where the execution gets stuck and the tests doesn't progress furthur.

**Expected behavior**
The tests to executing normally by timing out. For example, here is the output if I make no changes to the test, except grounding the capture pin. Meaning this is capturing on timer channel 1. It's clear the capture functions as expected, as gracefully times out.
```
*** Booting Zephyr OS build zephyr-v3.3.0-3460-gb10817ba25b2 ***
Running TESTSUITE pwm_loopback
===================================================================
START - test_capture_busy
E: PWM Capture already in progress
E: PWM capture already active
 PASS - test_capture_busy in 0.006 seconds
===================================================================
START - test_capture_timeout
W: pwm capture timed out
 PASS - test_capture_timeout in 1.003 seconds
===================================================================
START - test_continuous_capture

... not posting the entire log ...

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Impact**
Showstopper. It still persists in #57607 which is important for our project. I'm trying to debug why it's happening, but haven't had any luck yet.

**Environment (please complete the following information):**
 - OS:Ubuntu 22.04
 - SDK 0.16
 - b10817ba25b25f8a1b567234010f127cffaa34b6

FRASTM, 2023-05-16:
Is it due to timer 5 ? 
How is the PWM input capture chosen for the test case ?

On my nucleo_h743zi board, I configured 
```
	pwm_loopback_0 {
		compatible = "test-pwm-loopback";
		/* first index must be a 32-Bit timer */
		pwms = <&pwm2 2 0 PWM_POLARITY_NORMAL>,
			<&pwm5 1 0 PWM_POLARITY_NORMAL>;
```
timer2 channel 2 on pb3 (<&tim2_ch2_pb3>; /* CN7 PIN15 D23 */) and let  timer5 ch1 on PA0 (<&tim5_ch1_pa0>; /* CN10 PIN29 D32 */)

heinwessels, 2023-05-17:
> Is it due to timer 5 ? How is the PWM input capture chosen for the test case ?

Thanks for having a look! I was actually also looking into this yesterday, and saw the same. With the following inputs the tests passed for me.
```
Timer 2 Channel 2
Timer 2 Channel 4
Timer 3 Channel 2 (some tests failed, but it's because it's a 16-bit timer, and the tests are designed for 32-bit)
```
I've only had it fail for Timer 5 on Channel 2 and 4. Can you confirm that it fails for you too? If so it means it could possibly be a bug in the LL drivers?

heinwessels, 2023-05-17:
> Well, not sure about the order in

The first pwm is the output, meaning the PWM generator, and the second is the input, or PWM capture, which we are testing. So you need to set the second one to TImer 5 Channel 2. There's a diff on how to achieve this in the issue description.

FRASTM, 2023-05-17:
The pb seems specific to Timer 5 as With TImer 4 capture on channel 1 or 2, test passed
Note that I also have set the 	`st,prescaler = <255>; ` for each Timer

heinwessels, 2023-05-30:
Yeah, @FRASTM seems to be correct.

closed 2023-05-30 by heinwessels