zephyrproject-rtos/zephyr #57638: HAL Module Request: hal_hisilicon

xingrz, 2023-05-08:
## Origin

https://github.com/xingrz/hal_hisilicon

## Purpose

This repository provides source of components support for the HiSilicon Hi3861 SoC.

## Mode of integration

Please integrate this project as a module according to the generic Zephyr HAL modules structure.

## Maintainership

* @xingrz 

## Pull Request

#57363

## Description

I am uncertain whether this repository should be considered a "HAL module." It contains sources for two components:

* **loaderboot** - The source of a bootable image that was imported from the original SDK. This component is required by the hiburn runner that I have [implemented](https://github.com/zephyrproject-rtos/zephyr/pull/57363/commits/95c28da24b6742effe5c6bfbe6f83661cadf80dc) in the PR above. To flash the board, this image will be sent to and run on the RAM of the SoC. It will receive data sent from the `hiburn` on the host side and write it onto the flash.
* **flashboot** - This is a few lines of bare metal code that I wrote. The fixed romcode of the Hi3861 always expects the entry to be located in RAM instead of Flash. The only purpose of this image is to bridge this gap by jumping to the Zephyr kernel on the Flash.

Although the manufacturer of this SoC offers an official [LiteOS SDK](https://gitee.com/openharmony/device_soc_hisilicon/tree/master/hi3861v100/sdk_liteos), I did not use it in its entirety. Instead, I selectively chose the loaderboot from it.

Furthermore, all the drivers have been implemented in accordance with the user manual from HiSilicon, so no headers from the original SDK were imported into this repository.

## Dependencies

None.

## Revision

https://github.com/xingrz/hal_hisilicon/tree/73eb704ff40f4da93fbbebdf77ef3460f027d088

## License

Apache-2.0

henrikbrixandersen, 2023-05-08:
> ## License
> Apache-2.0

This is not correct. At least some of the files (i.e. `hi3861/boot/flashboot/riscv_init.S` are under 3-clause BSD.

xingrz, 2023-05-08:
> > ## License
> > Apache-2.0
> 
> This is not correct. At least some of the files (i.e. `hi3861/boot/flashboot/riscv_init.S` are under 3-clause BSD.

You're right. But the original LiteOS SDK says it was released under Apache-2.0 License.

https://gitee.com/openharmony/device_soc_hisilicon/tree/master/hi3861v100/sdk_liteos/license

xingrz, 2023-05-08:
> You're right. But the original LiteOS SDK says it was released under Apache-2.0 License.
>
> https://gitee.com/openharmony/device_soc_hisilicon/tree/master/hi3861v100/sdk_liteos/license

Oh, I see what the issue is now. The manufacturer has released their code from multiple sources. Initially, they provided zip files on their website, but now they release the code on Gitee, where all sources have been re-licensed with Apache 2.0. I will update my repository with the latest sources.

xingrz, 2023-05-08:
> > ## License
> > Apache-2.0
> 
> This is not correct. At least some of the files (i.e. `hi3861/boot/flashboot/riscv_init.S` are under 3-clause BSD.

I just updated and switched to the OpenHarmony branch. Now the code is definitely licensed under Apache 2.0.

keith-zephyr, 2023-05-12:
Can you contribute the `flashboot` directly into the Zephyr tree?  This code is your own, correct?

The `loaderboot` code probably makes sense as a module, but it's definitely not a HAL.

xingrz, 2023-05-13:
> Can you contribute the `flashboot` directly into the Zephyr tree? This code is your own, correct?

Yes, but I'm not sure which directory I should place it in.

henrikbrixandersen, 2023-05-13:
> Yes, but I'm not sure which directory I should place it in.

Perhaps a subdirectory to the soc directory?

keith-zephyr, 2023-05-15:
@xingrz - Can you drop the "hal" from the proposed module name?  How about `hisilicon_loaderboot` or just `hisilicon`?

xingrz, 2023-05-16:
> @xingrz - Can you drop the "hal" from the proposed module name? How about `hisilicon_loaderboot` or just `hisilicon`?

Sure. However, without the "hal" prefix, is it appropriate to use this repository to declare the west blobs required for WiFi in the future?

keith-zephyr, 2023-05-23:
> > @xingrz - Can you drop the "hal" from the proposed module name? How about `hisilicon_loaderboot` or just `hisilicon`?
> 
> Sure. However, without the "hal" prefix, is it appropriate to use this repository to declare the west blobs required for WiFi in the future?

Note that blobs are not permitted to be hosted in the Zephyr main tree or modules. See details [here](https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#hosting).  However the module can be used to reference the blob hosted somewhere else.

Please use  `hisilicon` as the module name.  This allows flexibility to add additional code from HiSilicon in the future.

Once this issue gets updated, I'll add the TSC label so that this issue gets voted on by the TSC.  Assuming it's approved, then maintainers will create the `zephyrproject-rtos/hisilcon`.

xingrz, 2023-05-23:
> Note that blobs are not permitted to be hosted in the Zephyr main tree or modules. See details [here](https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#hosting). However the module can be used to reference the blob hosted somewhere else.

Yes, I'm aware of that.

I've updated the title and the description of this issue.

keith-zephyr, 2023-05-30:
The TSC noticed that there are some pre-built libraries/object files in the proposed module.
https://github.com/xingrz/zephyr_module_hisilicon/tree/master/hi3861/loaderboot/libs

If these libraries are not needed, please delete from the module.  If the libraries are required, you'll need to follow the process for adding binary blobs.

xingrz, 2023-05-31:
> If these libraries are not needed, please delete from the module. If the libraries are required, you'll need to follow the process for adding binary blobs.

I understand the concern. These libs are indeed necessary. However, it's important to note that these libs will not be compiled into the Zephyr binary. The loaderboot.bin itself is similar to FLM files used during JLink flash.

While it is possible to reimplement a pure open-source alternative, it would require a significant amount of time.

Instead, would it be feasible if I provide the loaderboot.bin separately in another location, such as my own GitHub repo, telling the developers that they need to obtain this additional bin file for flashing in the README for the board?

carlocaione, 2023-06-30:
is there any progress on this discussion?

nashif, 2023-07-12:
Marking as rejected by the TSC due to the binaries in the module.

closed 2023-08-15 by nashif