zephyrproject-rtos/zephyr #57646: Intel CAVS 25 tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported fails.

gbarkadiusz, 2023-05-08:
**Describe the bug**
There are three failing tests during building.
`tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32`
`tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all`
`tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256`

**To Reproduce**
`twister -W --hardware-map ${HOME}/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/ -v`

**Expected behavior**
Passes.

**Impact**
Annoyance.

**Logs and console output**
```
Renaming output directory to /home/ztest/zephyrproject/zephyr/twister-out.5
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.3.0-1778-g1851950977d5
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/testplan.json

Device testing on:

| Platform          | ID   | Serial device   |
|-------------------|------|-----------------|
| intel_adsp_cavs25 | None |                 |

INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/3 intel_adsp_cavs25         tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all  ERROR Build failure (device)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all/build.log
INFO    - 2/3 intel_adsp_cavs25         tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256  ERROR Build failure (device)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256/build.log
INFO    - 3/3 intel_adsp_cavs25         tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32  ERROR Build failure (device)
ERROR   - see: /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32/build.log

INFO    - 3 test scenarios (3 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 3 test configurations passed (0.00%), 0 failed, 3 errored, 0 skipped with 0 warnings in 3.31 seconds
INFO    - In total 3 test cases were executed, 0 skipped on 1 out of total 555 platforms (0.18%)
INFO    - 0 test configurations executed on platforms, 3 test configurations were only built.

Hardware distribution summary:

| Board             | ID   |   Counter |
|-------------------|------|-----------|
| intel_adsp_cavs25 | None |         0 |
INFO    - Saving reports...
INFO    - Writing JSON report /home/ztest/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/ztest/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.crc32 on intel_adsp_cavs25 error (Build failure)
INFO    - 2) tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256 on intel_adsp_cavs25 error (Build failure)
INFO    - 3) tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all on intel_adsp_cavs25 error (Build failure)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p intel_adsp_cavs25 -s tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all
INFO    - or with west:
INFO    - west build -p -b intel_adsp_cavs25 -T tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.all
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```
And build.log contains.

```
...
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ztest/zephyr-sdk-0.16.0/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc
CMake Warning at /home/ztest/zephyrproject/zephyr/CMakeLists.txt:828 (message):
  No SOURCES given to Zephyr library: lib__libc__common

  Excluding target from build.

-- Configuring done
-- Generating done
...

FAILED: CMakeFiles/app.dir/src/main.c.obj
ccache /home/ztest/zephyr-sdk-0.16.0/xtensa-intel_s1000_zephyr-elf/bin/xtensa-intel_s1000_zephyr-elf-gcc -DKERNEL -DTC_RUNID=f9cc00c812124f5b03a196ef68e53943 -D__ZEPHYR__=1 -I/home/ztest/zephyrproject/zephyr/subsys/mgmt/mcumgr/transport/include -I/home/ztest/zephyrproject/zephyr/include -I/home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256/zephyr/include/generated -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/cavs -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/cavs/include -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/cavs/include/intel_tgl_adsp -I/home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/common/include -I/home/ztest/zephyrproject/zephyr/subsys/mgmt/mcumgr/util/include -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/include -I/home/ztest/zephyrproject/zephyr/subsys/testsuite/ztest/include -I/home/ztest/zephyrproject/modules/hal/xtensa/include -I/home/ztest/zephyrproject/modules/hal/xtensa/zephyr/soc/intel_tgl_adsp -I/home/ztest/zephyrproject/modules/crypto/tinycrypt/lib/include -isystem /home/ztest/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/ztest/zephyr-sdk-0.16.0/xtensa-intel_s1000_zephyr-elf/bin/../lib/gcc/xtensa-intel_s1000_zephyr-elf/12.2.0/include -isystem /home/ztest/zephyr-sdk-0.16.0/xtensa-intel_s1000_zephyr-elf/bin/../lib/gcc/xtensa-intel_s1000_zephyr-elf/12.2.0/include-fixed -fno-strict-aliasing -Os -imacros /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/mgmt.mcumgr.fs.mgmt.hash.supported.sha256/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always --sysroot=/home/ztest/zephyr-sdk-0.16.0/xtensa-intel_s1000_zephyr-elf/xtensa-intel_s1000_zephyr-elf -imacros /home/ztest/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ztest/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/ztest/zephyrproject/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/src/main.c
In file included from /home/ztest/zephyrproject/zephyr/include/zephyr/mgmt/mcumgr/mgmt/mgmt.h:13,
                 from /home/ztest/zephyrproject/zephyr/tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/src/main.c:11:
/home/ztest/zephyrproject/zephyr/include/zephyr/mgmt/mcumgr/smp/smp.h:31:10: fatal error: zcbor_common.h: No such file or directory
   31 | #include <zcbor_common.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```
**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0
 - West: v1.0.0
 - Latest main [4694b15](https://github.com/zephyrproject-rtos/zephyr/commit/4694b156054e1d23e8385edc84157a247defe24f)

nordicjm, 2023-05-09:
Builds fine here using `main`:
```
./scripts/twister -T tests/subsys/mgmt/mcumgr/fs_mgmt_hash_supported/ -p intel_adsp_cavs25
ZEPHYR_BASE unset, using "/tmp/aa/zephyr"
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.3.0-3631-g81c979502354
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /tmp/aa/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    3/   3  100%  skipped:    0, failed:    0, error:    0
INFO    - 3 test scenarios (3 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 3 of 3 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 27.31 seconds
INFO    - In total 3 test cases were executed, 0 skipped on 1 out of total 567 platforms (0.18%)
INFO    - 0 test configurations executed on platforms, 3 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /tmp/aa/zephyr/twister-out/twister.json
INFO    - Writing xunit report /tmp/aa/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /tmp/aa/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

nordicjm, 2023-05-09:
Closing as issue is not repducable.

closed 2023-05-09 by nordicjm

caco3, 2023-06-02:
I have the same issue using an nrf52 board.

My project works fine with Zephyr 3.3, but with 3.4-RC1 I get this error:
```bash
In file included from /opt/zephyrproject-v3.4.0-rc1/zephyr/include/zephyr/mgmt/mcumgr/mgmt/mgmt.h:13,
                 from /opt/zephyrproject-v3.4.0-rc1/zephyr/include/zephyr/mgmt/mcumgr/grp/img_mgmt/img_mgmt.h:12,
                 from /home/me/rcd-sensor/src/bt_dfu.c:15:
/opt/zephyrproject-v3.4.0-rc1/zephyr/include/zephyr/mgmt/mcumgr/smp/smp.h:31:10: fatal error: zcbor_common.h: No such file or directory
   31 | #include <zcbor_common.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
[41/340] Building C object CMakeFiles/app.dir/src/peripherals.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/me/rcd-sensor_build
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/me/rcd-sensor
-- CMake version: 3.21.1
-- Cache files will be written to: /home/me/.cache/zephyr
-- Zephyr version: 3.4.0-rc1 (/opt/zephyrproject-v3.4.0-rc1/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.14.0")
CMake Error at /opt/zephyrproject-v3.4.0-rc1/zephyr/cmake/modules/zephyr_module.cmake:77 (message):
  ERROR: ManifestImportFailed: project <Project bsim
  ('/opt/zephyrproject-v3.4.0-rc1/tools/bsim') at
  908ffde6298a937c6549dbfa843a62caab26bfc5> value west.yml

Call Stack (most recent call first):
  /opt/zephyrproject-v3.4.0-rc1/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  /opt/zephyrproject-v3.4.0-rc1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /opt/zephyrproject-v3.4.0-rc1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

```

nordicjm, 2023-06-05:
@caco3 that is https://github.com/zephyrproject-rtos/zephyr/issues/57198

caco3, 2023-06-05:
@nordicjm  Thanks!