zephyrproject-rtos/zephyr #57647: zephyr_get(... MERGE) ignores local scoped variables

tejlmand, 2023-05-08:
**Describe the bug**
Already defined local scoped variables are not considered with `zephyr_get(... MERGE)`.

For example the variable set in below line is being cleared by `zephyr_get(.... MERGE)`:
https://github.com/zephyrproject-rtos/zephyr/blob/3bf7f83f213f96daa9a17cd3ddd4cff0a200bb9d/samples/application_development/out_of_tree_board/CMakeLists.txt#L6

**To Reproduce**
Build https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/application_development/out_of_tree_board with trace output on and notice that the local scoped variable is cleared.
*note*: due to Zephyr also adding the current app dir as root search path, then no errors are immediately seen.

**Expected behavior**
The local scoped variable should be included in the list.

**Impact**
This is an oot of tree example, so users following the approach presented but with other paths may experience that things are not working.

**Logs and console output**
none

**Environment (please complete the following information):**
N/A

**Work around**
Set the variable as cache variable using `set(<var> CACHE PATH "")`, however this has the disadvantage of manipulating the cache and thus potentially conflict with user provided cache values.

nordicjm, 2023-05-09:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57583

tejlmand, 2023-05-10:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57583

closed 2023-05-10 by tejlmand