zephyrproject-rtos/zephyr #57651: Frdm_k64f tests/net/lib/mqtt_sn_client/net.mqtt_sn.client fails

gbarkadiusz, 2023-05-08:
**Describe the bug**
This testcase contains three tests.
```
mqtt_sn_client.test_mqtt_sn_connect_no_will
mqtt_sn_client.test_mqtt_sn_connect_will
mqtt_sn_client.test_mqtt_sn_publish_qos0
```
but only the last one is failing.
```
TESTSUITE mqtt_sn_client failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  66.67% [mqtt_sn_client]: pass = 2, fail = 1, skip = 0, total = 3 duration = 1.035 seconds
- PASS - [mqtt_sn_client.test_mqtt_sn_connect_no_will] duration = 0.254 seconds
- PASS - [mqtt_sn_client.test_mqtt_sn_connect_will] duration = 0.299 seconds
- FAIL - [mqtt_sn_client.test_mqtt_sn_publish_qos0] duration = 0.482 seconds
------ TESTSUITE SUMMARY END ------
```

**To Reproduce**
```
twister --device-testing --hardware-map <path_to>/map.yml -T ./tests/net/lib/mqtt_sn_client/ -v
```

**Expected behavior**
Passes.

**Impact**
Annoyance.

**Logs and console output**
```
Renaming output directory to <path_to>/twister-out.9
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.3.0-3578-g1f43e126a841
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report <path_to>/twister-out/testplan.json

Device testing on:

| Platform   | ID                                               | Serial device                                                                                     |
|------------|--------------------------------------------------|---------------------------------------------------------------------------------------------------|
| frdm_k64f  | 02400b0132044e4500417009997b00316781000097969900 | /dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_02400b0132044e4500417009997b00316781000097969900-if01 |
| frdm_k64f  | 02400b0140214e45001b1018c89200349e11000097969900 | /dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_02400b0140214e45001b1018c89200349e11000097969900-if01 |

INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 frdm_k64f                 tests/net/lib/mqtt_sn_client/net.mqtt_sn.client     FAILED Failed (device 6.265s)
ERROR   - see: <path_to>/twister-out/frdm_k64f/tests/net/lib/mqtt_sn_client/net.mqtt_sn.client/handler.log

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 errored, 0 skipped with 0 warnings in 13.67 seconds
INFO    - In total 3 test cases were executed, 0 skipped on 1 out of total 567 platforms (0.18%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board     | ID                                               |   Counter |
|-----------|--------------------------------------------------|-----------|
| frdm_k64f | 02400b0132044e4500417009997b00316781000097969900 |         1 |
| frdm_k64f | 02400b0140214e45001b1018c89200349e11000097969900 |         0 |
INFO    - Saving reports...
INFO    - Writing JSON report<path_to>/twister-out/twister.json
INFO    - Writing xunit report <path_to>/twister-out/twister.xml...
INFO    - Writing xunit report <path_to>/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/net/lib/mqtt_sn_client/net.mqtt_sn.client on frdm_k64f failed (Failed)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p frdm_k64f -s tests/net/lib/mqtt_sn_client/net.mqtt_sn.client
INFO    - or with west:
INFO    - west build -p -b frdm_k64f -T tests/net/lib/mqtt_sn_client/net.mqtt_sn.client
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```
but the handler.log contains more information.
```
...
FAIL - test_mqtt_sn_publish_qos0 in 0.482 seconds
===================================================================
TESTSUITE mqtt_sn_client failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  66.67% [mqtt_sn_client]: pass = 2, fail = 1, skip = 0, total = 3 duration = 1.035 seconds
- PASS - [mqtt_sn_client.test_mqtt_sn_connect_no_will] duration = 0.254 seconds
- PASS - [mqtt_sn_client.test_mqtt_sn_connect_will] duration = 0.299 seconds
- FAIL - [mqtt_sn_client.test_mqtt_sn_publish_qos0] duration = 0.482 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 0d2a5692968da714fabde3d9820c1cc5
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Linux
 - Target: Frdm_k64f
 - Toolchain: Zephyr SDK 0.16.0
 - West version: v0.13.1
 - Latest main [84ba03a](https://github.com/zephyrproject-rtos/zephyr/commit/84ba03ae3fbbc71729a434d9fe888d739d545771)

gbarkadiusz, 2023-05-10:
According to the handler.log file.
There is problem with `msg_send` function.
```
Assertion failed at WEST_TOPDIR/zephyr/tests/net/lib/mqtt_sn_client/src/mqtt_sn_client.c:40: assert_msg_send: (msg_send_data.called not equal to called)
msg_send called 0 times instead of 1
```

hakehuang, 2023-05-16:
I can reproduce this issue.

gbarkadiusz, 2023-05-16:
@hakehuang 
So, could you explain to me how to update Zephyr to your version?
After `git checkout main` and `git pull`, I've got at most `zephyr-v3.3.0-3980-g7bdae081e3`.

hakehuang, 2023-05-16:
@gbarkadiusz , you mean zephyr-v3.3.0-572-g04386be57e8a?
`git checkout -b temp zephyr-v3.3.0-572-g04386be57e8a`

gbarkadiusz, 2023-05-16:
@hakehuang I thought that this two commands 
```
git checkout main
git pull
```
are enough to get the latest zephyr version.

gbarkadiusz, 2023-05-16:
According to the `twisterlib` the Zephyr version is created by `git describe` command.
This command returns the version of the project according to these roles.
`<tag><number_of_commits_since_the_tag><abbreviated_commit_hash>`
So, my reported bug occurred with a newer Zephyr version.

hakehuang, 2023-05-16:
> According to the `twisterlib` the Zephyr version is created by `git describe` command. This command returns the version of the project according to these roles. `<tag><number_of_commits_since_the_tag><abbreviated_commit_hash>` So, my reported bug occurred with a newer Zephyr version.

@gbarkadiusz thanks for this clarify. I leave the ok commit is just to help to use git bisectto find the first failure commit. This issue is  a real issue.  I am checking the root cause. will feedback once found. Thanks for quick follow up

dleach02, 2023-05-17:
@hakehuang, the commit you flagged is really just changing the default logging level if one is not set. So maybe a log level change is not exposing the problem? e.g. if you did the same logging level in an earlier point it would also fail?

hakehuang, 2023-05-18:
> @hakehuang, the commit you flagged is really just changing the default logging level if one is not set. So maybe a log level change is not exposing the problem? e.g. if you did the same logging level in an earlier point it would also fail?

yes.

gbarkadiusz, 2023-05-18:
@hakehuang I've got the same result of bisect session.
```
6fe83ea810ca35ca21bc3bc460dbc0fde332a559 is the first bad commit
commit 6fe83ea810ca35ca21bc3bc460dbc0fde332a559
Author: Kamil Kasperczyk <kamil.kasperczyk@nordicsemi.no>
Date:   Tue Apr 4 14:08:30 2023 +0200

    logging: Changed log template to consider default log level
    
    Added using LOG_DEFAULT_LEVEL kconfig option in log module
    template for assigning the default log level to newly created
    module instance, instead of assuming always INFO level.
    
    Signed-off-by: Kamil Kasperczyk <kamil.kasperczyk@nordicsemi.no>

 subsys/logging/Kconfig.template.log_config | 6 +++++-
 1 file changed, 5 insertions(+), 1 deletion(-)
```

dleach02, 2023-05-22:
can we try this test after bumping the stack sizes? Or try putting CONFIG_TEST_EXTRA_STACK_SIZE to 1024 to see if it changes the behavior here?

hakehuang, 2023-05-22:
> mqtt_sn_client

same failure

```
D: Program MPU region at index 0x6
D: [06] 0x200065c0 0x200065ff 0x00208208 0x00000001
D: disable region 0x7
D: disable region 0x8
D: disable region 0x9
D: disable region 0xa
D: disable region 0xb

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/lib/mqtt_sn_client/src/mqtt_sn_client.c:40: assert_msg_send: (msg_send_data.called not equal to called)
msg_send called 0 times instead of 1

```

gbarkadiusz, 2023-07-24:
It shouldn't be signed as stale, because mqtt_sn_client still fails.

dleach02, 2024-02-02:
I have not been able to reproduce the reported problem

closed 2024-02-02 by dleach02