zephyrproject-rtos/zephyr #57655: STM32WL SOC missing LOG_MODULE_REGISTER

AnachronisticPenguin, 2023-05-08:
**Describe the bug**

The STM32WL SOC functions in `soc/arm/st_stm32/stm32wl`  are missing a call to register the module with the logging module. While the code in `soc.c` does not make any calls to the logging system, the power state specific code in `power.c` does make calls at the DEBUG level.

This manifests as a compile error for any program that uses the STM32WLxx SOC, enables the `PM` module, and is compiled with a logging level of `DEBUG`, i.e., `CONFIG_LOG_DEFAULT_LEVEL=4`

**To Reproduce**

1) Start a new project with the following `prj.conf` file:

```
CONFIG_LOG=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_PM=y
```

2) And the following `src/main.c` file:

```
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#define LOG_LEVEL CONFIG_LOG_DEFAULT_LEVEL
LOG_MODULE_REGISTER(log_tester);

int main(void){
        return 0;
}
```

3) Attempt to compile the program for a board that uses the STM32WL SOC:

```
$ west build -b nucleo_wl55jc . -p
```

4) See the following error:

```
[157/167] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/oliver/zephyrproject/zephyr/samples/stm32wl55jc_soc_logging_error/build/zephyr/zephyr_pre0.map 
: && ccache /home/oliver/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/oliver/zephyrproject/zephyr/samples/stm32wl55jc_soc_logging_error/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/oliver/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/home/oliver/zephyrproject/zephyr/samples/stm32wl55jc_soc_logging_error/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && cd /home/oliver/zephyrproject/zephyr/samples/stm32wl55jc_soc_logging_error/build/zephyr && /usr/bin/cmake -E echo
/home/oliver/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(power.c.obj): in function `pm_state_set':
/home/oliver/zephyrproject/zephyr/soc/arm/st_stm32/stm32wl/power.c:58: undefined reference to `log_const_soc'
/home/oliver/zephyr-sdk-0.16.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(power.c.obj): in function `pm_state_exit_post_ops':
/home/oliver/zephyrproject/zephyr/soc/arm/st_stm32/stm32wl/power.c:102: undefined reference to `log_const_soc'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/oliver/zephyrproject/zephyr/samples/stm32wl55jc_soc_logging_error/build
```

**Expected behavior**
I expect the program to compile successfully.

**Impact**
Unable to use `DEBUG` level logging on any application that uses the `PM` module and an STM32WLxx SOC.

**Environment:**
 - OS: Linux
 - Toolchain:  Zephyr SDK
 - Commit SHA: `commit 6c93cbf7b496229172fa9508b561417bf91df0ae` of `main` branch.

**Additional context**
I have a patch that fixes this by adding a call to `LOG_MODULE_REGISTER` to `soc.c`. I am testing it and will submit a pull request once I'm certain that it works.

erwango, 2023-05-09:
> I have a patch that fixes this by adding a call to LOG_MODULE_REGISTER to soc.c. I am testing it and will submit a pull request once I'm certain that it works.

That should be the good thing to do. Don't hesitate to submit once you're ready!

AnachronisticPenguin, 2023-05-11:
Status update for those following along at home. I applied the following patch, which resolved the compilation error.

```
+++ b/soc/arm/st_stm32/stm32wl/soc.c
@@ -18,6 +18,12 @@
 
 #include <stm32wlxx_ll_system.h>
 
+#include <zephyr/logging/log.h>
+
+#define LOG_LEVEL CONFIG_SOC_LOG_LEVEL
+LOG_MODULE_REGISTER(soc);
+
+
 /**
  * @brief Perform basic hardware initialization at boot.
  *
-- 
```

However, I got a stack overflow when running the (incredibly simple) program listed above logging set to level 4 (`DEBUG`); there is no stack overflow at lower logging levels. The console output is:

```
[00:00:00.000,000] <dbg> os: setup_thread_stack: stack 0x20001600 for thread 0x200006f8: obj_size=1088 buf_start=0x20001640  buf_size 1024 stack_ptr=0x20001a40
[00:00:00.000,000] <dbg> os: setup_thread_stack: stack 0x200014c0 for thread 0x20000678: obj_size=320 buf_start=0x20001500  buf_size 256 stack_ptr=0x20001600
[00:00:00.000,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_init: [2] 0x20001600 0x150b000a
[00:00:00.000,000] <dbg> os: setup_thread_stack: stack 0x20000940 for thread 0x200001b8: obj_size=832 buf_start=0x20000980  buf_size 768 stack_ptr=0x20000c80
*** Booting Zephyr OS build zephyr-v3.3.0-3137-g4691835645b8 ***
[00:00:00.000,000] <dbg> os: k_sched_unlock: scheduler unlocked (0x200006f8:0)
[00:00:00.000,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_init: [2] 0x20001600 0x150b000a
[00:00:00.000,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x2
[00:00:00.000,000] <dbg> mpu: region_init: [2] 0x20000940 0x150b000a
[00:00:00.128,000] <dbg> mpu: mpu_configure_region: Configure MPU region at index 0x2
[00:00:00.128,000] <dbg> mpu: region_allocate_and_init: Program MPU region at index 0x2
[00:00:00.128,000] <dbg> mpu: region_init: [2] 0x200014c0 0x150b000a
[00:00:00.129,000] <err> os: ***** MPU FAULT *****
[00:00:00.129,000] <err> os:   Stacking error (context area might be not valid)
[00:00:00.129,000] <err> os:   Data Access Violation
[00:00:00.129,000] <err> os:   MMFAR Address: 0x200014e0
[00:00:00.129,000] <err> os: r0/a1:  0x200014c8  r1/a2:  0x0000000c  r2/a3:  0x200015b0
[00:00:00.129,000] <err> os: r3/a4:  0x08002df1 r12/ip:  0x200014c0 r14/lr:  0xfffffffd
[00:00:00.129,000] <err> os:  xpsr:  0x14d8f400
[00:00:00.129,000] <err> os: Faulting instruction address (r15/pc): 0x553fc541
[00:00:00.129,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.129,000] <err> os: Current thread: 0x20000678 (unknown)
[00:00:00.223,000] <err> os: Halting system
```

Thread `0x20000678` is the idle thread. This runtime stack overflow is resoved by increasing the idle thread stack size in `prj.conf`:

```
CONFIG_IDLE_STACK_SIZE=1024
```

I have submitted a [pull request](https://github.com/zephyrproject-rtos/zephyr/pull/57782) to apply the patch.

closed 2023-05-15 by fabiobaltieri