zephyrproject-rtos/zephyr #57657: Frdm_k64f samples/subsys/usb/mass/ fails.

gbarkadiusz, 2023-05-08:
**Describe the bug**
The sample.yml contains three samples.
```
sample.usb.mass_ram_none  
sample.usb.mass_ram_fat 
sample.usb_device_next.mass_ram_none
```
but the only one fails.
`sample.usb_device_next.mass_ram_none`

**To Reproduce**
`twister --device-testing --hardware-map <path_to>/map.yml -T ./samples/subsys/usb/mass/ -v`

**Expected behavior**
Passes.

**Impact**
Annoyance.

**Logs and console output**
```
Renaming output directory to <path_to>/twister-out.12
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.3.0-3578-g1f43e126a841
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report <path_to>/twister-out/testplan.json

Device testing on:

| Platform   | ID                                               | Serial device                                                                                     |
|------------|--------------------------------------------------|---------------------------------------------------------------------------------------------------|
| frdm_k64f  | 02400b0132044e4500417009997b00316781000097969900 | /dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_02400b0132044e4500417009997b00316781000097969900-if01 |
| frdm_k64f  | 02400b0140214e45001b1018c89200349e11000097969900 | /dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_02400b0140214e45001b1018c89200349e11000097969900-if01 |

INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 5/7 frdm_k64f                 samples/subsys/usb/mass/sample.usb.mass_ram_none   PASSED (device 3.925s)
INFO    - 6/7 frdm_k64f                 samples/subsys/usb/mass/sample.usb.mass_ram_fat    PASSED (device 5.516s)
INFO    - 7/7 frdm_k64f                 samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none  FAILED Timeout (device 64.573s)
ERROR   - see: <path_to>/twister-out/frdm_k64f/samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none/handler.log

INFO    - 7 test scenarios (7 test instances) selected, 4 configurations skipped (4 by static filter, 0 at runtime).
INFO    - 2 of 7 test configurations passed (66.67%), 1 failed, 0 errored, 4 skipped with 0 warnings in 69.99 seconds
INFO    - In total 3 test cases were executed, 4 skipped on 1 out of total 567 platforms (0.18%)
INFO    - 3 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board     | ID                                               |   Counter |
|-----------|--------------------------------------------------|-----------|
| frdm_k64f | 02400b0132044e4500417009997b00316781000097969900 |         1 |
| frdm_k64f | 02400b0140214e45001b1018c89200349e11000097969900 |         2 |
INFO    - Saving reports...
INFO    - Writing JSON report <path_to>/twister-out/twister.json
INFO    - Writing xunit report <path_to>/twister-out/twister.xml...
INFO    - Writing xunit report <path_to>/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none on frdm_k64f failed (Timeout)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p frdm_k64f -s samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none
INFO    - or with west:
INFO    - west build -p -b frdm_k64f -T samples/subsys/usb/mass/sample.usb_device_next.mass_ram_none
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```

but the handler.log contains information like this.
```
[00:00:00.000,000] <inf> main: No file system selected
[00:00:00.002,000] <err> usbd_msc: Cannot register LUN SD
[00:00:00.002,000] <err> usbd_init: Failed to initialize class instance
[00:00:00.002,000] <err> usbd_init: Failed to init configuration 1
[00:00:00.002,000] <err> main: Failed to initialize device support
[00:00:00.002,000] <err> main: Failed to enable USB
```
It seems like this sample is prepared for check the external card reader but frdm_k64f does not support this feature.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0
 - West version:  v0.13.1
 - Latest main: [68d1d7d](https://github.com/zephyrproject-rtos/zephyr/commit/68d1d7d1b4754a9cabde830dc378eebe28aed3a9)

henrikbrixandersen, 2023-05-10:
Could be the same issue as seen here: https://github.com/zephyrproject-rtos/zephyr/issues/57654#issuecomment-1541718040

mmahadevan108, 2023-05-15:
@gbarkadiusz I ran the sample using west
`west build -b frdm_k64f zephyr\samples\subsys\usb\mass` 
The sample works as expected.

hakehuang, 2023-05-16:
@mmahadevan108 , this case is failed for very long time, you need use this config
```
west build -b frdm_k64f zephyr\samples\subsys\usb\mass -- -DCONF_FILE=usbd_next_prj.conf -DCONFIG_LOG_DEFAULT_LEVEL=3
```
and with below config it is ok
```
west build -b frdm_k64f zephyr\samples\subsys\usb\mass -- -DCONFIG_LOG_DEFAULT_LEVEL=3
```

gbarkadiusz, 2023-05-16:
@hakehuang You're right, after changing this parameter in the sample.yml, it's passing.
Before.
```
...
sample.usb_device_next.mass_ram_none:
    min_ram: 64
    depends_on: usb_device
    platform_allow:
      - nrf52840dk_nrf52840
      - frdm_k64f
    extra_args: CONF_FILE="usbd_next_prj.conf"
    extra_configs:
      - CONFIG_LOG_DEFAULT_LEVEL=3
    tags:
      - msd
      - usb
    harness: console
    harness_config:
      type: multi_line
      ordered: true
      regex:
        - "No file system selected"
        - "The device is put in USB mass storage mode."
 ...
```
After
```
...
sample.usb_device_next.mass_ram_none:
    min_ram: 64
    depends_on: usb_device
    platform_allow:
      - nrf52840dk_nrf52840
      - frdm_k64f
#    extra_args: CONF_FILE="usbd_next_prj.conf"
    extra_configs:
      - CONFIG_LOG_DEFAULT_LEVEL=3
    tags:
      - msd
      - usb
    harness: console
    harness_config:
      type: multi_line
      ordered: true
      regex:
        - "No file system selected"
        - "The device is put in USB mass storage mode."
...
```
So, maybe we need to prepare the PR for this change?

hakehuang, 2023-05-16:
@gbarkadiusz , no this is an issue to this config, and as it is for usbd_next_prj.conf so maybe usb maintainer need check this issue.
@tomaszduda23

tmon-nordic, 2023-06-05:
`USBD_MSC_LUNS_PER_INSTANCE` is one thing, but for some reason the device does not seem to respond with the USBS to `Mode Sense(6)` on LUN 0.

tmon-nordic, 2023-06-05:
The way endpoint stall in `Mode Sense(6)` is handled seems to differ between nRF and K64F.

If the bulk in endpoint stall at
https://github.com/zephyrproject-rtos/zephyr/blob/98810bc7a1649b4a8ccffa55176559f81512e283/subsys/usb/device_next/class/usbd_msc.c#L508-L515
is removed, then everything works fine. The stall is handled correctly on nRF52840DK therefore I think it is K64F udc driver. Note that just removing the stall is not really a solution.

On K64F, the CSW enqueued in `msc_send_csw()` doesn't make it to the bus. Note that this is the first queued IN data after IN endpoint STALL in the whole communication since USB reset (therefore it seems unlikely that `Mode Sense(6)` command is the culprit but rather a first victim).

PR #58827 (2023-06-06): samples: usb: mass: Set next stack max LUNs to 3
Mass storage sample contains separate Kconfig options for RAM, Flash and SDMMC disk drivers. All disk drivers can be enabled simultaneously and therefore the maximum number of concurrent LUNs is 3.

Configure Mass Storage class maximum LUNs to 3 to solve initialization failures on targets that enable more than one disk driver at a time. Each additional supported LUN increases the Mass Storage class memory usage by 48 bytes (struct scsi_ctx) on 32-bit targets. Allocating 3 SCSI context structures allows the sample to work regardless what disk dri [... cut]