zephyrproject-rtos/zephyr #57672: Common kernel object field for unifying kernel object stats reporting

peter-mitsis, 2023-05-08:
## Introduction

I am exploring an idea to unify kernel object stats reporting.

### Problem description

There are multiple places in the kernel where we collect statistical data about various kernel objects (thread runtime stats, memory usage stats for memory blocks, heaps, memory slabs, ...). Each of these have their own method to report stats. I would like to simplify this with a new framework. I understand that there is currently a stats subsystem (subsys/stats). If I am following it correctly, it is great for permanent features (drivers), but not so good for kernel objects that may be reused or have a lifespan.

### Proposed change

Kernel objects where possible, can be linked together and individually enabled to track statistical data. Linking them together can allow tools to walk the lists to automatically retrieve information in a structured way.

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

Before a kernel object can be linked to track that data, the object type must be registered during early system initialization. These registered object types will be tracked in a doubly linked to make it easier to cycle through them all.

    Obj Type 1 <--> Obj Type 2 <--> Obj Type 3 <--> ....

To put it a little more concretely, this could be something like ...

    Thread Objects <--> Memory Block Objects  <--> Heap Objects <--> ...

Each object type node would contain information such as  ...

```
struct k_obj_type {
    node  : doubly linked list node
    list   : list that contains all registered objects of this type
    stats  : pointer to descriptor that describes how to retrieve/report stats data (configurable)
   name / id : pointer to a string or unique ID to easily identify the object type (configurable)
}
```

The stats descriptor referenced above would follow the same pattern for each object type ...

```
struct stats_desc {
    raw_size  : size of structure used for internal tracking of statistics
    query_size : size of structure used for reporting the statistics
    raw : Function pointer to retrieve raw data
    query : Function pointer to retrieve reported data
    reset  : Function pointer to reset a kernel object's stats
    disable  : Function pointer to disable a specific kernel object from gathering stats
    enable  : Function pointer enable a specific kernel object to gather stats
}
```

Kernel objects would have embedded within them an optional `struct k_obj  obj;` field.
```
struct k_obj {
    node  : node in k_obj_type's list (doubly linked for fast insertion/removal)
    type  : a back pointer to the k_obj_type
    name  : optional pointer to a name (does not have to be unique)
    stats  : optional pointer to raw stats data (if NULL, stats are not tracked for it)
    .... future expansion ... ?
}
```

### Dependencies

This should not have too much in the line of immediate dependencies on other components. However, it some aspects of it may touch upon other features (stats subsystem, tracing/tracking, other?)

### Concerns and Unresolved Questions

I expect that there will be concerns around at least duplication, footprint and performance.

Why not use the stats subsystem? As far as I can tell (and I may be wrong) the stats subsystem is geared towards permanent things. It is not geared for items that may be re-used or dropped (such as threads). When it comes to resetting the stats, it assumes that the stats structure can be zeroed. This is not always ideal (for example when the stats for memory usage track peak usage, resetting the peak or current usage to zero does not make sense when memory is still in use). It also appears rather cumbersome to set up for individual kernel objects.

Footprint - Yes, it will increase footprint. However, this is configurable. I am currently thinking that there can be a core component that can be configured by kernel object (CONFIG_KOBJ, CONFIG_KOBJ_STATS, CONFIG_KOBJ_THREADS, CONFIG_KOBJ_MEM_BLOCKS, ...) so that we do not completely blow up the footprint. Just enable it for the parts that are wanted.

Does this duplicate tracing / tracking? Perhaps somewhat. Or, perhaps some of the tracing/tracking can be rolled into this in the future as this could link together the kernel objects that people want to track as well.

Performance - Enabling stats gathering will impact performance somewhat--hopefully not too much.

What are the APIs? This is not 100% settled yet, but they are going to focus around the k_obj and k_obj_type structures--initialization, links and unlinks. There may variations for stats linking as well since stats are a configurable sub-option to the K_OBJ.  The various stats routines are going to be accessed via macros--something like K_OBJ_STATS_xxx().

The major benefit that I see is that since all the stats objects (and potentially other objects) are linked together, 3rd party tools/debuggers can auto-extract their stats information.

carlescufi, 2023-05-16:
Architecture WG:

- Presentation by @peter-mitsis, slides to be attached.
- @stephanosio asks how this is related to the kobj infrastructure for userspace. @peter-mitsis needs to look into it
- @galak asks whether this would replace driver statistics. @peter-mitsis sees this as a parallel effort
- @carlescufi and @mbolivar-nordic ask whether the feature can be turned off entirely, the answer is yes
- @nashif points out that the name can be misleading, as @stephanosio already mentioned
- @peter-mitsis will ping @andyross to ensure he reviews this RFC
- @nashif asks how the actual stats will be retrieved from memory and out of the SoC. @microbuilder points out that right now mcumgr has the functionality to query stats, as well as using gdb.
- @henrikbrixandersen thinks it does make sense to have two different stat systems, because one (this one) is more of a debugging tool and the other (driver stats) is about gathering info about hardware peripheral. The latter is much more restricted in scope and width.

peter-mitsis, 2023-05-23:
[Object Core.pptx](https://github.com/zephyrproject-rtos/zephyr/files/11544795/Object.Core.pptx)

Slides uploaded.

andyross, 2023-05-23:
I don't think I have many concerns other than the cosmetic one: "kernel object / kobj" is a term of art already appropriated by the userspace subsystem, so we need to untangle that somehow (though adding these stats to existing kobj structs makes perfect sense of course).

It also wouldn't hurt to write up a section on expected synchronization behavior.  This has been a persistent wart on all our tracing/enumerating APIs.   Some stats ("how many?") can be read atomically and don't need it, others only make sense in multi-word sets ("received" X "transmitted") and absolutely need it.  I don't know necessarily that the API needs to address this directly but we should have a best practices kind of story explaining how to do it correctly.

Also: the enumeration API for this almost certainly does need locking if the list of objects is going to be runtime-mutable, which it likely is.

closed 2023-10-02 by peter-mitsis